lvgl/lvgl #239: Problem with CPU usage

wxd0216, 2018-05-08:
It's a great library, I'm playing with it.

I want to use it to control a screen(1440X540). So, I created  a task (vTest_iChangePictur) by FreeRTOS, periodically call lv_task_handler in 5ms. I have also ported the Demo from the pc_simulator, I tracked the CPU usage rate. Not too good for me.

Logger		1402		<1%
vTest_iChangePictur		294451		83%
InputTask		0		<1%
OutputTask		0		<1%
vTest_iDMAFinish_Ta		0		<1%
IDLE		0		<1%
SchedTask		63906		15%
Ethernet_Task		5489		1%
Tmr Svc		710		<1%

I'm trying to provide an empty  **ex_disp_flush**  function to confirm whether the problem is caused by the underlying driver. But the results are the same.

Could you offer me some advice? 

The attachment is my configuration file.

[lv_conf.zip](https://github.com/littlevgl/lvgl/files/1982895/lv_conf.zip)

Thank you every so much.

kisvegabor, 2018-05-08:
Hi,

What happens if you change `LV_REFR_PERIOD` to 500 ms?

wxd0216, 2018-05-10:
I changed the LV_REFR_PERIOD, and do the some test.

the following is the result:
[image]

I randomly counted ten CPU usage for each LV_REFR_PERIOD.

It did change, but I didn't feel good enough.

Is there a better option?

thanks!

kisvegabor, 2018-05-10:
Thank you for the measurement!

Please try an other thing too:
1. Revert LV_REFR_PERIOD to 500 ms
2. Do not register any input devices. Often the input device driver blocks the CPU

Does it change something?

wxd0216, 2018-05-10:
For now，I didn't register any input drivers.

void lv_display_init(void)
{
    /***********************
     * Initialize LittlevGL
     ***********************/
    lv_init();

    lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
    lv_disp_drv_init(&disp_drv);                    /*Basic initialization*/

    /*Set up the functions to access to your display*/
    disp_drv.disp_flush = ex_disp_flush;            /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/

    disp_drv.disp_fill = ex_disp_fill;              /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
    disp_drv.disp_map = ex_disp_map;                /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/

#if 0
    /*Optionally add functions to access the GPU. (Only in buffered mode, LV_VDB_SIZE != 0)*/
    disp_drv.mem_blend = ex_mem_blend;              /*Blend two color array using opacity*/
    disp_drv.mem_fill = ex_mem_fill;                /*Fill a memory array with a color*/
#endif

    /*Finally register the driver*/
    lv_disp_drv_register(&disp_drv);  
}

wxd0216, 2018-05-10:
For now，I didn't register any input drivers.

kisvegabor, 2018-05-11:
Hi,

Thank for the new results!

The "core" functions like `lv_obj_set_pos/size` checks if the value is the same like the current and does nothing in this case. However this functionality is not added consequently in higher level function. I will review it for the next release! 

Seeing your measurements there is something strange: 10 times slower screen refresh period (500 ms instead of 50 ms) leads to only  half CPU usage (30 % instead of 60 %). It seems to me not  the drawing is responsible for the high CPU usage. 

I have a few other questions (please set the refresh period back to 50 ms):
1. What is the CPU usage if you do not refresh anything on the screen? (not create `update_gauge`)
2. What if you don't register the display driver? Now you should enable `update_gauge`. From this we will see the time of rendering (without transferring the image to the screen)
3. How did you call `lv_task_handler` and `lv_tick_inv()`?
4. What is the speed of your MCU?

wxd0216, 2018-05-14:
Hi,
Thanks for your formatting.

1.yes, I have turned on compiler optimization.
2.My next plan is to use GPU.

When I use GPU, I will check the performance with the **# benchmark application**.

Thank you for your support and let me know more about # LV.

closed 2018-05-14 by wxd0216

kisvegabor, 2018-05-14:
Okay, I'm waiting for the news. :)
On this large display the GPU should help a lot to speed up the drawing.