lvgl/lvgl #240: demo stops

liubingyan123, 2018-05-08:
Hi, I am workong on porting lvgl to stm32f413 discovery board, and it works. but the picture stops when I slide from "Write" tab to "List" tab, shown as flowing picture, I cheak the heap and stack(demo runs on FreeRTOS),  but find nothing abnormal， how does it happen? what will cause it? please offer me some advices, thank you!
[image]

closed 2018-05-08 by liubingyan123

kisvegabor, 2018-05-08:
Does it solved meanwhile?

liubingyan123, 2018-06-14:
yes,  I have solved this problem, and lvgl works well in STM32F413 discovery board.

kisvegabor, 2018-06-14:
Can you tell us what was the problem? Just to learn from it.