lvgl/lvgl #241: preserve btn icon after clrscreen

Masi2017, 2018-05-08:
Hi all,
I created a btn and then once btn is pressed I want to load a container to clear the whole screen but still I need to preserve the btn Icon , what should be done?

kisvegabor, 2018-05-09:
Hi,

So you want to keep the button on the top to be always visible regardless the loaded container?

Masi2017, 2018-05-14:
@kisvegabor Yes , I added the btn to the group and after that I created the container, I focused the btn but btn is not visible.

Masi2017, 2018-05-14:
I have only up/down/left/right and OK key. now I want to have a keyboard and once user typed a character I should search from a table and display a list of options that matches with the character ( same as address book) , see the below photo:
[image]

once options are displayed user may wanted to selected from them , but all the keys are assigned to the keyboard. One solution can be using up/down key if one of the btns from the first row of the keyboard or last row have been selected ( if for example "`1#"  is selected form keyboard pressing UP key do nothing) . I need to know which btn from the keyboard or at least which row currently has been selected and if it's the first row and user presses UP arrow key , I can focus other object rather than keyboard and user can move among them using Left/Right arrow keys and also retuns back to the keyboard.

kisvegabor, 2018-05-16:
If the container covers the icon then it won't be visible. But you can create a simple layer system: 
1. Create two objects on a screen  with transparent style.
2. Everything created on the second transparent object will covert the objects on the first.
More info here: https://littlevgl.com/basics#layers
 
For suggestions I have an idea: Similarly to *Button matrix* you can set a new map for the keyboard as well. Your new map should have an extra row on the top with the suggestions.
Maybe it would be better to use a Button matrix instead of the Keyboard because the Button matrix is more flexible.

Masi2017, 2018-05-17:
oh sorry the previous explanation was for another issue!!! It was for #242! my bad!!

kisvegabor, 2018-07-17:
As there was no activity here for while I suppose you solved the issue!

If you have further questions about this topic please comment here and reopen the issue.

closed 2018-07-17 by kisvegabor