lvgl/lvgl #243: Hang with 100% CPU usage

vihai, 2018-05-09:
Hello,

I'm testing the library, latest git, on an ARM board with a framebuffer based small LCD (128x128).

I successfully compiled the hello world, it starts fine and displays the label. The tick loop runs a bunch of times (8 or so) and then lv_task_handler() freezes with an infinite loop.

I tracked it down to lv_task_exec() call in lv_misc/lv_task.c:101

This is the code in its minimal form:

```
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    /*Linux frame buffer device init*/
    fbdev_init();

    /*Add a display the LittlevGL sing the frame buffer driver*/
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.disp_flush = fbdev_flush;      /*It flushes the internal graphical buffer to the frame buffer*/
    lv_disp_drv_register(&disp_drv);

    /*Handle LitlevGL tasks (tickless mode)*/
    while(1) {
        lv_tick_inc(5);
        
        usleep(5000);
        
        lv_task_handler();
    }

    return 0;
}
```

kisvegabor, 2018-05-09:
Hi,

Please try to update to the `beta` branch. There were some fixes in lv_task.

vihai, 2018-05-11:
Switched to beta branch but the issue is still there. I will try digging a bit deeper.

kisvegabor, 2018-05-11:
`lv_task_exec` calls the LittlevGL related functions. It would be great to know which function causes the freeze.

vihai, 2018-05-11:
Okay, I traced the problem down to lv_vdb_get() in lvgl/lv_core/lv_vdb.c

This is the line:

```
    while(vdb_state != LV_VDB_STATE_ACTIVE);
```
Setting aside the issue for which it doesn't become active, this kind of code is a big no-no for any application in which the CPU is supposed to be halted most of the time, it eats power and wastes CPU cycles.

I suppose it does wait for a virtual framebuffer to become usable (to synchronize with refresh?) before writing.

I think you should restructure your library to be fully event-driven (asynchronous) with no blocking (or infinitely looping!) code :)

kisvegabor, 2018-05-11:
Do you call `lv_flush_ready()` at the end of your disp_flush function?

kisvegabor, 2018-05-15:
Which infinite loop do you mean?

Just to be sure: in _lv_drivers/display/fbdev.c_ really there is a `lv_flush_ready()` at the end of `fbdev_flush()`?

And please check `fbdev_flush` doesn't return at the beginning during the error checking. (I just updated this error handling to call `lv_flush_ready` on error too)