lvgl/lvgl #245: the default objects are too big for my tiny lcd，how to configure？

BFDX, 2018-05-10:
hi, i have a tiny lcd screen with a resolution of 162x132，i tried some objects, i found they are too big for my screen , is there any options i can take to configure the object with a small style?

BFDX, 2018-05-11:
thank you for reply， after i use lv_obj_set_size() the list object get smaller, but as you can see below, i couldn't get full screen display, the left and up screen area are wasted a lot. how do i configure then? if i want a object to use the full screen for displaying.
[image]
**my code:**
void demo_create(void)
{
    /*Crate the list*/
    lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
    lv_label_set_style(list1, &style);
    lv_obj_set_size(list1, 150, 120);
    lv_obj_align(list1, NULL, LV_ALIGN_IN_TOP_LEFT, 20, 40);

    /*Add list elements*/
    lv_list_add(list1, SYMBOL_FILE, "New", list_btn_action);
    lv_list_add(list1, SYMBOL_DIRECTORY, "Open", list_btn_action);
    lv_list_add(list1, SYMBOL_CLOSE, "Delete", list_btn_action);
    lv_list_add(list1, SYMBOL_EDIT, "Edit", list_btn_action);
    lv_list_add(list1, SYMBOL_SAVE, "Save", list_btn_action);
}

qwert1213131, 2018-05-11:
You can have a try to modify the LV_DPI <lv_conf.h>.
such as 60.

kisvegabor, 2018-05-11:
@BFDX 
You wrote: 
`    lv_obj_align(list1, NULL, LV_ALIGN_IN_TOP_LEFT, 20, 40);`

If you want a full screen object why not position it exactly into the corner: 
`   lv_obj_align(list1, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);`

Item's height comes from it's style. I updated to documentation to tell some info about it: https://littlevgl.com/object-types/list-lv_list
The height comes from the font height  + the vertical padding (style.body.padding.ver) property of the button's current style.

kisvegabor, 2018-07-07:
As this issue wasn't active for months I close it.
If you have further feel free to reopen it!

closed 2018-07-07 by kisvegabor