lvgl/lvgl #248: [Question] Porting SDL 1.2 to lvgl

avafinger, 2018-05-13:
@kisvegabor ,
Can you give some directions to port SDL 1.2 to lvgl, the easiest way?
I will try to port a small SDL program but I only know the basics so if you have something I could start with would be great. Documentation, samples, anything.
BR,

LiuZhe1, 2018-05-14:
what do you mean "port SDL 1.2 to lvgl"?
SDL is the c compiler package for eclipse. for lvgl you have to install sdl to eclipse, not to lvgl.

avafinger, 2018-05-14:
SDL is a C library to access and draw directly on the hardware, not specific to Eclipse. I have a program in C that draws some objects to the surface using SDL, i just want to port that program, not the SDL library per se. I don't want to use eclipse, just to have some 'function' correlation. There are some objects in PNG format that i will need to draw, that may be complex but i just need some basics to start.

kisvegabor, 2018-05-14:
Hi,

So do you want to use some SDL features (like drawing PNG) in LittlevGL?

avafinger, 2018-05-15:
Hi @kisvegabor 

I have pushed the SDL program i wish to run with **lvgl** to GitHub (so anyone can have a look at)  **instead of** SDL (X11/Desktop). It seems unrealistic to have 130 FPS on a 2.8TFT  SPI which defaults to 62 FPS:
https://github.com/avafinger/ballfield-1.0#arm64

I just want to learn the basics on how to port that to **lvgl**, the program makes use of loading PNG image (array of images) and render it to the surface and switch the double buffer. (i am not expert in SDL )
I have seen **png2c** for manipulating PNG but i was thinking to have it dynamically, and if you plan to add this in a future release would be really great.

Just think of this as a learning path, like a function to draw a PNG image, and a correlation between the two.
I also wanted to measure the FPS with lvgl, can you point to your FPS test program?

kisvegabor, 2018-05-16:
Hi,

Adding a PNG decoder library to LittlevGL would be very useful. It is a planned feature but I need to find some time to add it.

Here is the Benchmark application of LittelvGL: https://github.com/littlevgl/lv_examples/tree/master/lv_apps/benchmark

avafinger, 2018-05-16:
Thank you for that. I will try and see the results.
In my humble opinion, LittelvGL should have its own PNG / JPEG decoder and not be dependent on any external lib like libpng / libjpeg. I will try to help somehow. Yeah, time is a constraint for me too. ;)

kisvegabor, 2018-05-16:
For LittelvGL the it would be the best to have a function like .`png_get_pixels(x1, y1, x2, y2, buffer);`

So any contribution to this direction would be awesome! :)

kisvegabor, 2018-07-16:
I opened a dedicated issue for PNG/JPG decoder support: https://github.com/littlevgl/lvgl/issues/273

If you have ideas in this field please use that thread.

closed 2018-07-16 by kisvegabor