lvgl/lvgl #249: SSD1306 Driver

jano2358, 2018-05-13:
Hi!

I wanted to use the LV with the SSD1306 and since I didn't see any driver for it I begin to write it.
So far the driver begins to work, I made the function fill (right now it writes a buffer to be a bit faster, then I write the whole buffer to the screen) and I can draw things over there (still having problems with dejavu fonts, didn't try other yet, I don't know why, but this is for later). 
The question I have today is what is supposed to be made by the functions map and flush?

C47D, 2018-05-13:
Hi,

There's a driver for the SSD1306 on the new_api branch of the lv_drivers repo, i didn't check if that driver have the map and flush functions but maybe you can take a look: https://github.com/littlevgl/lv_drivers/tree/new_api/display

jano2358, 2018-05-13:
Great! there is also a driver for the ILI9341, which is the next thing I wanted to have. Now all I need to do is learn how to use it. Thanks!

C47D, 2018-05-13:
No problem, just be aware of the API, as the branch name points out it's using a new API, maybe @kisvegabor can help us to know the differences between the "old" and "new" API.

Regards

kisvegabor, 2018-05-14:
Hi,

the "new_api" means, new API for drivers and not for LittelvGL. So it's still fully compatible with "old" LittlevGL.
The main contributor of these updates is @Zaltora. I'm sure he can help if you stuck with it. Moreover - because the new API is still in development - any feedback are very welcome!

jano2358, 2018-05-14:
Perfect! I was wondering about compatibilities too, thanks for the info on that. I will try it soon.

kisvegabor, 2018-07-15:
As there was no activity here for a while I close this issue.

If you have further questions or remarks related to drivers, please open an issue in [lv_drivers](https://github.com/littlevgl/lv_drivers) repository.

closed 2018-07-15 by kisvegabor