lvgl/lvgl #251: Fast update on Discovery STM32F4/STM32F7

bkht, 2018-05-14:
Hi,

I just discovered lvgl, it is very great!
Using Discovery STM32F, below is my code for transferring the pixels to a discovery board.
But I would like to use DMA (or DMA2D) to speed it up, and to offload the MCU.
Is there some example code to use DMA with this?

`static void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
	// In the flush function you can use DMA or any hardware to do the flushing in the
	// background, but when the flushing is ready you have to call lv_flush_ready();

	/*The most simple case (but also the slowest) to put all pixels to the screen one-by-one*/

    int32_t x;
    int32_t y;
    for(y = y1; y <= y2; y++) {
        for(x = x1; x <= x2; x++) {
            /* Put a pixel to the display. For example: */
            /* put_px(x, y, *color_p)*/
        	uint32_t pixelColor = 0xff000000 |
        			color_p->red << 16 |
					color_p->green << 8 |
					color_p->blue;
        	lcd.DrawPixel(x, y, pixelColor);
            color_p++;
        }
    }

    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_flush_ready();               /*Call 'lv_fluh_ready()' when ready*/
}

BTW, I'm Running mbed-os on the platform and board drivers (BSP_DISCO_F469NI, LCD_DISCO_F469NI)

`

bkht, 2018-05-14:
OK, thank you very much. Performance is not super, but about OK. :-)
I will see if I can figure this, and give feedback once I solved it.

kisvegabor, 2018-05-15:
Hi,

AC6 (or SystemWorkbranch) is an Eclipse based IDE from STM MCUs. There are lot of example project for AC6 from STM. This post contains a getting started too:  https://littlevgl.com/blog/8/embedded-gui-on-stm32-discovery-board-with-littlev-graphics-library

However I don't know mbed-os but there are some STM32 project for AC6  in [lv_boards](https://github.com/littlevgl/lv_boards) repo.  All od them uses DMA to transfer pixels.

bkht, 2018-05-16:
Hi Gabor, thank you very much for pointing me to this. I use eclipse with the mbed and ARM_GCC.
But I have update problems with it, not finding the update repositories, so it needs fixing sooner or later anyway. Now I'm also diving into m2m DMA with interrupt on mbed, but I need to write some code using HAL (or maybe CMSIS). Starting with AC6 would certainly help me learning and finding a solution for mbed, I hope. I enjoy your library :-) Thanks to using AA, it looks very good.

caveman1996, 2018-05-16:
For people that start a new project for these boards (without the help of the mBed platform), use Atollic TrueStudio. AC6 need more configuration to make a project compile and from what I saw from my class mates, Atollic look simpler.

kisvegabor, 2018-05-16:
@bkht  Then good luck with the DMA. :) I hope the example will help!
 
@caveman1996 Yes, Atollic is a very popular alternative. However I didn't faced any configuration difficulties with AC6.

bkht, 2018-05-16:
Hi, lots to learn here! Thank you very much! I will check out AC6 and Atollic TrueStudio.

bkht, 2018-05-17:
Hi Gabor,
I tested a simple memory-to-memory DMA with interrupt in mbed, using HAL functions, and got this working very quickly.
If for updating the display we use DMA, instead of plotting all the pixels, do we use m2m DMA to update the display?
Or how does this look like?
Kind regards,
Jack.

kisvegabor, 2018-05-18:
Here I used `HAL_DMA_Start_IT` in `disp_flush`: https://github.com/littlevgl/lv_boards/blob/master/stm32f429_discovery_no_os/hal_stm_lvgl/tft/tft.c#L191

Yes, I'm interested in such a project! However I never used mbed-os so far I'm curious about it!

bkht, 2018-05-19:
Hi Gabor,
for me, it gives some scattered horizontal lines, evenly spread over the vertical height on the display.

Now use DMA without interrupt and wait for the transfer to complete.
It is a lot faster than without using DMA. DMA with interrupt would be better, to free the
MCU to do better things than waiting, I will try it later.

    /**
    * @brief  Updates just a region of the screen containing an object
    * @note   It configures DMA for memory to memory transfers
    * @retval None
    */
    static void tft_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
    {
        // Updates just a rectangular region of the screen containing an object
        // such as a button, meter, or chart, that needs to be updated.
        // In the flush function you can use DMA to do the flushing in the background.
        // Each transfer is a contiguous part of source and destination,
        // basically a horizontal line, so that an DMA transfer can handle it.
        // For every new vertical position we first need to calculate the address
        // for the destination frame buffer.
        // Because color_p just only holds the data of the rectangular region,
        // it just needs to increment that pointer by the DMA size for every line.
        // Once the flushing is ready we have to call lv_flush_ready();

        /*Return if the area is out the screen*/
        if(x2 < 0) return;
        if(y2 < 0) return;
        if(x1 > TFT_HOR_RES - 1) return;
        if(y1 > TFT_VER_RES - 1) return;

        /*Truncate the area to the screen*/
        int32_t act_x1 = x1 < 0 ? 0 : x1;
        int32_t act_y1 = y1 < 0 ? 0 : y1;
        int32_t act_x2 = x2 > TFT_HOR_RES - 1 ? TFT_HOR_RES - 1 : x2;
        int32_t act_y2 = y2 > TFT_VER_RES - 1 ? TFT_VER_RES - 1 : y2;

        x1_flush = act_x1;
        y1_flush = act_y1;
        x2_flush = act_x2;
        y2_fill = act_y2;
        y_fill_act = act_y1;
        buf_to_flush = color_p;

        // A faster than the pixel-by-pixel approach is to put all pixels to the
        // screen using DMA
        static const uint32_t *ptrFBStart = lcd.GetFBStartAdress();
        uint32_t dmaSize = ((x2_flush-x1_flush) + 1);
        while(y1_flush <= y2_fill) {
            uint32_t FBIndex = (uint32_t)&ptrFBStart[y1_flush * TFT_HOR_RES + x1_flush];

            // Start the DMA transfer using polling mode
            if (HAL_DMA_Start(&hdma_memtomem_dma2_stream0,
                    (uint32_t)color_p,
                    (uint32_t)FBIndex,
                    dmaSize) != HAL_OK)
            {
                while(1);    // Halt on error
            }
            // Polling for transfer complete, if not using the interrupt
            HAL_DMA_PollForTransfer(&hdma_memtomem_dma2_stream0,
                                    HAL_DMA_FULL_TRANSFER, 1000);
            color_p += dmaSize;
            y1_flush++;
        }

        // The most simple case (but also the slowest) to update the screen
        // pixel-by-pixel
    //    for(int32_t y = y1; y <= y2; y++) {
    //        for(int32_t x = x1; x <= x2; x++) {
    //            // Put a pixel to the display.
    //            // For example: put_px(x, y, *color_p)
    //            lcd.DrawPixel(x, y, color_p->full);
    //            color_p++;
    //        }
    //    }

        // Inform the graphics library that you are ready with the flushing,
        // by calling 'lv_fluh_ready()' when ready
        lv_flush_ready();
    }

    /**
    * Enable DMA controller clock
    * Configure DMA for memory to memory transfers
    *   hdma_memtomem_dma2_stream0
    */
    static void MX_DMA_Init(void)
    {
        // DMA controller clock enable
        __HAL_RCC_DMA2_CLK_ENABLE();

        NVIC_ClearPendingIRQ(DMA2_Stream0_IRQn);
        HAL_NVIC_DisableIRQ(DMA2_Stream0_IRQn); 	// DMA IRQ Disable

        // Configure DMA request hdma_memtomem_dma2_stream0 on DMA2_Stream0
        hdma_memtomem_dma2_stream0.Instance = DMA2_Stream0;
        hdma_memtomem_dma2_stream0.Init.Channel = DMA_CHANNEL_0;
        hdma_memtomem_dma2_stream0.Init.Direction = DMA_MEMORY_TO_MEMORY;
        hdma_memtomem_dma2_stream0.Init.PeriphInc = DMA_PINC_ENABLE;
        hdma_memtomem_dma2_stream0.Init.MemInc = DMA_MINC_ENABLE;
        hdma_memtomem_dma2_stream0.Init.PeriphDataAlignment = DMA_PDATAALIGN_WORD;
        hdma_memtomem_dma2_stream0.Init.MemDataAlignment = DMA_MDATAALIGN_WORD;
        hdma_memtomem_dma2_stream0.Init.Mode = DMA_NORMAL;
        hdma_memtomem_dma2_stream0.Init.Priority = DMA_PRIORITY_LOW;
        hdma_memtomem_dma2_stream0.Init.FIFOMode = DMA_FIFOMODE_ENABLE;
        hdma_memtomem_dma2_stream0.Init.FIFOThreshold = DMA_FIFO_THRESHOLD_FULL;
        hdma_memtomem_dma2_stream0.Init.MemBurst = DMA_MBURST_SINGLE;
        hdma_memtomem_dma2_stream0.Init.PeriphBurst = DMA_PBURST_SINGLE;

        // Set the transfer complete interrupt callback function
        hdma_memtomem_dma2_stream0.XferCpltCallback = XferCpltCallback;
        hdma_memtomem_dma2_stream0.XferErrorCallback = XferErrorCallback;

        if (HAL_DMA_Init(&hdma_memtomem_dma2_stream0) != HAL_OK)
        {
            _Error_Handler(__FILE__, __LINE__);
        }

        // DMA interrupt init
        // DMA2_Stream0_IRQn interrupt configuration
        NVIC_SetVector(DMA2_Stream0_IRQn, (uint32_t )my_irq_handler);
        HAL_NVIC_SetPriority(DMA2_Stream0_IRQn, 1, 0);  // IRQn_Type, uint32_t, uint32_t
        HAL_NVIC_EnableIRQ(DMA2_Stream0_IRQn);  // DMA IRQ Enable
    }

I hope this helps others.

kisvegabor, 2018-05-22:
Hi,

What happens if you change `LV_VDB_SIZE` (in lv_conf.h). Let's half it! Do you see twice as much "lines"?

bkht, 2018-05-22:
Hi Gabor,

When I half it like

    #define LV_VDB_SIZE         (10 * LV_HOR_RES)

I have about doubled the number of horizontal lines (about 48 lines), where I can better see the buttons graphical items in place. Note how this correlates to the vertical resolution of the display, which is 480.

When I half it like

    #define LV_VDB_SIZE         (5 * LV_HOR_RES)

I have again doubled the number of horizontal lines, where I can even better see the buttons graphical items in place.

If I make it 1, It draws every line, but is is very slow.

Kind regards,
Jack.

kisvegabor, 2018-05-22:
It seems somehow you don't flushes the last line. However you code seems sine because you wrote  `while(y1_flush <= y2_fill) `

bkht, 2018-05-22:
Hi Gabor, thank you for the quick response! The code where I wrote above, 3 days ago, actually works just fine, but it doesn't use interrupts.
I tried to implement a DMA with interrupt approach, but that gives some trouble so far.
With the code I posted this morning, I have those limited number of lines across the screen, as that code is missing to keep track where it left off, I think.
Somehow I have to create multiple DMA transfers to update the graphics in horizontal lines, in order to be able to use DMA over some contiguous frame buffer space.
So I need to trigger DMA transfers until the object has been entirely repainted and keep track when the last part has been written and then lv_flush_ready can be called.
But I wasn't successful in that approach, yet. Once I manage to do that, I will post the code here.

bkht, 2018-05-22:
Hi, it works better, but I have two small screens next to each other in top half of the display.
So, it appears if the destination pointer updates half the amount for each DMA.
I have to check this a bit more. I use 24 bit mode and so 32-bit DMA transfers, that's correct in my config:

    /**
    * @file lv_conf.h
    * 
    */

    //#if 0 /*Remove this to enable the content (Delete the last #endif too!)*/

    #ifndef LV_CONF_H
    #define LV_CONF_H

    /*----------------
    * Dynamic memory
    *----------------*/

    /* Memory size which will be used by the library
    * to store the graphical objects and other data */
    #define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
    #if LV_MEM_CUSTOM == 0
    #define LV_MEM_SIZE    (32U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
    #define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
    #define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
    #else       /*LV_MEM_CUSTOM*/
    #define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
    #define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
    #define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
    #endif     /*LV_MEM_CUSTOM*/

    /*===================
    Graphical settings
    *===================*/

    /* Horizontal and vertical resolution of the library.*/
    #define LV_HOR_RES          (800)
    #define LV_VER_RES          (480)
    #define LV_DPI              100

    #define TFT_HOR_RES LV_HOR_RES
    #define TFT_VER_RES LV_VER_RES

    /* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
    * Required for buffered drawing, opacity and anti-aliasing
    * VDB makes the double buffering, you don't need to deal with it!
    * Typical size: ~1/10 screen */
    //#define LV_VDB_SIZE         (20 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
    #define LV_VDB_SIZE         (20 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
    #define LV_VDB_ADR          0                  /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

    /* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
    * The flushing should use DMA to write the frame buffer in the background*/
    #define LV_VDB_DOUBLE       0       /*1: Enable the use of 2 VDBs*/
    #define LV_VDB2_ADR         0       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

    /* Enable anti-aliasing (lines, and radiuses will be smoothed) */
    #define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

    /*Screen refresh settings*/
    #define LV_REFR_PERIOD      50    /*Screen refresh period in milliseconds*/
    #define LV_INV_FIFO_SIZE    32    /*The average count of objects on a screen */

    /*=================
    Misc. setting
    *=================*/

    /*Input device settings*/
    #define LV_INDEV_READ_PERIOD            50                     /*Input device read period in milliseconds*/
    #define LV_INDEV_POINT_MARKER           0                      /*Mark the pressed points  (required: USE_LV_REAL_DRAW = 1)*/
[... 160 lines cut ...]
    #if USE_LV_LIST != 0
    #define LV_LIST_FOCUS_TIME  100 /*Default animation time of focusing to a list element [ms] (0: no animation)  */
    #endif

    /*Drop down list (dependencies: lv_page, lv_label)*/
    #define USE_LV_DDLIST    1
    #if USE_LV_DDLIST != 0
    #define LV_DDLIST_ANIM_TIME     200     /*Open and close default animation time [ms] (0: no animation)*/
    #endif

    /*Roller (dependencies: lv_ddlist)*/
    #define USE_LV_ROLLER    1
    #if USE_LV_ROLLER != 0
    #define LV_ROLLER_ANIM_TIME     200     /*Focus animation time [ms] (0: no animation)*/
    #endif

    /*Slider (dependencies: lv_bar)*/
    #define USE_LV_SLIDER    1

    /*Switch (dependencies: lv_slider)*/
    #define USE_LV_SW       1

    #endif /*LV_CONF_H*/

    //#endif /*Remove this to enable the content*/

kisvegabor, 2018-05-23:
Congratulation! :)

`lv_tick_inc(1);` should go to a 1 ms Timer interrupt and `lv_task_handler()` to a loop called in every 3..30 ms (no so critical).

effeci, 2018-06-13:
@bkht I would like to know if you have any news on a f746 demo. Thanks!

bkht, 2018-06-13:
@effeci Hi, de demo works very well, thanks to Gabor, who pointed me to some directions.
Note, I use Mbed-OS, and DMA with interrupt for littlevgl.
It is just a screen with some widgets on it.
The buttons control LEDs and the slider does move a point in the chart.
Just discovering littlevgl.
I hope to contribute to it later as well, if I find some time.

effeci, 2018-06-13:
Glad to eat you back! Is it possible to have any access to this demo?

bkht, 2018-06-13:
Sure! I'm learning from people here and I like to help other people as well. :-)
Just uploaded, here you go:
https://github.com/bkht/DISCO-F469NI_LCD_Demo2

effeci, 2018-06-14:
Thank you! Is this compatible with the f746ng development board? I agree with you but, in my case I feel a little bit ashamed of not having nothing to share !

kisvegabor, 2018-07-15:
@bkht Thank you for sharing your project! I'm sure it will help others who also uses mbedOS!

I close this issue because the goal seems to be reached. If you have further questions or remarks feel free to comment here and reopen this issue.

closed 2018-07-15 by kisvegabor

nmathon, 2018-10-04:
Hi everyone. Did you test it on F746NG ? 
Thanks

embeddedt, 2018-11-20:
@effeci @nmathon In general you will need to make some changes in order to get an F4 demo running on an F7, due to pin changes and possible hardware incompatibilities.