lvgl/lvgl #252: Round line ends or flat related to line angle possible?

bkht, 2018-05-14:
Hi,
thanks for this great library!
I notice that the line ends are either ending in a vertical or in a horizontal fashion, I think depending on the slope.
Would it be nice if it could be the related to the angle of the line and optionally have round line ends?
Best Regards!

kisvegabor, 2018-05-15:
Hi,

Your welcome! :) And thank you for the suggestion!

Yes, it's horizontal or vertical now depending on the slope as you said.

I would be great to add at least the perpendicular line ending, but it's not as simple as it seems. Because to properly handle opacity you need to care about not drawing a pixel twice. And to keep a good performance the advanced trigonometrical functions should be avoided.

So it's a planned feature but needs some time to implement.

bkht, 2018-05-16:
Hi Gabor,
thank you very much!
I see it's very important to draw the pixel only once, and performance is indeed important too.
You did an awesome job!

I tested below code (but not in littlevgl), and it works well.
https://github.com/ArminJo/STMF3-Discovery-Demos/blob/master/lib/graphics/src/thickLine.cpp
I didn't use 'drawThickLineSimple' as this once doesn't fill all the pixels.
The function 'drawThickLine' does work wonderfull,  it has perpendicular line endings (caps), it uses 'drawLineOverlap'.
I discovered you library just a few days ago, and like to use it is for hobby and learning purposes. 
Thanks a lot!
Regards,
Jack.

kisvegabor, 2018-05-16:
Thank you very much Jack!
I will have look. 
I mark this suggestion to be part of the next release!

bkht, 2018-05-16:
Thank you very much!

kisvegabor, 2018-05-20:
Hi,
I was working on the perpendicular line ending. It's quite challenging to keep well optimized both for memory and performance but I was doing a good progress! It's not working for line direction but works for one. It should similar for the other cases too. 
I added anti-aliasing as well.

Looks like this (different colors means different parts of drawing)
[image]

bkht, 2018-05-22:
Hi Gabor,

this is a routine that draws thick circles, without drawing pixels twice or missing pixels.
Not sure how fast it will be, but I think it is OK.
I implemented this on the Discovery F469NI.

    void BSP_LCD_DrawThickCircle(uint16_t Xpos, uint16_t Ypos, uint16_t Radius, uint16_t lineThickness)
    {
        Radius += (lineThickness/2);
        uint16_t Radius2 = Radius - lineThickness;

        // Good, may not be the fastest
        // Does not draw overlapping pixels
        BSP_LCD_SetTextColor(DrawProp[ActiveLayer].TextColor);	// Line Draw Color
        BSP_LCD_DrawHLine(Xpos - Radius, Ypos, lineThickness);	// Left Middle
        BSP_LCD_DrawHLine(Xpos + Radius2, Ypos, lineThickness);	// Right Middle
        BSP_LCD_DrawVLine(Xpos, Ypos - Radius, lineThickness);	// Top Middle
        BSP_LCD_DrawVLine(Xpos, Ypos + Radius2, lineThickness);	// Bottom middle
        uint32_t TextColor = BSP_LCD_GetTextColor();			// Pixel Draw Color
        uint32_t RR = Radius * Radius;
        uint32_t R2R2 = Radius2 * Radius2;
        for(int16_t y = -Radius; y < 0; y++) {
            for(int16_t x= -Radius; x < 0; x++) {
                uint32_t r2 = x * x + y * y;
                if ((r2 <= RR) && (r2 >= R2R2)) {
                    BSP_LCD_DrawPixel(Xpos+x, Ypos+y, TextColor);
                    BSP_LCD_DrawPixel(Xpos+x, Ypos-y, TextColor);
                    BSP_LCD_DrawPixel(Xpos-x, Ypos+y, TextColor);
                    BSP_LCD_DrawPixel(Xpos-x, Ypos-y, TextColor);
                }
            }
        }
    }

kisvegabor, 2018-05-23:
Hi Jack,

Thank you for sharing and testing. However  **Thick circle** drawing is already implemented in LittlevGL 
 I examined the algorithm and found that current method is faster then STM's algorithm because LittlevGL:
- calculates only a 1/8 circle and draws the other parts by mirroring which is quite fast
- draws lines instead of single pixels which also faster
- applies anti-aliasing

I was thinking a lot about **Tick line**. I checked your suggested algorithm an I found that improving the current line drawing mechanism would result a faster drawing and faster implementation because the core (not perpendicular line ending) drawing is very fast and already anti-aliased. So the picture above shws the current state.

The **Arc drawing** is still questionable. As this is not implemented in LittelvGL at all, I think it would be better to use an existing algorithm instead of develop one from the beginning. However I didn't have time to try arc drawing in the practice. :(   
 
> This is 7 times faster than using the thick arc function, because of the math in there.

Do you mean this circle drawing is 7 times faster then the Arc drawing?

bkht, 2018-05-23:
Hi Gabor, you got me!
I verified making sure the arc is actually closed like the circle, same diameter 100, and line thickness 30, then I measured:
Arc: 0.025987 Sec
Circle: 0.006373 Sec
So the difference then is 4.078 (I think I forgot to make the arc a full circle before).
Anyway, it is quite hard to develop (or find) very fast algorithms that allow anti-aliasing and opacity. :-) 
I have tested quite a lot of algorithms the last days, also faster math at the cost of precision, but it is not very impressive, yet. I also got stuck on a thick circle consisting of just fast horizontal lines, but I will keep that part in mind, as it only doesn't render a small part correctly. Maybe I still find a solution for it later. Sometimes I need to sleep a night to get new ideas.
Anyway, very interesting stuff.
Kind regards,
Jack.

kisvegabor, 2018-05-24:
Hi,

What a huge difference in performance!
 
Thank you for the testing and searching! 
Once I finished the line drawing I will also think about the Arc drawing algorithm. We will surly find the right way to implement it!

bkht, 2018-05-27:
Hi Gabor,
above the fastest possible thick circle, without duplicated pixels.
Below the fastest possible thick arc, without duplicated pixels, but this one is of course a lot slower than the circle, because of the calculations involved (though optimized).

    void BSP_LCD_DrawThickArc(int Xpos, int Ypos, int Radius, int startAngle, int endAngle, int lineThickness)
    {
        Radius += (lineThickness/2);
        uint16_t Radius2 = Radius - lineThickness;
        int16_t deg;
        BSP_LCD_SetTextColor(DrawProp[ActiveLayer].TextColor);		// Line Draw Color
        uint32_t TextColor = BSP_LCD_GetTextColor();				// Pixel Draw Color

        // Good, may not be the fastest
        // Does not draw overlapping pixels
        deg = BSP_LCD_FastAtan2(-Radius, 0);
        if (deg < 0) deg+= 360;
        if (deg > 360) deg-= 360;
        if ((deg >= startAngle) && (deg <= endAngle))
            BSP_LCD_DrawHLine(Xpos - Radius + 1, Ypos, lineThickness);	// Left Middle
        deg = BSP_LCD_FastAtan2(Radius2, 0);
        if (deg < 0) deg+= 360;
        if (deg > 360) deg-= 360;
        if ((deg >= startAngle) && (deg <= endAngle))
            BSP_LCD_DrawHLine(Xpos + Radius2, Ypos, lineThickness);		// Right Middle
        deg = BSP_LCD_FastAtan2(0, -Radius);
        if (deg < 0) deg+= 360;
        if (deg > 360) deg-= 360;
        if ((deg >= startAngle) && (deg <= endAngle))
            BSP_LCD_DrawVLine(Xpos, Ypos - Radius + 1, lineThickness);	// Top Middle
        deg = BSP_LCD_FastAtan2(0, Radius2);
        if (deg < 0) deg+= 360;
        if (deg > 360) deg-= 360;
        if ((deg >= startAngle) && (deg <= endAngle))
            BSP_LCD_DrawVLine(Xpos, Ypos + Radius2, lineThickness);		// Bottom middle
        uint32_t RR = Radius * Radius;
        uint32_t R2R2 = Radius2 * Radius2;
        for(int16_t y = -Radius; y < 0; y++) {
            for(int16_t x= -Radius; x < 0; x++) {
                uint32_t r2 = x * x + y * y;
                if ((r2 <= RR) && (r2 >= R2R2)) {
                    deg = BSP_LCD_FastAtan2(x, y);
                    if (deg < 0) deg+= 360;
                    if (deg > 360) deg-= 360;
                    if ((deg >= startAngle) && (deg <= endAngle))
                        BSP_LCD_DrawPixel(Xpos+x, Ypos+y, TextColor);
                    deg = BSP_LCD_FastAtan2(x, -y);
                    if (deg < 0) deg+= 360;
                    if (deg > 360) deg-= 360;
                    if ((deg >= startAngle) && (deg <= endAngle))
                        BSP_LCD_DrawPixel(Xpos+x, Ypos-y, TextColor);
                    deg = BSP_LCD_FastAtan2(-x, y);
                    if (deg < 0) deg+= 360;
                    if (deg > 360) deg-= 360;
                    if ((deg >= startAngle) && (deg <= endAngle))
                        BSP_LCD_DrawPixel(Xpos-x, Ypos+y, TextColor);
                    deg = BSP_LCD_FastAtan2(-x, -y);
                    if (deg < 0) deg+= 360;
                    if (deg > 360) deg-= 360;
                    if ((deg >= startAngle) && (deg <= endAngle))
                        BSP_LCD_DrawPixel(Xpos-x, Ypos-y, TextColor);
                }
            }
        }
    }

    uint16_t BSP_LCD_FastAtan2(int x, int y)
    {
        // Fast XY vector to integer degree algorithm - Jan 2011 www.RomanBlack.com
        // Converts any XY values including 0 to a degree value that should be
        // within +/- 1 degree of the accurate value without needing
[... 62 lines cut ...]
        if(negflag & 0x10) degree = (90 - degree);

        // 3. Degree is now 0-90 range for this quadrant,
        // need to invert it for whichever quadrant it was in
        if(negflag & 0x02)   // if -Y
        {
        if(negflag & 0x01)   // if -Y -X
                degree = (180 + degree);
        else        // else is -Y +X
                degree = (180 - degree);
        }
        else    // else is +Y
        {
        if(negflag & 0x01)   // if +Y -X
                degree = (360 - degree);
        }
        return degree;
    }

The tested arc has a diameter of 100, and line thickness of 30.
Full 360 degrees arc takes 0.013448 Sec;
Half 180 degress arc takes 0.011895 Sec;

Kind regards,
Jack.

kisvegabor, 2018-05-29:
Hi Jack,

Thank you very much for the updates!

**Circle**
Sound great, but unfortunately I don't kno how fast it is compared to LittelvGL's circle drawing. If you have some time can you compare them?
In LittelvGL the circle drawing happen in `lv_draw_rect` (lv_draw.c) because the circle is drawn as a rectangle with a big radius. The drawing algorithm uses `px_fp` and `fill_fp` functions.

**Arc**
I will definitely try this algorithm! As it uses only horizontal/vertical line and pixel drawing it will be easy to add to LittlevGL. 

**Line**
I tried two different approach to to draw lines with perpendicular ending but I wasn't happy with them. Still I have a third idea to which I have much hope! I will test it and get back with the result. 
It seems you have good environment for speed testing. Can I send you an easy-to-port version of the new line drawing algorithm to compare it with others?

bkht, 2018-06-01:
Hi Gabor,
I was just diving into it, but have some difficulty to isolate or extract the circle drawing from lv_draw_rect.
Can you please help me a bit by explaining it a bit more?
Or could I just draw a circle from the main function, by calling lv_draw_rect?
Regards,
Jack.

kisvegabor, 2018-06-02:
Hi,

Wow, that arc look awesome! My third line drawing concept seems working. I just need some tests and minor fixes on it. After that I will dive into the your arc drawing algorithm. :)

Yes, the most simple way is to use `lv_draw_rect` from the main function directly.

By the way, if you can compare the circle drawing algorithm of LittlevGL with others would you compare the line drawing algorithms as well? It will use the same `px_fp` and `fill_fp`.

kisvegabor, 2018-06-02:
My line drawing algorithm seems working! It draws perpendicular ending and can apply anti-aliasing too. There is not division or trigonometrical functions in it at all. So I hope it's fast. 

[image]

Here I drawn circles on the junction points. It seems working precisely.
[image]

bkht, 2018-06-02:
Hi Gabor,
I'm trying to draw at least something with lv_draw_rect
The slider, button, lmeter, gauge, switch, chart, text boxes, all work fine.
When after that I execute below code,  but nothing appears.
I'm quite clueless to get a circle, or something with it, I'm sorry, I'm still a bit new with littlevgl, I guess.

    static lv_area_t cords;
    cords.x1 = 10;
    cords.y1 = 10;
    cords.x2 = 400;
    cords.y2 = 240;
    static lv_area_t mask;
    mask.x1 = 10;
    mask.y1 = 10;
    mask.x2 = 400;
    mask.y2 = 240;
    static lv_style_t style_rect;
    style_rect.body.border.color = LV_COLOR_WHITE;
    style_rect.body.grad_color =  LV_COLOR_GREEN;
    style_rect.body.main_color =  LV_COLOR_LIME;
    style_rect.body.opa = LV_OPA_70;
    style_rect.body.padding.inner = 8;
    style_rect.body.padding.hor = 5;
    style_rect.body.padding.ver = 5;
    style_rect.body.radius = LV_RADIUS_CIRCLE;
    style_rect.body.shadow.type = LV_SHADOW_BOTTOM;
    style_rect.body.shadow.width = 10;
    style_rect.body.shadow.color = LV_COLOR_LIME;
    style_rect.line.color = LV_COLOR_SILVER;
    style_rect.line.width = 3;
    style_rect.line.color = LV_COLOR_RED;
    style_rect.text.color = LV_COLOR_HEX3(0x333);
    lv_draw_rect(&cords, &mask, &style_rect);

kisvegabor, 2018-06-04:
Hi,

The draw/refresh system of LittlevGL is quite complex inside. By simply
calling lv_draw_rect won't draw anything visible because it draws into an
internal buffer called VDB (virtual display buffer). However, the drawing
still happens, but the result stays in the internal buffer.

To test it similarly to your algorithms you should change the value of
px_fp and fill_fp function pointers. They have a "mask" parameter, but if
you draw directly to  display you don't need to deal with it. These
function are in lv_draw.c.

<https://github.com/littlevgl/lvgl/issues/252#issuecomment-394098526>, or mute
the thread

bkht, 2018-06-04:
Hi Gabor,
I was thinking it only draws in the background, not visible...
For disp_fill, I use below code. For px_fp, I can't seem to find information.
I'm afraid, I can't test it, as I can't verify, what I'm drawing.

    lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
    lv_disp_drv_init(&disp_drv);                    /*Basic initialization*/
    /*Set up the functions to access to your display*/
    disp_drv.disp_fill = tft_fill;
    disp_drv.disp_map = tft_map;
    disp_drv.disp_flush = tft_flush;            /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/
    /*Finally register the driver*/
    lv_disp_drv_register(&disp_drv);                /*Register the driver in LittlevGL*/

    /**
    * Fill a rectangular area with a color
    * @param x1 left coordinate of the rectangle
    * @param x2 right coordinate of the rectangle
    * @param y1 top coordinate of the rectangle
    * @param y2 bottom coordinate of the rectangle
    * @param color fill color
    */
    static void tft_fill(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t color)
    {
        /*Return if the area is out the screen*/
        if(x2 < 0) return;
        if(y2 < 0) return;
        if(x1 > TFT_HOR_RES - 1) return;
        if(y1 > TFT_VER_RES - 1) return;

        /*Truncate the area to the screen*/
        int32_t act_x1 = x1 < 0 ? 0 : x1;
        int32_t act_y1 = y1 < 0 ? 0 : y1;
        int32_t act_x2 = x2 > TFT_HOR_RES - 1 ? TFT_HOR_RES - 1 : x2;
        int32_t act_y2 = y2 > TFT_VER_RES - 1 ? TFT_VER_RES - 1 : y2;

        uint32_t x;
        uint32_t y;

        /*Fill the remaining area*/
        for(x = act_x1; x <= act_x2; x++) {
            for(y = act_y1; y <= act_y2; y++) {
                my_fb[y * TFT_HOR_RES + x] = color.full;
            }
        }
    }

kisvegabor, 2018-06-04:
Hi,

Using the `fill` function is also possible but I'm afraid it won't give intended result in performance.

Unfortunately wee need some hack to properly test the performance. 
Please have look at this part: https://github.com/littlevgl/lvgl/blob/master/lv_draw/lv_draw.c#L75

Here the library sets  fill (`fill_fp`) and put px (`px_fp`) to use in the drawing functions. 
As you have `BSP_LCD_DrawVLine` I suppose you have a similar function to draw a rectangle and put a pixel to the display too. To make them compatible with LittlevGL a simple wrapper will be required. For example:
```c
    void fill_wrapper(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa)
    {
    	/*`opa` and `mask` are don't care*/

    	BSP_LCD_DrawRect(coords->x1, coords->y1, lv_area_get_width(coords), lv_area_get_height(coords), color.full);
    }
```
And finally the `fill_fp` function pointer variable in lv_draw.c need to be updated to the wrapper:
```c 
void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = fill_wrapper;
```

I know it's a little bit complicated. Sorry for the inconvenience.  Just a little explanation: 
The intended usage of LittelvGL is not just draw things but create objects instead. This way the library can handle the drawing inside and do a lot of optimization on it. That's why LittlevGL hasn't got a drawing interface.

kisvegabor, 2018-06-06:
Hi Jack,

I tried your arc drawing algorithm in LittlevGL and it worked!

I applied opacity on it to test the pixels are really not drawn twice. And it worked as well!
[image]

In v5.2 I will create a simple arc object using this algorithm.

Only the anti aliasing is missing but I will learn this code in more detail to see how to add it!

So THANK YOU very much for searching and testing this algorithms. It was exteremly useful!

bkht, 2018-06-06:
Hi gabor,

now I have this:

    void fill_wrapper(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa)
    {
        /*`opa` and `mask` are don't care*/
        lcd.SetTextColor((uint32_t) color.full);
        lcd.DrawRect((uint16_t) coords->x1, (uint16_t) coords->y1,
                (uint16_t) lv_area_get_width(coords), (uint16_t) lv_area_get_height(coords));
    }

    // Rectangle or Circle or what?
    static lv_area_t cords;
    cords.x1 = 110;
    cords.y1 = 10;
    cords.x2 = 500;
    cords.y2 = 240;
    static lv_area_t mask;
    mask.x1 = 10;
    mask.y1 = 10;
    mask.x2 = 400;
    mask.y2 = 240;
    static lv_style_t style_rect;
    lv_style_copy(&style_rect, &lv_style_pretty);
    style_rect.body.main_color =  LV_COLOR_LIME;
    style_rect.body.grad_color =  LV_COLOR_GREEN;
    style_rect.body.radius = LV_RADIUS_CIRCLE;
    style_rect.body.opa = LV_OPA_70;
    style_rect.body.border.color = LV_COLOR_WHITE;
    style_rect.body.border.width = 3;
    style_rect.body.border.part = LV_BORDER_RIGHT;
    style_rect.body.border.opa = LV_OPA_70;
    style_rect.body.shadow.color = LV_COLOR_LIME;
    style_rect.body.shadow.type = LV_SHADOW_BOTTOM;
    style_rect.body.shadow.width = 10;
    style_rect.body.padding.inner = 8;
    style_rect.body.padding.hor = 5;
    style_rect.body.padding.ver = 5;
    style_rect.text.color = LV_COLOR_HEX3(0x333);
    style_rect.text.font = &lv_font_dejavu_30;
    style_rect.text.letter_space = 1;
    style_rect.text.line_space = 1;
    style_rect.text.opa = LV_OPA_50;
    style_rect.image.color = LV_COLOR_HEX(0x202020);
    style_rect.image.intense = LV_OPA_50;
    style_rect.image.opa = LV_OPA_50;
    style_rect.line.color = LV_COLOR_RED;
    style_rect.line.width = 3;
    style_rect.line.opa = LV_OPA_50;
    lv_draw_rect(&cords, &mask, &style_rect);

    void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = fill_wrapper;

But what gets drawn in the buffer is still invisible, to my eyes. :-)

kisvegabor, 2018-06-07:
Hi,

I used the FastAtan2 function. It seems really well optimized. 
I found a few points where the Arc drawing can be optimized. I attached the modified algorithm. Can you compare it with original if you have time? I totally understand you because I also change my job right now :)
[lv_arc.zip](https://github.com/littlevgl/lvgl/files/2079031/lv_arc.zip)
BTW the GUI in the video looks great!

> void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = fill_wrapper;

This part should be in `lv_draw.c` to modify the `fill_fp` variable there.

For rectangle drawing a the `px_fp` function should be wrapped too.

And because it's a special test case you shouldn't call `lv_task_handler` because it will redraw the  screen and overwrite the drawings outside of the library. So try something like this:

```c

//Init the style, area, mask, etc
...

//Draw the rectangle
lv_draw_rect(&cords, &mask, &style_rect);

while(1);     //Stop here, don't let LittlevGL to run
```

Take your time, your professional job(s) are more important now. So don't fell any pressure. Play with it if you really have free time.

bkht, 2018-06-07:
Hi Gabor,

thank you very much!
To make the arc the same size, I added this line at the very start.

    Radious += (lineThickness/2);

Comparing speed:
Arc: 0.012036
Arc2: 0.005585
The last one is the time of your code, it is more than twice as fast!
The radius = 100, line thickness 30. start angle 45, end angle 315.

Note there is a black line at the center, it might be a small thing to address:
[image]
[image]

Same code, but small adjustments, regarding mask, color and opa, not available in in de lcd code where I tested it.

    void BSP_LCD_DrawThickArc2(int Xpos, int Ypos, int Radius, int startAngle, int endAngle, int lineThickness)
    {
        Radius += (lineThickness/2);
        uint16_t r_out = Radius;
        uint16_t r_in = r_out - lineThickness;

        int16_t deg_base;
        int16_t deg;
        BSP_LCD_SetTextColor(DrawProp[ActiveLayer].TextColor);		// Line Draw Color
        uint32_t TextColor = BSP_LCD_GetTextColor();				// Pixel Draw Color

        int16_t x_start[4];
        int16_t x_end[4];
        deg = 270;//BSP_LCD_FastAtan2(-r_out, 0);
        if ((270 >= startAngle) && (270 <= endAngle)) BSP_LCD_DrawHLine(Xpos - r_out + 1, Ypos, lineThickness - 1);	// Left Middle
        if ((90 >= startAngle) && (90 <= endAngle)) BSP_LCD_DrawHLine(Xpos + r_in, Ypos,  lineThickness - 1);		// Right Middle
        if ((180 >= startAngle) && (180 <= endAngle)) BSP_LCD_DrawVLine(Xpos, Ypos - r_out + 1,  lineThickness - 1);	// Top Middle
        if ((0 >= startAngle) && (0 <= endAngle)) BSP_LCD_DrawVLine(Xpos, Ypos + r_in,  lineThickness - 1);		// Bottom middle

        uint32_t r_out_sqr = r_out * r_out;
        uint32_t r_in_sqr = r_in * r_in;
        int16_t x;
        int16_t y;

        for(y = -r_out; y < 0; y++) {
            x_start[0] = LV_COORD_MIN;
            x_start[1] = LV_COORD_MIN;
            x_start[2] = LV_COORD_MIN;
            x_start[3] = LV_COORD_MIN;
            x_end[0] = LV_COORD_MIN;
            x_end[1] = LV_COORD_MIN;
            x_end[2] = LV_COORD_MIN;
            x_end[3] = LV_COORD_MIN;
            for(x= -r_out; x < 0; x++) {

                uint32_t r_act_sqr = x * x + y * y;
                if(r_act_sqr > r_out_sqr) continue;

                deg_base =  BSP_LCD_FastAtan2(x, y) - 180;

                deg = 180 + deg_base;
                if ((deg >= startAngle) && (deg <= endAngle)) {
                    if(x_start[0] == LV_COORD_MIN) x_start[0] = x;
                } else if(x_start[0] != LV_COORD_MIN && x_end[0] == LV_COORD_MIN) x_end[0] = x - 1;

                deg = 360 - deg_base; //BSP_LCD_FastAtan2(x, -y);
                if ((deg >= startAngle) && (deg <= endAngle)) {
                    if(x_start[1] == LV_COORD_MIN) x_start[1] = x;
                } else if(x_start[1] != LV_COORD_MIN && x_end[1] == LV_COORD_MIN) x_end[1] = x - 1;

                deg = 180 - deg_base; //BSP_LCD_FastAtan2(-x, y);
                if ((deg >= startAngle) && (deg <= endAngle)) {
                    if(x_start[2] == LV_COORD_MIN) x_start[2] = x;
                } else if(x_start[2] != LV_COORD_MIN && x_end[2] == LV_COORD_MIN) x_end[2] = x - 1;

                deg = deg_base; //BSP_LCD_FastAtan2(-x, -y);
                if ((deg >= startAngle) && (deg <= endAngle)) {
                    if(x_start[3] == LV_COORD_MIN) x_start[3] = x;
                } else if(x_start[3] != LV_COORD_MIN && x_end[3] == LV_COORD_MIN) x_end[3] = x - 1;

                if(r_act_sqr < r_in_sqr) break;	/*No need to continue the iteration in x once we found the inner edge of the arc*/
            }

            if(x_start[0] != LV_COORD_MIN) {
                if(x_end[0] == LV_COORD_MIN) x_end[0] = x - 1;
                BSP_LCD_DrawHLine(Xpos+x_start[0], Ypos+y, x_end[0] - x_start[0]);
            }

            if(x_start[1] != LV_COORD_MIN) {
                if(x_end[1] == LV_COORD_MIN) x_end[1] = x - 1;
                BSP_LCD_DrawHLine(Xpos+x_start[1], Ypos-y, x_end[1] - x_start[1]);
            }

            if(x_start[2] != LV_COORD_MIN) {
                if(x_end[2] == LV_COORD_MIN) x_end[2] = x - 1;
                BSP_LCD_DrawHLine(Xpos-x_end[2], Ypos+y, LV_MATH_ABS(x_end[2] - x_start[2]));
            }

            if(x_start[3] != LV_COORD_MIN) {
                if(x_end[3] == LV_COORD_MIN) x_end[3] = x - 1;
                BSP_LCD_DrawHLine(Xpos-x_end[3], Ypos-y, LV_MATH_ABS(x_end[3] - x_start[3]));
            }

        }

    }

kisvegabor, 2018-06-07:
Wao, great! Thank you for testing it. :)
I will check the missing line in the center!

bkht, 2018-06-07:
Hi Gabor,
It actually is drawn one pixel too far to the right for some reason. When I change this
if ((180 >= startAngle) && (180 <= endAngle)) BSP_LCD_DrawVLine(Xpos - 1, Ypos - r_out + 1, lineThickness - 1);	// Top Middle
It is filling the gap. But it should be in the middle really... :-)
So maybe the arc is drawn 1 pixel to the left? :-)

kisvegabor, 2018-06-07:
I checked it with your settings (radius and angles) and it worked for me. Maybe it's because your `BSP_LCD_DrawHLine` function doesn't include the last pixel or something like that. Anyway with the built-in fill function of LittlevGL it doesn't skip that column. 

I will create a new object soon which uses the arc drawing. Hopefully it will work for you correctly as well.

bkht, 2018-06-07:
Hi Gabor,
yes, must be right here. :-)

kisvegabor, 2018-07-07:
Sorry Jack, somehow I didn't get notification from GitHub about your comment. :(

I created the new objects in `dev-5.2` branch:
1. lv_arc
2. lv_preload (a spinning arc)

Their API quite simple, I think you will see how to use them. Just don't forget to add the `USE_LV_ARC` and `USE_LV_PRELOAD` defines to lv_conf.h to enable the objects.

Please let me know what do you think if you had a chance to try them.

BTW, you can also try the new line drawing algorithm which is already used by `lv_line`.

TheGrovesy, 2018-07-12:
This is all looking very promising. I will try to give your 5.2 beta a play over the weekend (in my Visual Studio Simulation ;) )
The Arc work sounds very promising. I assume this will also become and type option for the Gauge object?

bkht, 2018-07-13:
Hi Gabor,
I'm very glad the arc works well.
I'm sorry, I was in hospital for operation and now recovering, all is going well with me.
The screen of my laptop is broken (vertical curtains of stripes, nothing readable), I now connected a monitor to it. The laptop obviously needs to be repaired soon.
I'm a bit limited on possibilities to work on this these days, but I hope to catch up soon, though. :-)
Kind regards,
Jack.

kisvegabor, 2018-07-14:
As line and arc drawing are implemented in `dev-5.2` I close this topic.

@bkht It was a pleasure to work with you on this improvements! Thank you very much for searching for these algorithms and testing them. :)

If you find bugs or have suggestions related to arc or line drawing please open a new issue.

closed 2018-07-14 by kisvegabor

bkht, 2018-07-15:
Hi Gabor,
very good work!!! Thank you very much!!!
Kind regards, Jack.

jpwright, 2019-01-11:
Gabor--first, thanks for your excellent work on this library, it has been a joy to use and I love to see the continuous progress.
I'm hoping to use arc w/ rounded edges in my application. However, I noticed that things look a bit strange right now due to the fact that the arc itself is not anti-aliased but the rounded edges are:
[image]
Ideally, we'd just have the entire arc be anti-aliased. Is this something you're planning to work on for the next version? If not, I might be able to put some time into it.

embeddedt, 2019-01-11:
@jpwright Arc antialiasing will hopefully be done by June when we release `v6.0`.

jpwright, 2019-01-14:
@embeddedt awesome! are there any plans to have a `dev-6.0` branch ahead of that?

embeddedt, 2019-01-14:
@jpwright The `dev-6.0` branch will be formed shortly after the release of `v5.3` this month. Do you want to start working on this feature now?

jpwright, 2019-01-15:
@embeddedt thanks for letting me know - I will keep an eye out for the `dev-6.0` branch and happy to contribute then if it makes sense.