lvgl/lvgl #253: lv_chart_set_type() - invalid conversion from int to lv_chart_type_t

bkht, 2018-05-14:
Hi,
I found a minor issue, when trying the Chart example, with this line:

    lv_chart_set_type(chart, LV_CHART_TYPE_POINT | LV_CHART_TYPE_LINE);

Error: invalid conversion from 'int' to 'lv_chart_type_t' [-fpermissive]

kisvegabor, 2018-05-15:
Hi,

Thank you!

With my settings it didn't cause warning. 
Introducing an `LV_CHART_TYPE_LINE_AND_POINT` type should solve it. What do you think?

bkht, 2018-05-16:
Hi Gabor,
I think the OR operation isn't the issue here, as the toolchain complains about the conversion from 'int' to 'lv_chart_type_t'.
It seems I just can't pass an int to the function.
I tried casting using (lv_chart_type_t), but same result.
I'm a bit puzzled, about this one... :-)
Just modifying the function so it can pass an int or enum could possibly be a solution, do you think?

kisvegabor, 2018-05-16:
Hi,
It's strange. I googled on this warning and found some C++ related topics. are you using C++?

bkht, 2018-05-16:
Yes, I'm using C++. I will try to figure this out.

kisvegabor, 2018-05-16:
I found this: https://stackoverflow.com/questions/11452920/how-to-cast-int-to-enum-in-c

So maybe it works:
```c
    static_cast<lv_chart_type_t>(LV_CHART_TYPE_LINE | LV_CHART_TYPE_POINT);
```

bkht, 2018-05-17:
Hi Gabor, you are awesome! This works!
I saw something similar on stackoverflow, but didn't test it yet, as I'm quite busy these days.
Because you have put your effort in finding a solution for me, I wanted to thank you and give you giveback :-)
So this worked, I do now have the dots in the graph:

    lv_chart_type_t type = static_cast<lv_chart_type_t>(LV_CHART_TYPE_POINT | LV_CHART_TYPE_LINE);
    lv_chart_set_type(chart, type);   /*Show lines and points too*/

Or just:

    lv_chart_set_type(chart, static_cast<lv_chart_type_t>(LV_CHART_TYPE_POINT | LV_CHART_TYPE_LINE));   /*Show lines and points too*/

I hope, I don't bother you too much, though... :-)

kisvegabor, 2018-05-18:
Awesome!

Bother? The opposite! I'm always happy to see so enthusiastic people like you! Keep this great work! :)

As the issue seems solved I close this issue.

closed 2018-05-18 by kisvegabor