lvgl/lvgl #255: Line meter - ideas

bkht, 2018-05-16:
For the line meter, a nice additional style could be a filled arc instead of (rainbow) lines.
Something like http://i.stack.imgur.com/UfubO.png
And instead of the rainbow it would be nice if we also could optionally just set levels for the Good (green), Warning (yellow), Error (red) regions, so the rainbow will then be replaced by regions after defining them.
Would that be a great idea for the future?
Drawing a filled arc (AA?) might be difficult though. I need to search for those arc algorithms...

bkht, 2018-05-16:
Hi Gabor,
yes, I see, it's probably a bit hard to find or solve... I will search and once I found something, I would let you know. problem also is that we want it be AA as well, right? Or could without AA be fine?

And the other suggestion, about defining regions with a specific color instead of the rainbow would probably not too hard, what do you think?

kisvegabor, 2018-05-16:
It would be better without AA. I can add the AA  once a non-AA version  is working.

Defining regions is a higher level feature which just draws more arcs. It should be easy once the arc drawing works.

Thank you very much in advance!  :)

bkht, 2018-05-16:
Hi Gabor,
Now I can plot an arc with some start and end angle and line thickness fairly fast!
Very straight forward, while fast, it sure plots duplicated pixels, to avoid missing pixels.
I tested this code (but not in littlevgl), together with the simple code that follows.
https://github.com/ArminJo/STMF3-Discovery-Demos/blob/master/lib/graphics/src/thickLine.cpp
The function 'drawThickLineSimple' is not needed, only those two: 'drawThickLine' which uses 'drawLineOverlap'.
Then I draw lines (using drawThickLine) from the inner to the outer arc (gaps will appear if line thickness is less than 3).

    void SOME_CLASS::drawThickArc(int x, int y, int r, int startAngle, int endAngle, int thickness)
    {
        // Very fast, but it has duplicated pixels, so improvement are possible
        int rDelta = -(thickness/2);
        int px1, py1, px2, py2, cx, cy;

        startAngle -= 90;
        endAngle   -= 90;

        if (startAngle!=endAngle)
        {
            for (int d=startAngle+1; d<endAngle+1; d++)
            {
                px1 = x + cos((d*3.14)/180) * (r+rDelta);
                py1 = y + sin((d*3.14)/180) * (r+rDelta);
                px2 = x + cos((d*3.14)/180) * (r+rDelta+thickness);
                py2 = y + sin((d*3.14)/180) * (r+rDelta+thickness);
                DrawThickLine(px1, py1, px2, py2, 3, 0);
            }
        }
        else
        {
            px1 = x + cos((startAngle*3.14)/180) * (r+rDelta);
            py1 = y + sin((startAngle*3.14)/180) * (r+rDelta);
            px2 = x + cos((startAngle*3.14)/180) * (r+rDelta+thickness);
            py2 = y + sin((startAngle*3.14)/180) * (r+rDelta+thickness);
            DrawThickLine(px1, py1, px2, py2, 2, 0);
        }
    }

Hope this helps?

kisvegabor, 2018-05-17:
Hi,

It helps a lot! If it draws some pixels twice we just can't use opacity on this object, I think it's better to have an arc drawer with no opacity support then don't have it at all.

I will check this code in more detail soon!

Because of your fast and great contribution I decided to use this issue as the main thread of arc drawing and I will redirect the similar request here.

bkht, 2018-05-17:
Thanks a lot. I can create a thick circle function that does not draw multiple pixels, That should be straight forward), this arc is probably among the fastest. The line thickness that I mentioned was for and arc diameter of 100, and thickness 30, that I used for testing. So for smaller arcs, we could get away even better.

bkht, 2018-05-17:
I have an idea, on how to plot the pixels once, maintaining a little array (size depending on arc diameter and line thickness), and keeping track of the pixels already plotted, but at the cost of a little bit performance and some memory footprint. I will test this approach, soon, if I find some time.
It's just an idea to test with opacity for me.

kisvegabor, 2018-05-18:
Sound great! I also have an algorithm in mind :)

kisvegabor, 2018-07-07:
As the discussion continues in #146 I close this issue.

closed 2018-07-07 by kisvegabor