lvgl/lvgl #256: lvgl slow start-up

Zaltora, 2018-05-16:
Hi, here is my situation:
When I turn on my product, the screen turns on instantly but the first lvgl interface takes about 1.5 seconds to start.
The problem is not Hardware and the driver works well.

In fact, I found a way for lvgl to start at the same time as the screen. I'm cheating on his time system adding 2 seconds at boot start setup with:
`lv_tick_inc(2000);`

someone noticed the same problem when turning on his screen?

Zaltora, 2018-05-16:
It is work with a lower value too:
` lv_tick_inc(200);`. 
Increment time, even a little, seems to avoid this long start-up.
Time Setup:
- `#define LV_REFR_PERIOD      200 ` (same problem with a value of 100)
- Timer Normal tick : 50 ms

kisvegabor, 2018-05-17:
Hi,

please update to the `beta` branch. I fixed something related to this.

Zaltora, 2018-05-17:
Update all branch ( tested beta and  dev-5.2). Problem still here.
What commit fixed that ?

kisvegabor, 2018-05-17:
I see. I'm just checking the commits but can't find it... 
I remember that I faced the same on ESP32 and found the origin of it as well but don't remember what was the solution... :(

Do you call `lv_tick_inc` in the tick hook of FreeRTOS? If yes then you should pass the tick period in milliseconds. E.g. `lv_tick_inc(portTICK_RATE_MS)`

Does it change anything?

kisvegabor, 2018-05-17:
I also suspect an overflow or something like this.

To debug the issue can you modify `lv_task_exec` function in lv_task.c like this:
```c
static bool lv_task_exec (lv_task_t* lv_task_p)
{
    bool exec = false;

    /*Execute if at least 'period' time elapsed*/
    uint32_t elp = lv_tick_elaps(lv_task_p->last_run);
    printf("elaps: %d\n", elp);                         /*THIS LINE IS ADDED*/

    if(elp >= lv_task_p->period) {
        lv_task_p->last_run = lv_tick_get();
        lv_task_p->task(lv_task_p->param);

        /*Delete if it was a one shot lv_task*/
        if(lv_task_p->once != 0) lv_task_del(lv_task_p);

        exec = true;
    }

    return exec;
}
```

Zaltora, 2018-05-17:
interesting result:
```
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 0
elaps: 1600
elaps: 1600
elaps: 200
elaps: 200
elaps: 200
elaps: 200
elaps: 200
elaps: 200
...
```

kisvegabor, 2018-05-17:
I suppose you have `lv_tick_inc(200)`. That's why lvgl thinks always 200 ms is elapsed

But why the `1600` ? Can you test it again with `lv_tick_inc(10)` (and really call it in 10 ms) ?

Zaltora, 2018-05-17:
yes, my timer is set for a period of "200"
`#define LV_REFR_PERIOD 200`
`#define coreDISPLAY_TICK (LV_REFR_PERIOD)  //in ms`

Test with a refresh at 10 ms:
[debug.txt](https://github.com/littlevgl/lvgl/files/2013188/debug.txt)
With a lower tick count (10 ms). It is work (or i don't see it). The problem seem to be with larger period count (or more evident with a larger value)

I can explain the `1600`. My application got a lot of task, and at start, a lot of device need to be initialized (sensors, wifi, ...). The screen task got a lower priority and need wait at this moment.
But the problem is not from the app. I already check by adding some driver command into the task. The screen task and  ` lv_task_handler();` loop are call lot of times and didn't affect the screen at start-up.

10ms counting seem work.

kisvegabor, 2018-05-17:
Great! :)

I will add this fix. It will be the part of v5.1.1 which will be released on the weekend.

closed 2018-05-17 by kisvegabor

Zaltora, 2018-05-17:
Add it got an advantage. I can keep my incrementation of time has a high value (200ms).
With your patch, it's instantaneous.
With my patch on timer, i need wait 200ms (the incrementation time )

kisvegabor, 2018-05-18:
Yes, indeed. I don't why didn't add this line so far. :)