lvgl/lvgl #258: Non-linear gradient

beibean, 2018-05-18:
Hi Gabor,

What a great job you and the community have done with this graphical library. I just have to congratulate you for the effort.

I wanted to bring to light a new possible improvement for future releases. As I have been able to check, when drawing gradient within objects (buttons for example), it is drawn as a linear gradient, which means that from one horizontal line to next one there is a fully color jump. This method may cause not a good effect when drawing it on a screen.

What about using a scattered gradient drawing method? This way the color jump would be smoothed so, final result will be nicer for our eyes.

See, I have quickly tested an algorithm option which is limitted to grayScale gradient and other issues. Here it is, so that you can imagine what I mean with all this:

[image]

One big question here is pixel calculating time, which is a way bigger than current gradient algorithm.

Let me know what you think about it.

Greetings ;)

kisvegabor, 2018-05-18:
Hi,

Thank you for the compliment! :)

That a very good idea! It could solve a lot of issues with the 16 bit color displays.
It seems to me a special kind of dithering, right?

As you mentioned it has some overheads, but we can make this optional with a  the style parameter (e.g. `style.body.gradient_smooth = 0 or 1`). This way user can choose between quality and speed.

So I think it would be a great update! :)

beibean, 2018-05-18:
What an accurate diagnosis!

I have implemented it for 16 bit color display and use a style parameter as you suggested :)

By monday I will let you know the algorithm I have used so we can think in the better way to make it suitable for all littlevGL requirements.

kisvegabor, 2018-05-18:
Awesome, thank you very much! :)

kisvegabor, 2018-05-22:
Thank you very much!

It's good idea to use predefined pattern. However even much larger patterns can be useful (e.g 32 lines)  to handle screen sized gradients.

So I will adapt it to LittelvGL! I'm sure a lot of people ill like it!

kisvegabor, 2018-12-18:
Sorry for being inactive for so long in this topic. I started to implement it but I just realized that a new of `lv_imgbtn` can be used for this purpose and that is even more versatile. 

See: https://github.com/littlevgl/lvgl/pull/569

If you add dithering to the parts of the image you will get a smooth result on lower color depth too.

 Anyway, thank you for this great idea!

closed 2018-12-18 by kisvegabor

beibean, 2018-12-19:
Sounds good!

I am currently out of this library, when I have time I will check `lv_imgbtn`.

Thank you!!