lvgl/lvgl #261: [Suggestion] Simplify encoder support

kl-cruz, 2018-05-22:
@kisvegabor I have idea to simplify working with encoder. Idea is to make possible working with lvgl and only rotarty encoder with one switch.
What does it mean?
For now keypad with groups is the best choice to work with encoder. Fortunately keypad has lot of keys (UP, DOWN, LEFT, RIGHT etc.). Unfortunately ofc for rotary encoders with embedded switch inside.

The idea to work with encoder is to make objects 'enterable':

pass every signal to object:
https://github.com/kl-cruz/lvgl/commit/5aafb4f36611ab37c17c1392b5eef2c8a2389cc2#diff-938b5672d7947f1085e7d98fb60a7c37L365

expand lv_res_t with additional values:
https://github.com/kl-cruz/lvgl/commit/5aafb4f36611ab37c17c1392b5eef2c8a2389cc2#diff-9c7730d0c276956f51a8a07e3a90130aR75

add handle for signal - stay focused on object:
https://github.com/kl-cruz/lvgl/commit/5aafb4f36611ab37c17c1392b5eef2c8a2389cc2#diff-afacddff4769d2799e684742d5ba06f6R233

and signal of the 'enterable' object:
```
...
} else if(c == LV_GROUP_KEY_ENTER) {
            if (!ext->is_entered)
            {
                ext->is_entered = true;
                res = LV_RES_ENT;
            } else
            {
                ext->is_entered = false;
                res = LV_RES_REL;
            }
        } else if(c == LV_GROUP_KEY_ESC) {
            ext->is_entered = false;
            res = LV_RES_REL;
        }
...
``` 
and of course focus object with another color for example green :)
https://github.com/kl-cruz/lvgl/commit/5aafb4f36611ab37c17c1392b5eef2c8a2389cc2#diff-afacddff4769d2799e684742d5ba06f6R278

Buttons do not need it, but for example it simplifies working with lists or sliders.
Group can react properly to play with object with proper signals. With these changes it is possible to enter to the edit mode of the object and exit from it.

My encoder has implemented only three group values:
```
LV_GROUP_KEY_PREV
LV_GROUP_KEY_NEXT
LV_GROUP_KEY_ENTER
```

Let me know what do you think about this idea!

kl-cruz, 2018-05-27:
Hi, 
Thanks for reply!
Looks good imo. It is another way to reach it, by do not expand control API but with expanding object API. Like it!
One thing I see here. Lets concentrate on roller. It should be 'enterable' with long press enter action, because normal click sends signal that something was clicked. You would like to enter to roller and choose proper member with short click. list object seems to be 'affected' too.
It could mean that input device implementation should cover long press action, but it is done in another place, in another way.
That's the only thing I see to discuss about

kisvegabor, 2018-05-29:
Hi,

I'm afraid I don't fully understand. Let's imagine a concrete case with two objects: a button and roller. We use a `bool entered` variable to store the current state.
1. Button is in focus and `entered == false`
2. Encoder is turned right -> Because `entered == false` send `LV_GROUP_KEY_NEXT` -> Roller is focused
3. Encoder is pressed -> `entered = true` and `style_mod` function update to entered style-> Send nothing
4. Encoder is turned right -> Because `entered == true` send `LV_GROUP_KEY_RIGHT` -> Roller selects the next item
5. Encoder is pressed -> Because `entered == true` send `LV_GROUP_KEY_ENTER` -> The roller's action callback will be called -> set `entered = false` and update the `style_mod` function to not entered style 
6. Encoder is turned right -> Because `entered == false` send `LV_GROUP_KEY_NEXT` -> button is in focus

However I'm not tested it should work. Or am I misunderstood something?

kisvegabor, 2018-05-30:
Hi,

You are right, I didn't consider these kind of cases. It's more problematic with a keyboard where it's typical to press multiple buttons. Obviously it's not usable to switch to Edit mode, press a  button, leave Edit mode and enter to Edit mode again to press next letter.

Now long press's only purpose is to call lv_btn's long press action. However to use it for changing between "edit" and "focus" mode is a more useful. In this case the whole "mode change" can happen internally and something similar to your implementation need to be added as  built-in feature.

kl-cruz, 2018-06-02:
Okay, I'll take care about encoder support then! I will test solution with `is_enterable` and move management to lower layer. Thanks for answers!

kisvegabor, 2018-06-04:
I also thank you for the contribution! :)

kisvegabor, 2018-06-19:
Hi,

I implemented my idea about the encoder support in `dev-5.2` branch.
I explained in nutshell here: https://github.com/littlevgl/lvgl/issues/277#issuecomment-398332940

Please let me know what do think about it.

kl-cruz, 2018-06-24:
Wow, looks amazing! I'll check it in this week! From quick review, I like function to manually set edit mechanism.
One thing I would like to add: imo it could be good idea to add one bit to objx about possibility enter to edit state (in lv_obj), because editing calendar, roller or slider is good idea, but for example button or checkbox should not allow to go into edit mode.

kisvegabor, 2018-06-25:
Good idea, I will add this functionality too.

StephanSy, 2018-07-10:
Hi!
Thanks for this new feature!

One comment: At least latest commit on branch dev-5.2 doesn't send LV_GROUP_KEY_ENTER_LONG anymore when this new feature is enabled.

One suggestion: It should be configurable if ENTER or ENTER_LONG key switches between navigation and editing.

BR
Stephan

StephanSy, 2018-07-12:
Hi!
I intend to use ENTER_LONG for some short cuts, yes.
Not everyone has only a rotary encoder, so I think would be a really a good idea to keep backwards compatibility when this feature is not used.

I attach a patch with a small change that should do the job as suggestion.

[lvgl_patch.txt](https://github.com/littlevgl/lvgl/files/2187286/lvgl_patch.txt)

BR
Stephan

kisvegabor, 2018-07-12:
Your proposal is awesome! It's a very good idea to just send a LONG_RESS signal.  
I applied your patch and pushed it!

Can you verify if it works?

StephanSy, 2018-07-18:
Hi! Thanks. Did a short test, works for me.
BR
Stephan

kisvegabor, 2018-07-18:
@StephanSy Great, thanks!

Still the implementation of kl-cruz's idea about "editable" bit is missing but after it, the encoder support will be ready :)

kisvegabor, 2018-07-25:
Hi,

I added the "editable" option. It uses a signal (`LV_SIGNAL_GET_EDIATBLE`) because this attribute is typical to the object type an not to individual objects.

To test:
1. Add `lv_group_enable_edit(g, true);` to  `lv_test_group_1`
2. Call `lv_test_group_1()`
3. Used the arrows to navigate 
4. Press enter on some non-editable  obejcts (e.g. button)
5. Long press enter on an editabe object (e.g. slider)
6. Edit with arrows

Please, let me know how it works for you.

kisvegabor, 2018-07-30:
To see the progress of `dev-5.2` I close this issue. 
If you have any remarks don't hesitate to reopen it! :)

closed 2018-07-30 by kisvegabor