lvgl/lvgl #264: Missing include in lv_theme.h

MiSimon, 2018-05-30:
Hi,

in the _lv_theme.h_ file there is no include for _lv_theme_material.h_, this leads to a compiler error if i try to use the material theme.
Adding the include to the other includes at the end of _lv_theme.h_ solves this problem.

Regards,
Michael

BesitzeRuf, 2018-05-30:
Hi,  lv_theme.h doesn't use anything from lv_theme_material.h. You should analyze more to identify which file requires the lv_theme_material.h...

BesitzeRuf, 2018-05-30:
Aha, sorry... I double checked the lv_theme.h... and all these includes are at the bottom of the file :D So yes, you're right, it should be included to the lv_theme.h and mby move all #include... to the top?

MiSimon, 2018-05-30:
The lv_theme_XXX.h files need the definition of lv_theme_t, so the includes need to stay at the bottom of the file.

BesitzeRuf, 2018-05-30:
There 2 solutions:

1. Reorganize headers.
2. Use forward declaration:
- a) set some name of struct for lv_theme_t type (for ex. lv_theme_s)
- b) in each lv_theme_XXX.h declare somewhere at the top "typedef struct lv_theme_s lv_theme_t; "

<details> 
  <summary>lv_theme.h</summary>

> 
> /**
>  *@file lv_themes.h
>  *
>  */
> 
> #ifndef LV_THEMES_H
> #define LV_THEMES_H
> 
> #ifdef __cplusplus
> extern "C" {
> #endif
> 
> /*********************
>  *    INCLUDES
>  *********************/
> #include "../../lv_conf.h"
> #include "../lv_core/lv_style.h"
> 
> #include "lv_theme_templ.h"
> #include "lv_theme_default.h"
> #include "lv_theme_alien.h"
> #include "lv_theme_night.h"
> #include "lv_theme_zen.h"
> #include "lv_theme_mono.h"
> #include "lv_theme_material.h"
> 
> /*********************
>  *    DEFINES
>  *********************/
> 
> /**********************
>  *    TYPEDEFS
>  **********************/
> 
> typedef struct lv_theme_s{
>     lv_style_t *bg;
>     lv_style_t *panel;
> 
> #if USE_LV_CONT != 0
>     lv_style_t *cont;
> #endif
> #if USE_LV_BTN != 0
>     struct {
>         lv_style_t *rel;
>         lv_style_t *pr;
>         lv_style_t *tgl_rel;
>         lv_style_t *tgl_pr;
>         lv_style_t *ina;
>     }btn;
> #endif
> 
> #if USE_LV_LABEL != 0
>     struct {
>         lv_style_t *prim;
>         lv_style_t *sec;
>         lv_style_t *hint;
>     }label;
> #endif
> 
[... 253 lines cut ...]
>  * @param font pointer to a font (NULL to use the default)
>  * @return pointer to the initialized theme
>  */
> lv_theme_t * lv_theme_material_init(uint16_t hue, lv_font_t *font);
> 
> /**
>  * Get a pointer to the theme
>  * @return pointer to the theme
>  */
> lv_theme_t * lv_theme_get_material(void);
> 
> /**********************
>  *      MACROS
>  **********************/
> 
> #endif
> 
> #ifdef __cplusplus
> } /* extern "C" */
> #endif
> 
> #endif /*LV_THEME_MATERIAL_H*/
> 

</details>

MiSimon, 2018-05-30:
I created a pull request for the missing include.
The reorganization of the header files is not directly related to this issue, so i did not include it in the pr.

kisvegabor, 2018-05-31:
Hi,

I added includes for all themes in `lv_theme.h` to enable the inclusion of all themes at once.

@MiSimon Thank you for highlighting the missing include. I merged your Pull Request.

@BesitzeRuf Thank your for sharing your thoughts! I considered to use of forward declaration. It really would be a more elegant and professional solution. However, as the current organization is quite simple and not requires any additional declaration in the lv_theme_XXX files I would stay with that.

closed 2018-05-31 by kisvegabor