lvgl/lvgl #266: [Feature Request] Right align label

upbeat27, 2018-05-31:
Right now a label can be left or center aligned.
A right align option should exist, I would think.

kisvegabor, 2018-06-01:
Hi,

Yes, it absolutely should exist.
v5.2 will support Right-to-left writing as well. With this I will implement this as well!

kisvegabor, 2018-07-25:
I added "Right aligned text" support to `dev_5.2`

It works like this:
```c
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(label, "Very long line\n"
                             "a\n"
                             "two words");
    lv_label_set_align(label, LV_LABEL_ALIGN_RIGHT);
``` 

[image]

A quick test would be welcome :)

I mentioned Right to left support as well, but release v5.2 quicker I delayed this feature. If somebody needs Right-to-left support, please tell it here: https://github.com/littlevgl/lvgl/issues/46

kisvegabor, 2018-07-30:
To see the progress of `dev-5.2` I close this issue. 
If you have any remarks don't hesitate to reopen it! :)

closed 2018-07-30 by kisvegabor