lvgl/lvgl #267: how to draw a mouse on the screen

parai, 2018-06-04:
Hi kisvegabor:
This is really a great  project, now I am trying to integrated it with my automotive project https://github.com/parai/as which is by default running on arm qemu. Now I have the display and mouse driver ready, but the mouse cursor doesn't show on the window, so I don't know where the mouse is, in fact from the log I can see the X,Y position but it's not friendly.

Could you hint me how to add a mouse cursor to be displayed on top of the windows?

[image]
ON              :pressed @(398,236)
ON              :pressed @(319,288)
ON              :pressed @(261,438)
ON              :pressed @(222,327)
ON              :pressed @(48,358)
ON              :pressed @(21,419)
ON              :pressed @(151,465)
ON              :pressed @(325,461)

kisvegabor, 2018-06-04:
Hi,

You can add a cursor with `void lv_indev_set_cursor(lv_indev_t *indev, lv_obj_t *cur_obj)`.
A cursor can be any object. For test I suggest to create a label with a short text, e.g. "<" but later an lv_img object should be used.

Side note:  
As the default button colors in the library are blueish and if you didn't modified the basic styles then I suspect that  you have an ABGR color coding  instead of ARGB (default in LittelvGL). If this is the case then you can swap the bytes in lv_color.h

parai, 2018-06-05:
Hi kisvegabor:
Thanks for you kelp, it works very well.

Next, I think I need to fix the ARGB issue.

kisvegabor, 2018-06-05:
Great! Your welcome!

misterbinny, 2018-07-29:
the lv_layer_sys() and lv_layer_top() functions don't seem to actually draw any of the objects when they're assigned to this layer. You will not see a mouse cursor unless you edit the lv_indev_set_cursor function in lv_indev.c and change the screen pointer to lv_scr_act() for the cursor object.

From: 

`lv_obj_set_parent(indev->cursor, lv_layer_sys()  );`

To:
`lv_obj_set_parent(indev->cursor, lv_scr_act()  );`

This can be confirmed with a simple test of assigning a drawn object to lv_layer_sys() or lv_layer_top(); 
Perhaps I've seriously misunderstood the API. 

The code for your input device should be something like: 

`
	    lv_indev_drv_init(&indev_drv);         //driver type
	    indev_drv.type = LV_INDEV_TYPE_POINTER;  //pointer type
            indev_drv.read = mouse_read;         //to callback function
	    lv_indev_t * mouse_indev = lv_indev_drv_register(&indev_drv); //register the driver in task queue
	    lv_obj_t * cursor_obj =  lv_img_create(lv_scr_act(), NULL); //create object
	    lv_img_set_src(cursor_obj, SYMBOL_POWER); //set its image
            lv_indev_set_cursor(mouse_indev, cursor_obj); // connect the object to the driver
`

And your callback function
`
bool mouse_read(lv_indev_data_t *data)
{
    data->point.x = posx;
    data->point.y = posy;
    data->state = LV_INDEV_STATE_REL;  //state of your mouse button ..press or release..
    return false;  			
}

`

kisvegabor, 2018-07-30:
Hi,

Your code seems fine. I tested it in `maser` and `dev-5.2` branches too and it worked for me.

I also added this to the test:
```c
 lv_obj_t * label = lv_label_create(lv_layer_sys(), NULL);
 lv_label_set_text(label, "Sys layer");
```

And I got this:
[image]

kisvegabor, 2018-07-30:
It also works here:
```c
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(label, "Sys layer");
    lv_obj_set_parent(label, lv_layer_sys());
```

kisvegabor, 2018-08-01:
Try `lv_obj_set_click(cursor_img, false)` to no block if it's on `lv_scr_act()`

Please test this:
1. In `lv_init` (`lv_obj.c`) you will find the creation of sys_layer and top_layer.
2. Change  `lv_obj_set_style(sys_layer, &lv_style_transp_fit);` to `  
    lv_obj_set_style(sys_layer, &lv_style_pretty);`
3. Now the sys_layer should cover everithing a white-graw graient rectangle.

Do you see the rectangle? If not then to debug what's going on you should check this part: https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_refr.c#L343
It starts the rendering of top and sys layers.

misterbinny, 2018-08-01:
The suggested test did not show a rectangle, but simply hid the cursor.  Perhaps the cursor is behaving the way it is unless VDB is used?

From the "lv_refr.c#L343" It looks like that will require Virtual Display Buffering, I am writing directly to a frame buffer in my implementation. 

I would like to get the VDB working, but I think I will have to dig into the code to understand what is going on; but that is off the topic of this thread.

kisvegabor, 2018-08-02:
I checked it and if you don't use VDB then the layers are not refreshed. It's a bug which should be solved by adding these two lines at the end of `static void lv_refr_area_no_vdb(const lv_area_t * area_p)` (lv_refr.c):
```c
    /*Also refresh top and sys layer unconditionally*/
    lv_refr_obj_and_children(lv_layer_top(), area_p);
    lv_refr_obj_and_children(lv_layer_sys(), area_p);
```

kisvegabor, 2018-09-04:
I added the suggested update in `dev-5.2`. If it does not solve your problem please reopen the issue.

closed 2018-09-04 by kisvegabor