lvgl/lvgl #268: Use Mutex

Masi2017, 2018-06-04:
Hi
I created a linked list for messages information, I'd like to show text from a static array in the created labels for the buttons . I have a receive task which receive messages from server, receive task may delete the oldest message from the liked list if max message number exceed , so I have to add a mutex since both tasks are using linked list at the same time and they might try to change that but I don't know how this can be added in littlevgl since as soon as keypad is pressed , prev or next object will be shown automatically and cursor will move to that button ( I am not sure even if object is not in the visible area this causes a problem or not?) , any idea bout this? can I handle this situation?

MiSimon, 2018-06-05:
Hi,
can you move the list operations from the receive task into its own lvgl task (create it with `lv_task_create`)?
Since the input handling is also a lvgl task, there should be no need for a mutex, since the two task will not be called in parallel.

kisvegabor, 2018-06-05:
I agree with @MiSimon 
I also think that it's the most simple and robust way!

Masi2017, 2018-06-07:
Thanks for the comments,  I'll try it.

Masi2017, 2018-06-19:
Another question, is that any way to know that when lvgl is idle? I mean my_disp_flush is done completely?

kisvegabor, 2018-06-20:
Please give us feedback about the previous topic. 

Now you can use the `lv_task_get_idle` to see how much percentage of time is spent in `lv_task_handler`.

However it would be possible to add function which can tell when will be the next task called.

Masi2017, 2018-06-26:
Thanks, About the previous topic, I decided not to use static array for now since I can not move list operations to lvgl task.

kisvegabor, 2018-06-26:
I see, thank you.

And have you found a solution?

kisvegabor, 2018-07-07:
As the issue is solved I close it.
If you have some time, please let us know how did you solve it.

closed 2018-07-07 by kisvegabor