lvgl/lvgl #270: flush not called

rs-development, 2018-06-09:
Hello,

First a big thanks for all your work, it looks awsome!

But i am running in a problem:
After initializing and registering the flush handler it put the task handler and the tick handler in the main loop. The flush handler gets called a few time to fill the display plain white. After that the flash callback is never called again. I have tried with multiple examples and so on.

Is there something obvious i have to activate to enable the display after first white screen?

rs-development, 2018-06-09:
lv_init();
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);           /*Basic initialization*/
    disp_drv.disp_flush = my_disp_flush;   /*Initialize the field here. See below.*/
    lv_disp_drv_register(&disp_drv);          /*Register the driver in LittlevGL*/

    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);

    while (1)
    {
    	  lv_tick_inc(1);
    	  HAL_Delay(1);
    	  lv_tick_inc(1);
    	  HAL_Delay(1);
    	  lv_tick_inc(1);
    	  HAL_Delay(1);
    	  lv_tick_inc(1);
    	  HAL_Delay(1);
    	  lv_task_handler();
      }

rs-development, 2018-06-10:
Found the error by myself,

Since i added the label before the first flush operation the label was already integrated in the flush buffer. I had an error in the color pointer handling, witch lead to each pixel be colored like the first one.

After i added a object later in the loop i recognized my mistake since the object was drawn fine....

closed 2018-06-10 by rs-development

kisvegabor, 2018-06-11:
Great! :)