lvgl/lvgl #273: Help wanted - PNG or JPG decoder

kisvegabor, 2018-06-12:
Hi,

I will redesign the image interface of LittelvGL to be extendable with image decoders. So you will be able to register a decoder function for example for PNG images and library will call it to decode the JPG image data.

I tried [upng](https://github.com/elanthis/upng) with LittelvGL and worked however it was slow and consumed a lot of memory to always decode the full image.  

It would be much better to start decoding from a given line, and get the next line with a function without storing the whole uncompressed image.  It would be awesome to supporting PNG and/or JPG images this way.

If you would like learn the decoding of PNG or JPG images and write a decoder which works like this, please apply here! :)

BesitzeRuf, 2018-06-12:
Hi Gabor,

it will be great if it will be possible to use our costum decords with your implementation of the interface (for ex. the image data, which contain color lookup table (for DMA2D)). This can be powerful instrument to draw GUI objects based on images.

C47D, 2018-06-12:
Hi,

Apart from upng, do you know any other decoders or documentation to get familiar with decoders?

kisvegabor, 2018-06-12:
@BesitzeRuf Yes, it's a good idea to use DMA2D to accelerate image decoding into a buffer. If you already know these DMA2D features can you tell which formats are supported? For example is it possible use 16 color palette (4 bit) stored as two colors/byte? 

@C47D 
I learned that both (png, jpg) uses a compression method called [Huffman coding](https://en.wikipedia.org/wiki/Huffman_coding) but in addition JPEG stores "color frequencies" of 8x8 blocks.
If found theses sites which seemed useful for me:
* PNG: https://www.w3.org/TR/PNG/#11IHDR
* JPG:  https://www.impulseadventure.com/photo/jpeg-huffman-coding.html
and of course wikipedia: [png](https://en.wikipedia.org/wiki/Portable_Network_Graphics), [jpeg](https://en.wikipedia.org/wiki/JPEG)

BesitzeRuf, 2018-06-14:
@kisvegabor Hm... In direct way it is not possible. The CLUT table for each layer support up to 256 bytes. Each cell fo the table will store 32bit value... Maybe in the same way as it is described here (http://parttimeelectronics.blogspot.com/2015/02/playing-with-chrom-art-accelerator-of.html) it can be possible, with some limitations (probably with 2 convertation steps)

avafinger, 2018-06-18:
@kisvegabor ,

I just came across to this, maybe this can help someway, originally written for C++ , according to the author, rename it to C and compile, or just get some ideas.

Take a look at https://raw.githubusercontent.com/lvandeve/lodepng/master/lodepng.cpp

reference: https://lodev.org/lodepng/

tjstyle, 2018-07-01:
@kisvegabor do you was try nemuisan work of libpng port for STM32's hardware?
here is the original article in Japanese:
http://nemuisan.blog.bai.ne.jp/?eid=192848#STM32F4

It was ported to F4 and F7 hardware, I also using that libpng for some project.

kisvegabor, 2018-07-02:
@tjstyle Thank you very much!
I had a quick look on the JPG library. Is it really decodes the JPG image image MCU-by-MCU (gives 8x8 px block one-by-one)?

tjstyle, 2018-07-02:
Hi @kisvegabor Sorry, I don't really understand with your question.
And actually in some F4 and F7 family's have hardware codec capability, but until now I can't make it work due poor documentation by ST.

kisvegabor, 2018-07-02:
I mean the library not de-compresses the whole image at once. Rather returns with every single MCU. MCU is the 8x8 px basic unit of JPG image and not "microcontroller" here.  

I've just realized, you meant "lib/display/jpeg/" and not "lib/Utilities/JPEG/" (as it uses the hardware codec)

Just to be sure I downloaded the ZIP file from here: https://onedrive.live.com/?id=36F4D1230F8A673C%214346&cid=36F4D1230F8A673C

tjstyle, 2018-07-02:
Oh okay, before I think is "microcontroller".
Yes, it's on "lib/display/jpeg/" with jd_decomp for decompressing the jpg file.

avafinger, 2018-07-16:
Hi @kisvegabor ,
I'm following his thread but hw decode/encode (jpg/png) is not always available or documented on arm world, initially, i thought using the latest lodepng would improve things but memory allocation/deallocation is very expensive. Maybe a one heap scheme with suballoc blocks would speed up things but still need to decompress the whole image. If i come up with something workable i will share.

kisvegabor, 2018-07-16:
@avafinger I just realized there is no way to decompress a PNG image from an arbitrary position. But I learned that theoretically, PNG needs only a two line buffer the decode it (the difference between the lines is stored) but I haven't found a library which works this way.
A faster, but still not memory friendly solution would be still a big step ahead. So I'm curious what you will find! :)

As an alternative to PNG (to be fast, and memory effective) I was thinking about a simple run-length encoding. For icons, and images with a few colors and without frequent color change it might result a great compression ration. I guess even 90%.

avafinger, 2018-07-26:
@kisvegabor,
I've tried to benchmark the latest lodepng with a BLOCK/PARCEL memory allocator and see if decoding could be improved without resorting to libc malloc/realloc that i supposed was slow for realloc. This memory allocator predates the Win16 bit era where memory was scarce and could be fragmented very easily. 

The memory manager is very simple and works like this: 
* alloc a big chunk of memory
* when memory is requested it sub-alloc the BLOCK in PARCEL and split if necessary
* when memory is freed, it looks for PARCELS marked as free and joins them to have a continuous PARCEL
* This code was lying around for ~20 years, if i recall correctly, the author is WARD / SMITH, I added some memory checking, changed a bit here and a bit there to adapt to 32bit and 64bit.

A surprisingly similar method is used by **wmem**, see here: https://code.wireshark.org/review/gitweb?p=wireshark.git;a=blob;f=doc/README.wmem 

See here for a detailed description about BLOCK and CHUNK
https://github.com/eapache/wof_alloc

I have here another memory manager very similar to the wof_alloc, using POOLS, but i thought it should be as simple as possible to have a real improvement. TBH i think one needs to spend lots of time on this to gain little performance, not sure if worth it. I mean about memory manager!

I tried to benchmark the function but i get very different timings, sometimes better, sometimes noticeably slower. Maybe you take a look and find this interesting and improve it, so it can be useful.

Attached is the code you can have a look and see, do some benchmarks, adjust structure and remove some memory corruption checking, or call ReclaimMemory not so frequently with the expense of having the BLOCK more fragmented, but this would speed up things.

**About the code and Benchmark**
Build testpng example with usual **libc malloc/realloc**:
gcc -O2 -o testpng example_decode.c lodepng.c block.c (you may need to add zLib to the build)

**Benchmark:**
./testpng **ic.png**

**Build with BLOCK/PARCEL malloc/realloc:**
gcc -DMEM_BLOCK -O2 -o testpng example_decode.c lodepng.c block.c

**Benchmark:**
./testpng **ic.png**

I used **decodeTwoSteps(const char* filename)** to benchmark
 error = lodepng_decode32(&image, &width, &height, png, pngsize);

Results vary every new run for the better and for the worse, maybe you try with lvgl and see if you get a real improvement.

**Adjusting the Block size:**
* Edit block.c and change **BLOCKSIZE**
* Alignment could help for some arch 

Attached is the source code.
[decode_png.zip](https://github.com/littlevgl/lvgl/files/2233146/decode_png.zip)

kisvegabor, 2018-07-29:
Hi,

I tested the **loadpng** and it worked perfectly! :)

[image]

But one thing was not clear to me. What is the preferred way to load a PNG image from buffer? I found the `lodepng_buffer_file` function but it is static.

avafinger, 2018-07-30:
Great to see this progress!

I'm not sure i understand the question, you would like something to read a stream in a small chunk and then decode, read again and decode, until no more stream to read? This would save memory but would be really slow. IMHO.It is just a guess.

I think **decodeGeneric** should be re-written for that purpose.

kisvegabor, 2018-08-01:
I mean if you convert to a png file a C array and store it in the flash as constant data then you can load it as it were a file's data.

kisvegabor, 2018-09-03:
Finally, I added the image decoder interface to LittelvGL. It wasn't easy to find the best way to do it to keep the good performance and provide flexibility. I hope you will like the result!

I merged it to `dev-5.2` and wrote a post about it: https://littlevgl.com/blog/50/use-png-images-in-littlevgl
I used `lodepng` as @avafinger  suggested.

The image decoder interface required new image format which means you need to convert you images again with the new image converter tool: http://www.test.littlevgl.com/image-to-c-array

FYI, in `v5.2` 1, 2, 4 and 8 bit color formats will be natively supported.

avafinger, 2018-09-03:
@kisvegabor 
**Thanks for your work**, I will give a try next weekend.
I did not have a good answer to your question so a remained quiet... sorry.

kisvegabor, 2018-09-05:
@avafinger  No problem! Meanwhile, I fond the solution.

Thank you for the test in advance!

kisvegabor, 2018-10-05:
The new image decoder interface is released in v5.2.
Here you can see how to add your own decoder library: https://littlevgl.com/blog/50/use-png-images-in-littlevgl

If there are some news on this topic just reopen this issue.

closed 2018-10-05 by kisvegabor

m943040028, 2019-05-10:
Hi, I tried png_decoder.zip in the blog post, but found the red and blue channel are swapped.

Original file

[image]

Image on pc simulator

[image]

kisvegabor, 2019-05-10:
You can swap the red and blue color in `convert_color_depth()` in `png_decoder.c`.

m943040028, 2019-05-10:
Hi, thanks for your quick reply. 

My LV_COLOR_DEPTH is 32, it seems like convert_color_depth() doesn't have any effect.

I changed the backend png library to libpng, but result is the same.
After I invoked png_set_bgr(), result is correct.
http://refspecs.linuxbase.org/LSB_3.1.0/LSB-Desktop-generic/LSB-Desktop-generic/libpng12.png.set.bgr.1.html

It is strange since the SDL surface format is ARGB8888, but I need to tell the png decoder to change the pixel order to bgr to have a correct result.

I found the picture on the blog has the same result.
[image]

[image]

m943040028, 2019-05-10:
Hi kisvegabor,

It might be endian issue. PNG pixel format is R, G, B and followed by Alpha channel. (According to [this](https://stackoverflow.com/questions/2384111/png-file-format-endianness), png format is big-endian)

x86_64 is little-endian, thus it interprets as A, B, G and R, and our SDL surface is ARGB8888.
Thus, Red and Blue was swapped.

kisvegabor, 2019-05-10:
Wow, I haven't noticed it. 

I fixed `convert_color_depth` can you also verify if it really works well now?

m943040028, 2019-05-13:
Sure.
How do I get the fixed version? I tried the version on the blog post, but result is still the same.

kisvegabor, 2019-05-15:
Sorry, I uploaded it to the wrong place.
Now it should be on the original place: https://github.com/littlevgl/blog/raw/master/assets/png_converter/png_decoder.zip

m943040028, 2019-05-15:
Hi,

I tried but it would compile. Need a few modifications

Change
`
img_c[i]red = c->blue;
img_c[i]blue = c->red;
`
to
`
img_c[i].red = c.blue;
img_c[i.]blue = c.red;
`

Then It works

[image]

kisvegabor, 2019-05-15:
Thank you!

Fixed, again. :)

I tested it with a different version of LittelvGL and added the updates manually. I shouldn't trust in myself so much :)

nikunj8780, 2024-10-23:
`simple how to save image (png) file in our esp LittleFS storage and how to using in Arduino IDE to display this image in any display like ILI9488 ; LIL9441 any display but give atlist one demo code for Arduino IDE with LVGL 9.2.0 and TFT_eSPI 4.5.43`