lvgl/lvgl #274: Object roller: selected option not saved for LV_SIGNAL_CONTROLL/LV_GROUP_KEY_ENTER

StephanSy, 2018-06-12:
Hi!
Saving of the selected option for signal LV_SIGNAL_CONTROLL/LV_GROUP_KEY_ENTER is missing for the roller object. Thus the selected option is lost when the group focus is changed.
BR,
Stephan

if(ext->ddlist.action) ext->ddlist.action(roller);
         ext->ddlist.sel_opt_id_ori = ext->ddlist.sel_opt_id;

```c
} else if(c == LV_GROUP_KEY_ENTER || c == LV_GROUP_KEY_ENTER_LONG) {
    ext->ddlist.sel_opt_id_ori = ext->ddlist.sel_opt_id;  /* <= Missing line for fix */
    if(ext->ddlist.action) ext->ddlist.action(roller);
    ext->ddlist.sel_opt_id_ori = ext->ddlist.sel_opt_id;		/*Set the entered value as default*/
}
```

kisvegabor, 2018-06-12:
Hi,

It should be already fixed in `beta`. 
Is it working for you using the beta branch?

StephanSy, 2018-06-13:
Hi!
No, it is not working for me on the beta branch without above fix.

StephanSy, 2018-06-13:
Hi! There is a second problem here:
If lv_roller_set_selected() is called from outside the object on a roller object currently in focus, this setting is lost when the focus changes. 
I would propose provide an exported function for set from outside and different static one for a set triggered from a RIGHT/LEFT signal.
BR
Stephan

kisvegabor, 2018-06-14:
Hi,
**The original question**
It is your suggestion:
```c
   ext->ddlist.sel_opt_id_ori = ext->ddlist.sel_opt_id;  /* <= Missing line for fix */
    if(ext->ddlist.action) ext->ddlist.action(roller);
    ext->ddlist.sel_opt_id_ori = ext->ddlist.sel_opt_id;		/*Set the entered value as default*/
```
The 3. line is the same as your suggested fix. Now it saves twice the new id. Or am I missed something?

**lv_roller_set_selected() while focused**
As `lv_roller_set_selected` is a wrapper to `lv_ddlist_set_selected` what do think about adding a line like this: `ext->sel_opt_id_ori = ext->sel_opt_id;` to update the original value.

StephanSy, 2018-06-14:
Oh, sorry. I some got confused with the branches... It is fixed.

For the second problem:
- Add  `ext->sel_opt_id_ori = ext->sel_opt_id;` to `lv_roller_set_selected`
- Use `lv_ddlist_set_selected` instead of  `lv_roller_set_selected` inside function `lv_roller_signal`

BR,
Stephan

kisvegabor, 2018-06-14:
As it affects ddlist as well I think it should be added to lv_ddlist_set selected to fix both objects.

kisvegabor, 2018-07-15:
It is fixed in the `beta` branch So I close this issue.

If it's not working properly feel free to comment here and reopen the issue.

closed 2018-07-15 by kisvegabor