lvgl/lvgl #275: Not focus on object in a group

Masi2017, 2018-06-12:
Hi,
I was wondering if it is possible to focus on no object in a group? I created two groups and once the other group is selected I want to defocus the other otherwise user may get confused which item has been selected .

kisvegabor, 2018-06-12:
Hi,

I think the most simple solution would be to create a hidden object (`lv_obj_set_hidden(obj, true)`) and focus on it you would like to hide the focus.

Masi2017, 2018-06-15:
HI, Thanks I'll try this.

aenchevich, 2018-07-08:
Is this the same issue as : "[Group focus and page focus bugs](https://github.com/littlevgl/lvgl/issues/282) #282"  ?

Masi2017, 2018-07-10:
@aenchevich Yes can be but it's not completely the same . In fact in case of having two different groups, if assigned group is changed, the last focused object for the previous assigned group still remained focused.

kisvegabor, 2018-07-25:
@Masi2017 I fixed _"last focused object for the previously assigned group still remained focused"_ issue. Thank you for pointing this out. The update is in the `dev-5.2` branch.

As the original question is answered and the other issue is solved I close this issue. If it's not working properly don't hesitate to comment here! Thank you!

closed 2018-07-25 by kisvegabor