lvgl/lvgl #276: Complex lv_obj_del bug (with potentially simple solution)

upbeat27, 2018-06-12:
So, an example of the bug is this:
Say you have a page, and on this page are a number of objects.
These objects are in a group.
One of these objects is a ddlist.

If you want to delete all the objects (for example, to show a different set of objects, i.e. a different menu) you can clean them with `lv_obj_clean(lv_page_get_scrl(page))`. Note: this would happen also if you just delete the ddlist only.

If you have the ddlist in focus at this time, when it comes to delete the ddlist it will go into `lv_obj_del`, where it will delete all the children of the ddlist. A bit later it will delete the ddlist from the group, which will trigger `lv_group_focus_next`. This will then cause the invalidation of the ddlist object, which will trigger `lv_ddlist_signal` with `LV_SIGNAL_DEFOCUS` which will trigger `lv_ddlist_refr_size` which will eventually try to get the style of the ddlist label.

The problem is that at this point the children of the ddlist have been deleted, and the label is gone - `ext->label` is NULL, and it will attempt to get its style. Not good!

This bug would show up with other objects as well (I'm guessing those that have defocus signals).

Looking at `lv_obj_del` simply moving:
```
    /*Delete from the group*/
 #if USE_LV_GROUP
    if(obj->group_p != NULL) lv_group_remove_obj(obj);
 #endif
```
above deleting the children fixes it. The thing is, I am unsure if that introduces some other complicated side effect, perhaps in some other incredibly specific scenario such as this one.

kisvegabor, 2018-06-14:
Hi,

Thank you for examining this complex issue. 

Based on you suggestion I  changed to order of actions in `lv_obj_del`.
Can you test it?

upbeat27, 2018-06-15:
It fixes this bug, yes. Time will tell if it breaks something else ;)

closed 2018-06-15 by upbeat27