lvgl/lvgl #277: Enhancement: More flexibility for focused object style

StephanSy, 2018-06-13:
Currently, you can only have one common style modification per group for the focused object.
This is inflexible. For different type of objects different modifications should be possible.

Therefore I propose the enhancement that the objects should provide a method for the focus style modification.

BR,
Stephan

kisvegabor, 2018-06-14:
Hi,

It sounds a reasonable update. 

My idea it to modify the style in a signal, this way all object types can make their own updates. What do you think about it.

StephanSy, 2018-06-14:
Hi!
You mean a new signal?

There are already the signals LV_SIGNAL_FOCUS / LV_SIGNAL_DEFOCUS, so the objects could implement a change of their style triggered by these signals.

But as long as there is the current implementation of a common style modification by the group, a style modification by a object would be overwritten by the common one, wouldn't they?

We also need to consider that objects like the roller have several styles to be modified.

BR,
Stephan

kisvegabor, 2018-06-14:
Yes, I ment a new signal but it really can be in the FOCUS signal.

Now every group contains a unique style (`group->style_tmp`) which is the modified version of the object's current style. Using this temporary style avoids the overwriting of the real style (which can belong to other objects too).

And now the style modification (style_mod callback) is applied every time when `lv_obj_get_style` is called. However, you are right it would be enough to calculate it only once when the object comes into focus (and update it if the object's style changes meanwhile). It would be easy to add this update to the FOCUS signal.

And yes, objects can have multiple styles. Usually the one, in `obj->style_p` (lv_obj_t) is considered the most relevant "main" style so only this one is updated. But if the `group->style_tmp` would be an array, like `group->style_tmp[8]` all object types could index it like `LV_DDLIST_STYLE_BG`, `LV_DDLIST_STYLE_SEL` etc. So it seems possible to support the modification of the extra style too.

There is only one thing I can't imagine right now: how can the user affect the style modification? How should a style_mod callback look like now if it should affect so much object types and "internal" styles?

StephanSy, 2018-06-15:
Hi!
The objects could (optionally) provide a "`_set_focused_style(lv_obj_t *, lv_objectname_style_t , lv_style_t *)`" method. The object would contain a pair of pointers per style type, one pointer to the normal (defocused) style, one to the focused style. (Maybe the focused style pointer is default NULL if not set by user?).
If an object is focused or defocused, the object switches the applied style.
Maybe the object could also have a 3rd pointer per style type which points to the currently valid one and  this pointer is witched between the defocused and focused style.

BR,
Stephan

kisvegabor, 2018-06-17:
Hi,

It sounds very flexible because the user can set all the styles as he wants. However it would increase the memory footprint of all object even you don't want use this feature. 

If we handled the style update in the signal function (and object would do some predefined modifications on the basic styles) the user still would have the opportunity to update/replace the signal function by a unique one using the `lv_obj_set_signal_func()` function. It's not well documented yet but it's totally possible and "valid". And I suppose it would be required rarely.   

By the way can you mention some object types or cases where unique style modification would be required?

StephanSy, 2018-06-18:
Hi!
I found it necessary to modify the LV_ROLLER_STYLE_SEL style of a focused roller object. It should fit to the modified background style. 

For the system I am currently evaluating your library for,  we will only have a monochrome LCD and a rotary encoder with push button is the only input device. So I am looking for a way to discern a focused object from a focused and editable object. That is why I need more flexibility with the style modifications.

Replacing the signal functions would work, but this would not be user friendly way.

BR
Stephan

kisvegabor, 2018-06-19:
Hi,

I see. Already there was some discussion about controlling  with one encoder but there wasn't a correct solution/workaround so far.

But now I added this feature the `dev-5.2` branch! With `LV_GROUP_KEY_LONG_PRESS` you can change between "edit" and "navigate" mode. With `lv_group_get_editin()` you can get the current mode and decide whether to send `LV_GROUP_KEY_NEXT/PREV` on encoder rotate or `LV_GROUP_KEY_LEFT/RIGHT`. This can be enabled by `lv_group_set_edit_enable()`.

It should simplify the encoder usage a  lot.

But please keep in mind `dev-5.2` is development branch and can change.

kisvegabor, 2018-06-21:
I see.
The roller seems suitable but you should set a transparent style for the selected area. This way you will see the background if focused.

I modified the code you send in #278 issue:
```c
    roller[0] = lv_roller_create(lv_scr_act(), NULL);
    lv_roller_set_style(roller[0], LV_ROLLER_STYLE_SEL, &lv_style_transp_fit);
    lv_roller_set_visible_row_count(roller[0], 1);
```
Which resulted:
[image]

StephanSy, 2018-06-22:
Hi!
Interesting. With the new "edit group focus" feature, will it possible to have "focused edit" style with white text on black background?

I tried to get a more compact style with the following code:

```
lv_style_copy(&bg_style, lv_roller_get_style(roller[0], LV_ROLLER_STYLE_BG));
bg_style.body.border.width = 0;
bg_style.body.radius = 1;
bg_style.body.padding.hor = 2;
bg_style.body.padding.ver = 0;
lv_roller_set_style(roller[0], LV_ROLLER_STYLE_BG, &bg_style);
```

The results is ok:

[image]

Bit when I set `bg_style.body.radius = 1;`, it looks a bit strange ;-)

[image]

See also source file:

[main.c.txt](https://github.com/littlevgl/lvgl/files/2127384/main.c.txt)

kisvegabor, 2018-06-23:
Hi,

In `lv_conf.h` `LV_ANTIALAIS` is 0? It should be disabled for monochrome designs.

Yes you can set a black appearance for edit mode. 
I applied this code the create a black style:
```c
    lv_style_copy(&sel_style, &lv_style_transp_fit);
    sel_style.text.color = LV_COLOR_WHITE;
    lv_roller_set_style(roller[0], LV_ROLLER_STYLE_SEL, &sel_style);

    lv_style_copy(&bg_style, lv_roller_get_style(roller[0], LV_ROLLER_STYLE_BG));
    bg_style.body.border.width = 0;
    bg_style.body.radius = 2;
    bg_style.body.padding.hor = 2;
    bg_style.body.padding.ver = 0;
    bg_style.body.main_color = LV_COLOR_BLACK;
    bg_style.body.grad_color = LV_COLOR_BLACK;
    lv_roller_set_style(roller[0], LV_ROLLER_STYLE_BG, &bg_style);
```
I also added some extra space between the rollers.
[image]

StephanSy, 2018-06-27:
Hi!
Thanks, but actually I wanted to know if I can switch between the style black and white background when the object is switched between "focused" and "focus edit" with this new "lv_group_set_edit_enable()". But I guess for that you would need to modify the selected style of the roller.
BR,
Stephan

StephanSy, 2018-06-27:
Hi!
Also with LV_ANTIALIAS = 0 , the result for radius=0 looks wrong:

[image]

BR
Stephan

kisvegabor, 2018-06-29:
To test it please copy here or attache:
* the relevant part of you code
* your lv_conf.h
* and font you used

kisvegabor, 2018-07-25:
@StephanSy 
Finally, were you able to make it work?

StephanSy, 2018-07-26:
Hi!
Sorry, no I have not been able to reproduce this. Strange... 

Btw. in the meantime I changed to lv_label with a modified signal function instead of the lv_roller. This solves my problem of the focused navigate/edit style.
So I think we can close this ticket.
BR
Stephan

kisvegabor, 2018-07-26:
Awesome! Thank you  for the feedback!

closed 2018-07-26 by kisvegabor