lvgl/lvgl #278: Roller height calculation or label position wrong for theme mono

StephanSy, 2018-06-13:
Hi!
I have seen a problem with the height calculation when using the theme mono. 
The upper roller border is too close to the highest option. Especially bad when the roller has a thicker border because then the highest option cut off. Also the selected option is positioned too high on the "selected frame".
See my example below.
BR
Stephan

[image]

kisvegabor, 2018-06-14:
Hi,

As roller and ddlist are very similar do think it is related to this? https://github.com/littlevgl/lvgl/issues/174

StephanSy, 2018-06-14:
Uh, no idea. I was not able to understand what is causing the problem so far.
Sorry!

kisvegabor, 2018-06-14:
Hi,

I check it. It seams like a theme issue instead of roller issue. I added "Ág" characters to the roller ("Á" to see top of the characters and "g" to see the bottom) and I saw that the position is correct.
It seems to me only a little bit more line space is required in the mono theme. I added an update to `beta`

Oroginal:
[image]
 

More line space:
[image]

StephanSy, 2018-06-15:
Hi!
It is better, but still there is not enough space between the top label and the border.
This is a screen shoot after your latest update. You can still see that the top text is too close to the border. Also the selected option is too close to the "selected" frame.
BR
Stephan

[image]

kisvegabor, 2018-06-17:
Hi,

I tested it with "A|g" characters and looked like this for me (using the built in 10 px font):
[image]

Which font have you used? And can you send a screenshot with "AIg" texts on the roller?

StephanSy, 2018-06-18:
Hi!
Sorry, I was not able to get the special character printed. 
Maybe I am doing something wrong, but I don't have the time right now...
Have you tried a 3pt border? That is the default 'focused' style.
BR
Stephan

StephanSy, 2018-06-18:
[image]

kisvegabor, 2018-06-19:
Thank for the image. It seems there is a few pixel error. I will examine it deeper.

kisvegabor, 2018-06-19:
I checked it with 7px and 8px monochrome fonts on Font converters website. And found that there was zero (8px font) or one pixel (7px font) error.

Is it possible that there a few empty line in your font below the letters. Can you tell which font do you use? Or even send the font's "C" file?

StephanSy, 2018-06-21:
Hi
I am using the lv_font_dejavu_10. 

Here the source files.

[RollerTest.zip](https://github.com/littlevgl/lvgl/files/2123246/RollerTest.zip)

BR
Stephan

kisvegabor, 2018-06-21:
Hi,

Thank you!

I found a minor issue which caused 1 px error when calculating the size of the "selected" area. 
However using only numbers still looks ugly because number doesn't have lower parts like some letters have.

I suggest to use an other font which is designed to be small for low resolution displays. There are two on the font converter's website (namely: Syncronizer and Unscii)

With Syncronizer I got the this (a 2 px border in style mod function might be better): 
[image]

StephanSy, 2018-06-22:
Hi!
Yes, I am already looking for decent 8pt and 16pt fonts. This "syncronizer font" looks fine for 7&8pt. However, I was not able to find a decent larger font. Do you have a tip for me?
BR,
Stephan

kisvegabor, 2018-06-23:
Hi,

You can convert the 7px and 8px fonts in 14px and 16 px size. They will remain pixel perfect.

However this way the the glyphs are really just doubled which means for example a "/" character will be built from 2x2 blocks instead of smoother 1x1.

kisvegabor, 2018-06-24:
Or you can simply find 16px pixel fonts in google. Like this: https://fontstruct.com/fontstructions/show/1123988/properhitboxglobal

kisvegabor, 2018-06-29:
Hi,

Pixels fonts need to be converted in exactly the same height as they designed.
I played with it and found that it is a 9px font. I attached the result.

[px_font_9.c.zip](https://github.com/littlevgl/lvgl/files/2149397/px_font_9.c.zip)

kisvegabor, 2018-07-15:
As the original issue is solved I close this.

closed 2018-07-15 by kisvegabor