lvgl/lvgl #280: How to change the font of a single ddlist object?

StephanSy, 2018-06-14:
Hi!
I would like to change the font size of individual objects (not the default font size).
I tried the code below to change from default font size 20 to 10, but it seems I also need to change the spacing somehow.
Can you please help me?
BR, 
Stephan

```c
lv_obj_t * ddlist = lv_ddlist_create(lv_scr_act(), NULL);
lv_ddlist_set_options(ddlist, "Low\nMedium\nHigh");
lv_obj_set_pos(ddlist, LV_DPI / 4, LV_DPI / 4);
lv_group_add_obj(g, ddlist);                    /*Add the object to the group*/

lv_style_copy( &bg_style, lv_ddlist_get_style( ddlist, LV_DDLIST_STYLE_BG ));
lv_style_copy( &sel_style, lv_ddlist_get_style( ddlist, LV_DDLIST_STYLE_SEL ));

bg_style.text.font = &lv_font_dejavu_10;
sel_style.text.font = &lv_font_dejavu_10;

lv_ddlist_set_style( ddlist, LV_DDLIST_STYLE_BG, &bg_style );
 lv_ddlist_set_style( ddlist, LV_DDLIST_STYLE_SEL, &sel_style );
```

Before:

[image]

After:

[image]

kisvegabor, 2018-06-14:
Hi Stephan,

First of all then you for persistent work on reporting these issues!

Your code is fine it was tricky bug in the library. When the ddlist came to focus it started an "open" animation which modified its height. Meanwhile the style change also changed the height but animation overwritten it.

I fixed it in `beta`.

StephanSy, 2018-06-15:
Thank you! I have tested it and it works now also for me.
BR
Stephan

closed 2018-06-17 by kisvegabor