lvgl/lvgl #282: Group focus and page focus bugs

upbeat27, 2018-06-19:
Group focus bug (simple):
Focus callback is not called for `lv_group_focus_obj`, just `lv_group_focus_next` and `lv_group_focus_prev`.

Page focus (less simple):
An example setup:
- animations enabled and `lv_page_focus` is called with anim_time non-zero.
- multiple objects on a page, two of which (Object 1 and Object 2) can not be visible at the same time - so triggering `lv_page_focus` on either would result in the other not visible.
- Object 1 is visible, `lv_page_focus` is called on Object 2, not visible.
- before the animation starts to move focus to Object 2, `lv_page_focus` is called on Object 1
- `lv_page_focus` on Object 1 does nothing because it is technically visible, since the focus has not moved to Object 2 yet
- animation for page focus on Object 2 is done, Object 1 is not visible even though it was the last object to have `lv_page_focus` called on it

To fix this, the focus animation should still "happen" (really, no animation will happen) even when the object is already in focus, in order to cancel any already existing focus animation. The animation time should be 0 for this case, so the result is just deleting the existing animation.

@kisvegabor,  here is my take at this fix, but you may prefer some other flow for achieving the same goal.
Note that this modification also forces `anim_time` to 0 when animations are not enabled - in the current code, nothing will happen when calling `lv_page_focus` with a non-zero `anim_time` with animations disabled. In my eyes this way makes more sense.

```c
void lv_page_focus(lv_obj_t * page, lv_obj_t * obj, uint16_t anim_time)
{
    lv_page_ext_t * ext = lv_obj_get_ext_attr(page);
    lv_style_t * style = lv_page_get_style(page, LV_PAGE_STYLE_BG);
    lv_style_t * style_scrl = lv_page_get_style(page, LV_PAGE_STYLE_SCRL);

    lv_coord_t obj_y = obj->coords.y1 - ext->scrl->coords.y1;
    lv_coord_t obj_h = lv_obj_get_height(obj);
    lv_coord_t scrlable_y = lv_obj_get_y(ext->scrl);
    lv_coord_t page_h = lv_obj_get_height(page);

    lv_coord_t top_err = -(scrlable_y + obj_y);
    lv_coord_t bot_err = scrlable_y + obj_y + obj_h - page_h;

#if USE_LV_ANIMATION
    bool cancel_anim = false;
#else
    anim_time = 0;
#endif
    /*If obj is higher then the page focus where the "error" is smaller*/

    /*Out of the page on the top*/
    if((obj_h <= page_h && top_err > 0) ||
       (obj_h > page_h && top_err < bot_err)) {
        /*Calculate a new position and let some space above*/
        scrlable_y = -(obj_y - style_scrl->body.padding.ver - style->body.padding.ver);
        scrlable_y += style_scrl->body.padding.ver;
    }
    /*Out of the page on the bottom*/
    else if((obj_h <= page_h && bot_err > 0) ||
            (obj_h > page_h && top_err >= bot_err)) {
        /*Calculate a new position and let some space below*/
        scrlable_y = -obj_y;
        scrlable_y += page_h - obj_h;
        scrlable_y -= style_scrl->body.padding.ver;
    } else {
        /*Already in focus*/
#if USE_LV_ANIMATION
        cancel_anim = true;
#else
        return;
#endif
    }

    if(anim_time == 0) {
        lv_obj_set_y(ext->scrl, scrlable_y);
    }
#if USE_LV_ANIMATION
    else {
        lv_anim_t a;
        a.act_time = 0;
        a.start = lv_obj_get_y(ext->scrl);
        a.end = scrlable_y;
        a.time = (cancel_anim ? 0 : anim_time);
        a.end_cb = NULL;
        a.playback = 0;
        a.repeat = 0;
        a.var = ext->scrl;
        a.path = lv_anim_path_linear;
        a.fp = (lv_anim_fp_t) lv_obj_set_y;
        lv_anim_create(&a);
    }
#endif
}
```

kisvegabor, 2018-06-20:
Hi,

**Simple**
Thank you for pointing this out! 

**Complex**
Thank you for the examination and the precise description!
Your proposed solution sounds great! I would think in the similar way. :) 

You mentioned that nothing happen with zero anim. time if animations are disabled. True. I tried to follow a systematic way of thinking for these cases like in [lv_roller_set_selected](https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_roller.c#L123). So set `anim_time` or `anim_en` to zero if animations are disabled. It seems I forget to do this in this function.

Do you have time to add these updates in a Pull request?
Just in case of you have modifications in lvgl I run a code formatter on the source code so there are lot of changes in the `beta`. So it might be difficult to merge.

Zaltora, 2018-07-03:
I am not sure if it is related to this, but i want do something and it is didn't work:
I want use a keypad with group focus to scroll a page with 2 button ( up / down ).
But i can't do this with the example "lv_test_group".

If it is a normal behavior, can be great to add the possibility to scroll manually with function (it maybe already the case but i don't no how)

kisvegabor, 2018-07-03:
Hi,

Righ now `lv_page` does nothing if added to a group but it's really useful feature. So I just added it to `beta` branch and also fixed the original bug related to `lv_page_focus`.

Zaltora, 2018-07-03:
>  So I just added it

sry, you mean added the possibility to navigate with group ?

kisvegabor, 2018-07-03:
Yes. Now You can use LV_GROUP_KEY_DOWN/RIGHT to scroll down and LV_GROUP_KEY_UP_LEFT to scroll up the page.

kisvegabor, 2018-07-04:
Please let me know how it works for you when you've tested it.

Zaltora, 2018-07-04:
It is implemented in 5.2 branch ? Yes, i will do some feedback about it.
It is possible to scroll with a function too ? I want use a timer to do automatic scrolling when no activity. Will be very useful for all object with scrolling capabilities.

kisvegabor, 2018-07-06:
It's in the `beta` branch.

Adding scroll function is also a good idea. But should go to `dev-5.2`

kisvegabor, 2018-07-25:
The discussion about the mentioned new scrolling feature should continue in https://github.com/littlevgl/lvgl/issues/298

As the original issue is solved I close this topic. If you find some related bugs feel free to comment here.

closed 2018-07-25 by kisvegabor