lvgl/lvgl #283: How can I let the scroll bar display all the time?

zhaojigao, 2018-06-20:
Hi

I use keyboard to control a list.How can I let the scroll bar display all the time?
I tried using lv_list_set_style to solve this problem, but I did not solve it.

LV_LIST_STYLE_SCRL
LV_LIST_STYLE_SB

lv_list_set_style(mainMenulist, LV_LIST_STYLE_SB, &style_btn_rel);

Which style should I use ？LV_LIST_STYLE_SCRL or LV_LIST_STYLE_SB or other style?
Which item in the lv_style_t is used to control whether the scroll bar is always displayed?

please offer me some advices,

thank you!

kisvegabor, 2018-06-20:
Hi,

You can use `lv_list_set_sb_mode(mainMenulist, LV_SB_MODE_ON)`

zhaojigao, 2018-06-20:
Hi 
	
	I use the following code to set the style of the list. and use lv_page_set_sb_mode(mainMenulist, LV_SB_MODE_ON) let the scroll bar display all the time!

       static lv_style_t sb;
	lv_style_copy(&sb, &lv_style_pretty_color);
	sb.body.padding.hor = 1; 
       sb.body.padding.inner = 3; 
	sb.body.radius = 0;
	lv_list_set_style(mainMenulist, LV_LIST_STYLE_SB, &sb);

    Thanks very much!

closed 2018-06-20 by zhaojigao

kisvegabor, 2018-06-20:
Your welcome!