lvgl/lvgl #284: ddlist   BUG?

as564335sa, 2018-06-22:
Hi!
In the "ddlist" obj，if I add Chinese to options，then I can't click it accurately，But there is no problem in using English！
example：
lv_ddlist_set_options(eth0_ddlist,"关  闭\n静  态\n动  态");

I must click "动  态" to select "静  态".When I click "动  态", it selects "关  闭".
 
What causes this problem？

as564335sa, 2018-06-23:
Hi!
I am using the simulator and not testing on other hardware.I think this question is not related to the font used.Now I changed Chinese into a built - in sign，but the problem remains

This is the font I used:

`lv_theme_t *th = lv_theme_material_init(200,&lv_font_dejavu_30);
lv_theme_set_current(th);
lv_font_add(&msyhbd,&lv_font_dejavu_20);
lv_font_add(&iconfont,&lv_font_dejavu_20);
lv_font_add(&msyhbd_30,&lv_font_dejavu_30);`

`lv_obj_t *eth0_ddlist;
eth0_ddlist = lv_ddlist_create(eth0_cont,NULL);
lv_ddlist_set_options(eth0_ddlist,SYMBOL_CLOSE"\n"SYMBOL_WIFI"\n"SYMBOL_AUDIO);
lv_obj_align(eth0_ddlist,label2,LV_ALIGN_OUT_RIGHT_TOP,LV_DPI,0);
lv_ddlist_set_style(eth0_ddlist,LV_DDLIST_STYLE_BG,&lv_style_btn_rel);
lv_ddlist_set_style(eth0_ddlist,LV_DDLIST_STYLE_SEL,&lv_style_plain_color);
lv_ddlist_set_selected(eth0_ddlist,2);
lv_ddlist_set_action(eth0_ddlist,debug);`

the debug() is:
`lv_res_t debug(lv_obj_t *obj)
{
	printf("select %d\r\n",lv_ddlist_get_selected(obj));

	char *str = lv_ddlist_get_options(obj);
	char len = strlen(str);
	printf("text : ");
	for (char i; i < len; i ++)
	{
		printf("%02X ",str[i]);
	}
	printf("\r\n");
	return LV_RES_OK;
}`

now when i click SYMBOL_CLOSE;
printf:
select 0
text : 80 84 0A EF 80 AB 0A EF 80 80

when i click SYMBOL_WIFI;
printf:
select 0
text : 84 0A EF 80 AB 0A EF 80 80

when i click SYMBOL_AUDIO;
printf:
select 1
text : EF 80 AB 0A EF 80 80

And there is no problem when I use English,the same is the symbol font and the Chinese are all UTF-8

kisvegabor, 2018-06-26:
Hi,

Sorry for the late answer.I will check it soon, but unfortunately I'm a little bit busy now.

as564335sa, 2018-06-26:
Hi:
I seem to have found the wrong place today，in the function `static lv_res_t lv_ddlist_release_action(lv_obj_t * ddlist)`(file lv_ddlist.c , line 637)

`uint16_t letter_i;
 letter_i = lv_label_get_letter_on(ext->label, &p);`
When use English, it can return the clicking character index correctly.But when use symbol font or Chinese(utf8), it cannot return the correct index.I think this is the wrong reason, but I haven't made it clear yet.

kisvegabor, 2018-06-26:
Hi,

I fixed it in `beta` branch. In `lv_ddlist_release_action` the evaluation of the the clicked option was wrong. However your suggestion about  `lv_label_get_letter_on` was pretty useful!

Please, test it!

as564335sa, 2018-06-27:
Thank you very much！The problem was solved！

kisvegabor, 2018-06-27:
Great! :)

closed 2018-06-27 by kisvegabor