lvgl/lvgl #285: Strange Colors on HX8357D

AGlass0fMilk, 2018-06-24:
Hi,

I'm working on porting LVGL over to ARM mbed-os and I'm working on the STM32F407-DISC1 board. I am using an Adafruit 3.5" TFT LCD breakout with an HX8357D driver IC.

I have gotten to the point where the display has the right shapes being printed out, but I cannot get the colors to be correct. I have tried configuring the display driver to accept RGB(5/6/5) and BGR(5/6/5) as well as inverting the colors to no avail. It still looks very... messed up (see picture below).

Any idea on what I'm doing wrong? Any hints would be great. Looking forward to working with lvgl :)

Thanks!

[image]

AGlass0fMilk, 2018-06-24:
I think I fixed it by making sure the color was written out in the proper format (RGB 5/6/5):

```
uint8_t pix_color = 0;
pix_color = (color_p->red << 3) | ((color_p->green & 0x38) >> 3);
*((__IO uint8_t*)(0x60010000)) = pix_color;
pix_color = ((color_p->green & 0x7) << 5) | (color_p->blue);
*((__IO uint8_t*)(0x60010000)) = pix_color;
```

You can find the port I'm working on over at my lvgl fork:
https://github.com/AGlass0fMilk/lvgl

closed 2018-06-24 by AGlass0fMilk

kisvegabor, 2018-06-25:
Great!

Thank you for sharing you code!

I would like add more blog post to littelvgl.com. If you are interested in writing about your porting, please write me to: kisevegabor@gmail.com

tomtristan, 2019-02-28:
It seems that I've the same issue, colors distorted like the picture above. Where exactly did you put these code lines to fix the problem?

embeddedt, 2019-02-28:
@tomtristan Are LittlevGL and the display configured to accept the same color depth?

kisvegabor, 2019-02-28:
You can enable `LV_COLOR_SWAP_16` in lv_conf.h. It internally swaps the 2 byte

tomtristan, 2019-03-02:
[image]
[image]
[image]
[image]

I attached several picture of my progress so far - demo seems to be working, but there are issues with colors overall and pixels around the letters.
The LTDC is initialized for ARGB8888, and the lv_conf color depth is 32(when changed to 16 everything goes downhill, very distorted everything), so the configure should be fine. To my eye it seems that some color pixels are swapped in the STM32F746 disco kit ported no os atollic and has an issue somewhere in the tft.c file. There exists a possibility that there is a bad solder on a certain pin, but the second revision of the pcb is arriving soon, if the problem exists after soldering and testing it, it most probably is software related.

kisvegabor, 2019-03-02:
@tomtristan We have just launched a [Forum](https://forum.littlevgl.com). This topic is more like a question than development so please open a topic on the forum for it.