lvgl/lvgl #290: How to customize my own object

szjdb, 2018-06-27:
Hi, kisvegabor,
 Many Thanks for your effort for the nice GUI. 
How to customize my own object ? I can not find any imformation about it on your website.

Best Regards, 
szjdb

kisvegabor, 2018-06-27:
Hi,

How do you mean "customization"? What are your goals?

szjdb, 2018-06-27:
Hi, kisvegabor,
Just like constructing a new button or meter , which have different style than the existing object. I read the code and find hard to understand the drawing of the object. There have not such similar one like button_draw() function inside the GUI lib. Please give me some advice!

Best Regards,
szjdb

BesitzeRuf, 2018-06-28:
Hi szjdb, 

did you already check files "lv_btn.c", "lv_btn.h" from /lvgl//lv_objx/ directory?

szjdb, 2018-06-29:
Hi, BesitzeRuf ,

Yes I have checked the object file like lv_btn.c  and found that there have not the detail drawing function for that object. To  customize my own object  , I have to draw them and process the message queue  by myself. Would your please give me more advice. 

Best Regards,
szjdb

BesitzeRuf, 2018-06-29:
For the "button"object there is no any draw function defined, because it is created from the "container" object. The button object in this library just changing its visual style depending on incoming signals. The container object is responsible for the some additional parameters like horizontal/vertical fit, but the container itself is created from the base object "obj", which contains "lv_obj_design" function which performs the "drawing tasks". 

So I think if you have some specific draw function and you can not use already built-in function, you can create you own. The best place to implement this function will be in the "lv_draw" folder. Then in your own object implement a "design" function, which will do some stuffs for you by your needs and call your "draw function" from lv_draw folder. Don't forget to assign the design function by "    lv_obj_set_design_func(your_own_obj_pointer, lv_your_own_obj_design);" ... Please see more also in lv_img.c file, which is the most specific and simply object. I think you know, there is also lv_objx_templ.c. with some "guide" in comment, which can help you too.

szjdb, 2018-06-29:
Hi, BesitzeRuf ,
Thanks a lot!
I will follow your advice. Actully I was going to implememnt the gui for my diy digital osocilloscope. It have a window for the realtime waveform and some control button with some selecting menu, like some commercial DSO. There have not similar object for the waveform display in the existing lib, so I was considering the customization. It is really a tough thing for me.
Best Regards,
szjdb

BesitzeRuf, 2018-06-29:
Aha, 
so for the oscilloscope I recommend you to check also system monitor demo (in [lv_examples branch/lv_apps/sysmon.c](https://github.com/littlevgl/lv_examples/tree/master/lv_apps/sysmon)

kisvegabor, 2018-06-29:
@BesitzeRuf  First of all thank you for explaining all this things. I'm so happy see people are learning LittelvGL and help each other.

@szjdb Speaking about an oscilloscope I suppose you would like to display points and/or lines. As it was already mentioned you can set a new design function for your object where you can draw the required points or line with the already existing  `lv_draw_rect` and `lv_draw_line` functions. In addition [lv_chart](https://littlevgl.com/object-types/chart-lv_chart)  object might be suitable for this purpose.

szjdb, 2018-06-30:
Hi, kisvegabor,
Many Thanks!
I would like to package the  waveform window to an object for easy resizing and moving . Due to the realtime refresh rate reaching 20-50HZ, I don't know if the lv_chart can fit it.  I have tracked the drawing of one button and it is likely to redraw the whole display area step by step  when update the button. So it is a little bit heavy duty to update the waveform than just refresh the curve itself. I am not clear if it is possible to manage the low level drawing rule by myself. Would like to have your advice! 

Best Regards,
szjdb

kisvegabor, 2018-07-01:
Hi,

Your welcome! :)

The library redraws the whole screen only at first time. After it only the changed areas should be redrawn. However, if an extreme number of areas are changed (more then `LV_INV_FIFO_SIZE` in lv_conf.h) in `LV_REFR_PERIOD` time  then the whole screen will be redrawn. If it's the case then maybe you update things to often or try to increase  `LV_INV_FIFO_SIZE`.

An `lv_chart` with a plain background (no gradient, no radius, no border, just a mono color rectangle) and lines with full opacity should be the fastest thing you can achieve without hacking the library.
Please test it and if it is too slow then we can figure out something else. :)

kisvegabor, 2018-09-18:
As there was no activity here for a while I close it to keep the issues clear.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-09-18 by kisvegabor