lvgl/lvgl #291: Problems adding a new font

hydra, 2018-06-29:
I wanted to use my own custom font.
I don't want to modify any of the existing library code, just my own source files that are contained in my git repo, not the lvgl submodule.

there is an issue with lv_style.c as it uses `LV_FONT_DEFAULT` which is defined in the lv_conf.h, but it's not possible to compile lv_style.c as it complains with:

```
lvgl/lv_core/lv_style.c: In function 'lv_style_init':
lvgl/lv_core/../../lv_conf.h:134:33: error: 'TopazPlus_a1200' undeclared (first use in this function)
 #define LV_FONT_DEFAULT        &TopazPlus_a1200
                                 ^
```
Adding `LV_FONT_DECLARE(TopazPlus_a1200);` into lv_conf.h doesn't work since the `LV_FONT_DECLARE` is not defined yet, it's only defined after `lv_font.h` is included, which in the case of `lv_style.c` is AFTER `lv_conf.h` is included.

I think there's an underlying dependency issue with `lv_font.h`, I don't think it should have font functions and declarations mixed in with actual fonts.  I think perhaps there should be a `lv_default_fonts.h` file that contains the 'FONT DECLARATION INCLUDES' section, and `lv_style.c` should likely include something like `lv_user_fonts.h` which can have your custom fonts.

kisvegabor, 2018-06-29:
Hi,

So far didn't tried to set a custom font as a default font and didn't realized it's an issue.

Based on your suggestion, for me,  it look more simple to add a define in `lv_conf.h` like:
`#define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(TopazPlus_a1200)` 

And this define should be "called" in the end of `lv_font.h`. This way it would be easy to make all custom font declarations global.

What do you think?

hydra, 2018-06-29:
I created a PR for you with my current solution, it's here: https://github.com/littlevgl/lvgl/pull/292

Since it's my first PR for this project please study it closely and let me know if you need changes.

hydra, 2018-06-29:
I don't think your solution is workable if you have more than one custom font to add.

I want to use 2 custom fonts and no default/built-in fonts.

kisvegabor, 2018-06-29:
You use a backslash to crate multi-line defines:

```c
#define LV_FONT_CUSTOM_DECLARE    LV_FONT_DECLARE(font1) \
				  LV_FONT_DECLARE(font2) \
				  LV_FONT_DECLARE(font3)
```

My concern about the external header file is the include path. This way the user need to place lv_user_fonts.h to a predefined location. I know that it's the case with lv_conf.h as well however I think it's better to minimize the external resources like these.

hydra, 2018-06-29:
Yeah, I thought about that, I will try it.  Agreed with regards to needing another file and the paths.

kisvegabor, 2018-06-29:
Okay, thanks. I tried it and it worked for me.

hydra, 2018-06-29:
tried what? the PR or the above?

kisvegabor, 2018-06-29:
The `LV_FONT_CUSTOM_DECLARE` thing. 
I reviewed the PR but didn't tried so far.

hydra, 2018-06-29:
I'm not sure exactly what you tried at this point, have you tried changing the value of 'LV_FONT_DEFAULT' to something else?  That is the main issue.

kisvegabor, 2018-06-29:
Sorry for being inaccurate. So I tried this in lv_conf.h:

```c
#define LV_FONT_DEFAULT        &synch     /*Always set a default font from the built-in fonts*/

#define USE_LV_FONT_DEJAVU_10              0
#define USE_LV_FONT_DEJAVU_10_LATIN_SUP    0
#define USE_LV_FONT_DEJAVU_10_CYRILLIC     0
#define USE_LV_FONT_SYMBOL_10              0

#define USE_LV_FONT_DEJAVU_20              4
#define USE_LV_FONT_DEJAVU_20_LATIN_SUP    4
#define USE_LV_FONT_DEJAVU_20_CYRILLIC     4
#define USE_LV_FONT_SYMBOL_20              4

#define USE_LV_FONT_DEJAVU_30              4
#define USE_LV_FONT_DEJAVU_30_LATIN_SUP    4
#define USE_LV_FONT_DEJAVU_30_CYRILLIC     0
#define USE_LV_FONT_SYMBOL_30              4

#define USE_LV_FONT_DEJAVU_40              4
#define USE_LV_FONT_DEJAVU_40_LATIN_SUP    4
#define USE_LV_FONT_DEJAVU_40_CYRILLIC     4
#define USE_LV_FONT_SYMBOL_40              4

#define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(synch) \
							   LV_FONT_DECLARE(uniscii) \
```

(`synch` and `uniscii` are my two custom fonts.)

And I added `LV_FONT_CUSTOM_DECLARE` to the end of lv_font.h.

hydra, 2018-06-29:
yes that works and can remove the additional lv_user_fonts.h file.  I will update my PR.

hydra, 2018-06-29:
out of curiosity, why are the `lv_fonts` located in `lv_misc` ?  why not just have them as siblings along side the other `lv_*` folders?

kisvegabor, 2018-06-29:
Awesome, thanks! Only one thing, please add the PR to the beta branch.

About `lv_fonts` folder:
I asked the same from myself one minute ago. :) 
Earlier `misc` was different repo and it contained general programs (like linked list, malloc, fifo and even fonts). When it was added to `lvgl` I kept the fonts in it. 
However, I don't see any reason to keep it in `lv_misc`. In the next release I will move it outside of `lv_misc`. Thanks for pointing this out!

kisvegabor, 2018-07-07:
Fixed in #292

Thank you for your help!

closed 2018-07-07 by kisvegabor