lvgl/lvgl #295: Something wrong with Online Image to C Array Converter?

chaoking888, 2018-07-02:
Hi gabor,

When using Online Image to C Array Converter to convert a 1024*600 24-bit jpg image  which is shown in the attach file to C array, the last part of the data is garbled character. 

Thanks!
[image]

[image]

chaoking888, 2018-07-02:
In other words, is there any restrictions while using Online Image to C Array Converter? For example, the size of the picture can't exceed a certain value?

kisvegabor, 2018-07-02:
Hi,

I tested it the PC simulator and it worked for me.
I attached a screen shoot when using the image and a "C" file which is the converted image. The "C" file different for you? Or does it work with the attached C file?

[img_conv_test.zip](https://github.com/littlevgl/lvgl/files/2156082/img_conv_test.zip)

chaoking888, 2018-07-03:
Thanks for your response. And I have tested this two pictures using Online Image to C Array Converter, the results show that the parsed data of "img_conv_test.png" is incorrect which is the same as I tested the 1024*600 24-bit jpg image before, while the picture "test.png" of a smaller size is converted correctly. 

I thought there might be related to the size of the converted file. The size of "img_conv_test.png" after the conversion reaches 44.87MB. So would you test "img_conv_test.png" with the Online way to see if we have the same situation when you are free? Thanks a lot!

[image]

chaoking888, 2018-07-04:
I seem to have found the reason which is related to the network. The size of the converted C array by Online method is about 44MB, while the size of the file I download is smaller than the 44MB even I have tried many times, so that the last part of the .c file is garbled character. Um...I have no other way of doing this for the time being. But thank you all the same!

kisvegabor, 2018-07-06:
It's very strange.
On my server, I increased the memory size dedicated to PHP scripts. Hopefully, it will help.

And the converter scrips are available here too: https://github.com/littlevgl/lv_utils
You can try to run the scrips offline in a terminal.

kisvegabor, 2018-10-07:
I new image converter is released. Hope your issue is solved in it.

If don't or you find other issues just reopen this issue.

closed 2018-10-07 by kisvegabor