lvgl/lvgl #297: Const correctness in LittlevGL

hydra, 2018-07-05:
I noted today, when using const appropriately in my own code that LittlevGL wasn't very 'const correct'.

https://isocpp.org/wiki/faq/const-correctness

It would be good to start using `const` as appropriate on new code and updating existing code to use it.

In embedded systems using `const` can really help the compiler optimize the code in addition to the other readability and developer friendly gains it provides.

hydra, 2018-07-05:
for instance:

`v_obj_t * lv_obj_create(lv_obj_t * parent, lv_obj_t *copy);`

should probably be

`v_obj_t * lv_obj_create(lv_obj_t * parent, const lv_obj_t * const copy);`

that is, `lv_obl_create` never changes the pointer to `copy` and never modifies that structure pointed to by `copy`

kisvegabor, 2018-07-06:
Hi,

I tried to consciously place the `const` tags but I forget it in create functions.

However, I never thought that `lv_obj_t * const copy` or `const lv_obj_t * const copy` might have relevance as the key thing is the data pointed by the pointer won't changed (indicated by`const lv_obj_t * copy`).

DeonMarais64, 2018-07-10:
Hi,

DeonMarais64, 2018-07-10:
Hi, 
IMHO Hydra has a point as a matter of principle.
In the first place, it would make for easier understanding of the behaviour of the function,
Secondly, it would allow for constant declarations of objects (in flash for example, even though that may not be currently supported) to be used as templates for copying instead of needing to obtain them from read/write memory.

kisvegabor, 2018-07-11:
Hi,

To store data in flash only `const lv_obj_t * copy` is required to show data the pointed data is constant. I totally agree with it, this approach should be followed everywhere. 

I have concerns in regard to  `const lv_obj_t * const copy` (second `const`). Can you mention some cases where it is useful? Or do you think it also improves "readability"?

DeonMarais64, 2018-07-11:
Hi,
I agree, the "second" const is largely meaningless when a pointer is passed as a function parameter.

kisvegabor, 2018-07-11:
Great! @hydra, what do you think?

If you find forgotten `const`s in the code, please mention them here.

hydra, 2018-07-11:
for me, any improvement in const usage is welcomed.  the sooner you apply it the easier it will be to maintain in the future for the library and the users of it.

DeonMarais64, 2018-07-12:
Hi,
```C
void lv_obj_align(lv_obj_t * obj,lv_obj_t * base, lv_align_t align, lv_coord_t x_mod, lv_coord_t y_mod);
```
Note the missing space after first comma.

All the lv_x_get_x functions which take lv_obj_t * arguments ?

I may be wrong, but it seems to me that correcting all instances may not be a trivial matter.

kisvegabor, 2018-07-12:
Hi,

You are right near all `lv_x_get_x` functions can be updated! Thank you!
`lv_obj_align` is also correct!

kisvegabor, 2018-07-30:
Hi,

I added `const`s to all:
- `lv_.._create()`
- `lv_..._get_...()` 
- and other places (e.g. `lv_obj_align`)

Probably still you will find some places where `const` can be added but I think we covered the bulk of it.
Hence I close this issue!

Thank you for the contribution!

closed 2018-07-30 by kisvegabor