lvgl/lvgl #298: Scrolling object function

Zaltora, 2018-07-06:
Can be useful to add  scrolling function (x/y) for each object that use this mechanics.
That will allow people to do automatic scrolling with anything  (e.g: timer, interrupt, uart, ...)
Very good for rotating texts / windows.

kisvegabor, 2018-08-04:
Hi,

I implemented the following functions to achieve this:
```c
lv_page_scroll_up(page);
lv_page_scroll_down(page);

lv_win_scroll_up(page);
lv_win_scroll_down(page);
```

I found that the other `lv_page` related objects already has some kind of navigation interface:
- lv_list: `lv_list_up/down`
- lv_ta: lv_ta_cursor_up/down/left/right
- lv_tabview: it craetes `lv_page`s

Can you test it?

Zaltora, 2018-08-05:
I have test it, seem to work. I just found that scrolling top behavior is different than scrolling bot. it is maybe me or the content. we don't see the bottom bar (padding/inner) ?
Top picture (start):
 [image]
Bot picture (end): 
[image]

Custom problem:
I want just show the Text  in the object Page. but i don't no style config to do for this i want remove all border. It is possible to do that ?
Code:
```c
    static lv_style_t style_sb;
    lv_style_copy(&style_sb, &lv_style_plain);
    style_sb.body.main_color = LV_COLOR_BLACK;
    style_sb.body.grad_color = LV_COLOR_BLACK;
    //
    style_sb.body.border.color = LV_COLOR_BLACK;
    style_sb.body.border.width = 0;
    style_sb.body.border.opa = LV_OPA_70;
    style_sb.body.border.part = LV_BORDER_NONE;
    //
    style_sb.body.padding.hor = 0;          /*Horizontal padding on the right*/
    style_sb.body.padding.inner = 0;        /*Scrollbar width*/
    style_sb.body.padding.ver = 0;

    style_sb.body.radius = LV_RADIUS_CIRCLE;
    style_sb.body.opa = LV_OPA_60;

    lv_obj_align(pxPrvArgs->apxObjects[2], pxPrvArgs->apxObjects[1], LV_ALIGN_OUT_BOTTOM_LEFT, 0, 0);
    lv_obj_set_size(pxPrvArgs->apxObjects[2], 128, 64);
    lv_page_set_style(pxPrvArgs->apxObjects[2], LV_PAGE_STYLE_SB, &style_sb);
    lv_page_set_sb_mode(pxPrvArgs->apxObjects[2] , LV_SB_MODE_OFF);
```
What config i miss ? i try to change  width, padding. Set value seems to change nothing...

Questions:
- Why each object need a specific function for scrolling actions ? can't be global "obj" API ? Maybe they are too much different for that.
- Page and Win can be multidimensional (x/y) (top/down/left/right) ? It is maybe not implemented.

Idea: 
- Multidimensional object like window/Page  will allow to create a virtual screen size. Can be an interested feature to expand screen size ? user use input/API/keypad to navigate.

thank for this feature :). Really nice for me !

Zaltora, 2018-08-06:
Question/idea:
- the srolling mouvement  size is hard-coded ?  Second parameter to select the scroll speed can be great ? e.g: `lv_page_scroll_down(page, LV_SCROLL_FAST);`
the scroll speed is a factor  `factor * screen size`

Zaltora, 2018-08-08:
Problem about the bottom of page is not a problem. My setting about height was wrong.
Ignore pictures :)

Zaltora, 2018-08-10:
**Answer:**
1. thank you ! it is work :)
2. Ok.
3. Really nice !
4. You are right lv_page do it already. I missed that. I was thinking page only work vertically.

**Feature/question:**
i like the fact to design GUI without specify "pixel argument". The big advantage is to keep the same GUI code when the HW is not the same. It is maybe a big new feature.
the goal of my idea is to replace all pixel entry by some enumeration or API that will compute what is needed.
e.g for page:
`lv_page_scroll_hor(page, lv_scroll_speed(LV_SLOW));`
The API will determine what is a slow scrolling with the width of the screen.
instead of :
`lv_page_scroll_hor(page, 5);`
It more like a utility API idea.

**Tip/memory:**
I recently change the way i declare `typedef enum` to save memory. i share it if you want use it. I don't know disadvantage yet (maybe speed?, not sur). a `typedef enum` for me is a int encoded on 4 byte.
current declaration:
```c
typedef enum
{
    LV_SB_MODE_OFF 	= 0x0,   	/*Never show scrollbars*/
    LV_SB_MODE_ON  	= 0x1,   	/*Always show scrollbars*/
    LV_SB_MODE_DRAG = 0x2,   	/*Show scrollbars when page is being dragged*/
    LV_SB_MODE_AUTO = 0x3,   	/*Show scrollbars when the scrollable container is large enough to be scrolled*/
    LV_SB_MODE_HIDE = 0x4,	 	/*Hide the scroll bar temporally*/
    LV_SB_MODE_UNHIDE = 0x5,	/*Unhide the previously hidden scrollbar. Recover it's type too*/
} lv_sb_mode_t;
```
new declaration:
```c
enum
{
    LV_SB_MODE_OFF 	= 0x0,   	/*Never show scrollbars*/
    LV_SB_MODE_ON  	= 0x1,   	/*Always show scrollbars*/
    LV_SB_MODE_DRAG = 0x2,   	/*Show scrollbars when page is being dragged*/
    LV_SB_MODE_AUTO = 0x3,   	/*Show scrollbars when the scrollable container is large enough to be scrolled*/
    LV_SB_MODE_HIDE = 0x4,	 	/*Hide the scroll bar temporally*/
    LV_SB_MODE_UNHIDE = 0x5,	/*Unhide the previously hidden scrollbar. Recover it's type too*/
};
typedef uint8_t lv_sb_mode_t;
```

The current enumeration will be 1 byte, whatever the compiler is. Enumeration size depend on how the compiler implemented it and what option are available to define the size. 
Just an idea too :)

kisvegabor, 2018-08-10:
**1-4:** 
Great :)

**Feature/question:**
If you want to move the GUI to other displays you can: 
* `lv_page_scroll_hor(page, LV_HOR_RES / 2)` 
* `lv_page_scroll_hor(page, lv_obj_get_width(page) / 2)`
* `lv_page_scroll_hor(page, MY_SCROLL_SIZE)`

For me, it seems a suitable way to use the current API. What do you think?

**Tip/memory**
Thanks for sharing this! I also noticed the wasting of memory if `typedef enum` is not used carefully. As LittlevGL usually uses the enums in structures I use it this way: https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_objx/lv_page.h#L65 It needs less memory but a little bit more difficult to read the code.

C47D, 2018-08-10:
I didn't knew that neat trick to reduce enums size @Zaltora 👍 , surfing the web i found this compiler flags [-fshort-enums, -fno-short-enums](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0774b/chr1411640303038.html), those flags tell the compiler what size the enums will be (available on gcc as well).

One thing to notice:

> The -fshort-enums option can improve memory usage, but might reduce performance because narrow memory accesses can be less efficient than full register-width accesses.

Zaltora, 2018-08-10:
@kisvegabor : Yes you are right. I need think more before talk xD. I just wondering how make portable the shift parameters of a function like this `lv_obj_align(obj1, obj2, LV_ALIGN_TYPE, x_shift, y_shift)` .
Use the screnn size solve everything i guess :) 

**Bit field:**
I learn something important about it recently: [source](http://www.indiabix.com/technical/c/bits-and-bytes/)

> Bit fields are not portable. Because bit fields cannot span machine words, and because the number of bits in a machine word is different on different machines, a particular program using bit fields might not even compile on a particular machine.
>  Assuming that your program does compile, the order in which bits are assigned to bit fields is not defined. Therefore, different compilers, or even different versions of the same compiler, could produce code that would not work properly on data generated by compiled older code. Stay away from using bit fields, except in cases in which the machine can directly address bits in memory and the compiler can generate code to take advantage of it and the increase in speed to be gained would be essential to the operation of the program.

Question: use bit field mean that compiler will use bit operations. It is slower than use an variable like a byte ?

@C47D 

> The -fshort-enums option can improve memory usage, but might reduce performance because narrow memory accesses can be less efficient than full register-width accesses.

The problem is the same everywhere you use a variable smaller or larger than the size of the architecture.
To go faster (less memory for app ), we can use `int` but that variable can be too small. Other solution is to use `uint_fast32_t` definition (or equivalent 8,16,..) that allow  minimum 32 bit but can be 64 bit if needed by the architecture .
To allow this optimisation for the user, lvgl need maybe to create thiw own variable in `lv_conf.h`
e.g for area:
```c
typedef  int16_t lv_coord_t;  //< currently 
typedef  int_fast16_t  lv_coord_t;  //< faster size variable possible with minimum of 16 bit.
```

kisvegabor, 2018-08-18:
As the original issue is solved  I close it now.
If you have ideas or remarks related to the bitfields, please open a new issue for it.

closed 2018-08-18 by kisvegabor