lvgl/lvgl #300: Application Managed VDB Option

AGlass0fMilk, 2018-07-06:
Hello!

I recently ported LittleVGL over to mbed-os on a Nordic nRF52832. I am using it with a monochrome display. I discovered a caveat that I have been discussing with Gabor over email:

The issue and my proposal:

>Hi Gabor, 
>I just got the nRF52832 + Mbed-OS port working with a monochrome OLED display. I’m waiting for a color OLED display to come to try it with that.

>LittleVGL is incredibly flexible! I’ll try to write a guide as I go along. Porting LittleVGL to mbed was actually fairly simple.

>I had a thought while working with the monochrome display:

>My understanding of the VDB system used by LittleVGL is that it creates its own display buffer with element sizes based on the configured color-depth. For monochrome, the element size is one byte.

>The display I was using is only 128 X 64, and represents each pixel using one bit. So the whole frame buffer would only be 1kB (easily handled by most microcontrollers). However, the way I understand it, if I used the VDB feature of LittleVGL it would need to either be a partial VDB or it would allocate 8kB for the full display buffer. I also found it necessary to convert the VDB data to the proper format for the display before sending it over I2C. This makes it harder to use a DMA to speed up the transfer, and the only solution I could see would be to maintain an additional display buffer with the proper format.

>I was thinking maybe there is a way to provide LittleVGL with a buffer allocated and maintained by the application and provide a common interface to LittleVGL to modify the buffer. This allows the application to define how the buffer is formatted, enabling direct transfers with a DMA while minimizing redundant memory usage and wasteful conversion procedures.

>Is there something like this for LittleVGL currently? What are your thoughts?
>Best,
>George

Gabor's response:

>Hi George,

>I happily hear that! And thank for the writing of guide in advance.

>About VDB Your observation is correct. There is an overhead using monochrome displays. With some guys, we had a discussion about bit-based VDB support however it won't solve the storage of pixels in every specific way (bytes means columns or rows etc.)

>But your proposal sounds great. An interface like lv_px_set(buf, buf_w, x, y, color) can solve it. It would also make possible the use of 2 bit and 4 bit based VDB.

>There is a rounding function interface in LittelvGL which allows you, for example, to round all y coordinates to 8.

>What do you think?

>Best regards,
>Gabor

I am thinking that an "application-managed VDB" could be a configuration option. The user must then allocate their own display buffer and provide functions for reading/writing to it. This will allow users to optimize their memory usage and take advantage of DMA transfers by formatting the buffer as LittleVGL accesses it rather than right before a transfer. It will also drastically reduce memory overhead for monochrome/grayscale displays and overall make the library more flexible and compatible with a broader set of displays.

That being said, I just started using LittleVGL and am not very familiar with its internal workings. I'm looking at modifications/additions to [lv_core/lv_vdb.h/.c](https://github.com/littlevgl/lvgl/blob/fc05fad2770d62de8085c88e1a2b15c313576909/lv_core/lv_vdb.h) for where to put the app-managed VDB API and [lv_draw/lv_draw_vbasic.h/.c](https://github.com/littlevgl/lvgl/blob/f5d4bf8fc47a3fc7fdca4b29cf156fa768084643/lv_draw/lv_draw_vbasic.h) for where to modify how LittleVGL accesses the VDB in this scheme.

Any thoughts/suggestions?

kisvegabor, 2018-07-07:
Hi,

Thank you for opening the issue!

As you can see my idea is an interface like `lv_vdb_set_px(buf, buf_w, x, y, color)` which can be used in the functions of `lv_draw_vbasic.c` if set.

What do you think?

I remember @Zaltora and @upbeat27 were interested in monochrome displays.

Zaltora, 2018-07-11:
It should not be 2 function? it just learn how to write in a memory?
It is a good idea to outsource the working memory of lvlg. That what we talking about ?
I didn't take a look everywhere in lvgl core but you don't need to read VDB to do advanced features ?
Why just set "write function" ?

At the initialization of lvgl, it would be necessary to register:
-the memory buffer address (usefull with DMA ?)
-a way to write in VDB.
-a way to read VDB ??

kisvegabor, 2018-07-11:
You are right! A read function is required as well! How did I forget it?

But maybe these "functions" should be macros to avoid the performance loss because of a lot of function calls.

Zahriman, 2018-07-12:
Hi all,

Monochrome display here too and one byte is one column (not a row). While it is not that hard to write the ```disp_flush``` callback, it needs some conversion before writting to the i2c and then no DMA; and also memory overhead not to mention it.
Outsourcing the VDB management using read/write seems a good idea. Like @AGlass0fMilk I maintain an additional display buffer to retrieve data when not all of the 8 bits needs to be updated. I am going to investigate on the rounding function you mentioned and the area invalidation in multiple of 8 pixels as discussed in #84. I am not already familiar with all that lv_ stuff.

May I suggest inline functions instead of macros? Still 'C' functions but no function call overhead. They could be defined in the ```lv_conf.h``` inside a ```USE_MANAGED_VDB``` macro. If the macro is not defined, the VDB management then fallbacks to its actual behavior. This would allow custom management when required but do not force it.

What do you think?

Zahriman, 2018-07-12:
Using the rounding function did simplify the ```disp_flush``` callback implementation and I do not need to maintain anymore a supplementary display buffer.

kisvegabor, 2018-07-13:
@Zahriman The `inline` function idea sounds well! In this case, the VDB manager functions need to defined to not use any `lv_.._t` types because they are not visible in lv_conf.h. However, it seems manageable.
Good to hear the rounding function works!

As there will a lot of updates in drawing algorithms I add this improvement to the milestone of the next release.

AGlass0fMilk, 2018-07-13:
What about _weak_ inline functions? That way the application can just override them by redefining them.

kisvegabor, 2018-07-13:
@AGlass0fMilk Yes, as `weak` symbol are not in standard C we shouldn't use them to keep platform independency.

BesitzeRuf, 2018-07-13:
Hi all, can we declare some macro attribute for these functions? Than based on compiler everybody can define this macro as weak, __weak, etc. ?

kisvegabor, 2018-07-13:
@BesitzeRuf  Great idea! This way, `inline`, `weak` etc. everything is optional and possible!

AGlass0fMilk, 2018-07-19:
Hey Gabor, is there any feature branch for this yet? I'm getting some hardware in on Monday that may need this for optimal efficiency so I may start working on it. We can collaborate on this feature as it develops.

kisvegabor, 2018-07-19:
Awesome! 
I just created the `app_vdb` branch from the current `dev-5.2`. 
Basically, all writing of VDB happens in `lv_draw_vbasic.c` so hopefully the update will affect only this file.

kisvegabor, 2018-07-30:
@AGlass0fMilk is there any news about this topic?

kisvegabor, 2018-08-01:
@AGlass0fMilk Okay, please let me know if you start it. Probably, I will work on it soon and it wouldn't be good to do the same thing twice. :)

AGlass0fMilk, 2018-08-01:
@kisvegabor I'll start looking into it more today.

kisvegabor, 2018-08-01:
Awesome! :)

AGlass0fMilk, 2018-08-01:
After looking at associated code, I'm not sure how to proceed. It seems the size of lv_color_t is closely coupled with how the VDB is read/written to. We would also need to modify the lv_core/lv_vdb.h and lv_core/lv_vdb.c so that the VDB is not statically allocated (assuming LV_VDB_SIZE will be non-zero when using application managed VDB).

How could we decouple lv_color_t from reading/writing to the VDB without a big performance hit?

I don't know if I'll have the time in the coming days to work on this @kisvegabor, you may want to go ahead and get started. I may help out if you need it once we figure out how to proceed.

kisvegabor, 2018-08-01:
I didn't check it deeply so far and it's possible that it won't be so simple as it seemed. :)
As I'm thinking about it `lv_refr.c` need to be updated too because it makes some calculations with VDB size.

Calculating every bit of VDB (shift, multiply, mod operations) would decrease the performance. Fortunately, in `lv_vfill` and in some cases of `lv_vmap` the pixels are written sequentially. So it might be useful to have an `init` and a `set_next` function because it's easier to go to the next bit than calculating the position again.

Hopefully, at the weekend I can check!

kisvegabor, 2018-08-04:
Hi,

I managed to implement it but it really was quite comlicated. Finally I added a new function pointer to the display driver. (In v6.0 there will multi display support and this feature is specific the display and the library irself).  

I test it in the PC simulator like this:

1. In lv_conf.h there should be a new define: `#define LV_VDB_PX_BPP		XXX`. By default, it should be `LV_COLOR_SIZE` but here you an modify the bpp of the VDB. Here is my lv_conf.h: [lv_conf.h.zip](https://github.com/littlevgl/lvgl/files/2259352/lv_conf.h.zip)

2. Created three custom VDB write functions:
```c
/* Just add the same colored pixel (Set LV_VDB_PX_BPP to LV_COLOR_SIZE)*/
void lv_vdb_wr_same(uint8_t * buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
	lv_color_t * buf_c = (lv_color_t *) buf;
	buf_c += buf_w * y;
	buf_c += x;

	if(opa == LV_OPA_COVER) *buf_c = color;
	else *buf_c = lv_color_mix(color, *buf_c, opa);
}

/*Black/White. Store 8 pixel in one byte. Bytes are mapped vertically.  (Set LV_VDB_PX_BPP to 1)*/
void lv_vdb_wr_bpp1(uint8_t * buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
	buf += buf_w * (y >> 3) + x;
	if(lv_color_brightness(color) > 128) (*buf) |= (1 << (y % 8));
	else (*buf) &= ~(1 << (y % 8));
}

/*Store the brightness on 1 byte to make the screen grayscale  (Set LV_VDB_PX_BPP to 8)*/
void lv_vdb_wr_gray(uint8_t * buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
	buf += buf_w * y;
	buf += x;
	if(opa == LV_OPA_COVER) *buf = lv_color_brightness(color);
	else *buf = (lv_color_brightness(color) * opa + (*buf) * (LV_OPA_COVER - opa)) >> 8;
}
```

3. Registered one of the above functions in the display driver:
```c
    monitor_init();
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);            /*Basic initialization*/
    disp_drv.disp_flush = monitor_flush;
    disp_drv.vdb_wr = lv_vdb_wr_gray;  
    lv_disp_drv_register(&disp_drv);
```

4. Added three parts in `monitor.c` according to the three functions above (comment out the 2 unused)
```c
void monitor_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    /*Return if the area is out the screen*/
    if(x2 < 0 || y2 < 0 || x1 > MONITOR_HOR_RES - 1 || y1 > MONITOR_VER_RES - 1) {
        lv_flush_ready();
        return;
    }
    int32_t x;
	int32_t y;
	uint32_t buf_x = 0;
	uint32_t buf_y = 0;
	uint8_t * buf8 = (uint8_t *) color_p;
	lv_coord_t buf_w = x2 - x1 + 1;

//	/*lv_vdb_wr_bpp1*/
//	for(y = y1, buf_y = 0; y <= y2; y++, buf_y++) {
//		for(x = x1, buf_x = 0; x <= x2; x++, buf_x++) {
//			lv_color_t c;
//			if(buf8[buf_w * (buf_y >> 3) + buf_x] & (1 << (buf_y % 8))) c = LV_COLOR_WHITE;
//			else c = LV_COLOR_BLACK;
//
//			if(c.red) tft_fb[y * MONITOR_HOR_RES + x] = LV_COLOR_WHITE.full;
//			else tft_fb[y * MONITOR_HOR_RES + x] = LV_COLOR_BLACK.full;
//
//
//		}
//	}

	/*lv_vdb_wr_gray*/
	for(y = y1, buf_y = 0; y <= y2; y++, buf_y++) {
		for(x = x1, buf_x = 0; x <= x2; x++, buf_x++) {
			lv_color_t c;
			uint8_t tmp = buf8[buf_w * buf_y + buf_x];
			c = LV_COLOR_MAKE(tmp, tmp ,tmp);
			tft_fb[y * MONITOR_HOR_RES + x] = c.full;
		}
	}

//	/*lv_vdb_wr_same*/
//#if LV_COLOR_DEPTH != 24
//    int32_t x;
//    for(y = y1; y <= y2; y++) {
//        for(x = x1; x <= x2; x++) {
//            tft_fb[y * MONITOR_HOR_RES + x] = lv_color_to24(*color_p);
//            color_p++;
//        }
//
//    }
//#else
//    uint32_t w = x2 - x1 + 1;
//    for(y = y1; y <= y2; y++) {
//        memcpy(&tft_fb[y * MONITOR_HOR_RES + x1], color_p, w * sizeof(lv_color_t));
//
//        color_p += w;
//    }
//#endif

    /*Don't comment out below this*/
    sdl_refr_qry = true;
    /*IMPORTANT! It must be called to tell the system the flush is ready*/
    lv_flush_ready();

}
```

5. I added a "rounder" function to simulate a monochrome display where the bytes are mapped vertically.
```c
void rounder(lv_area_t *a)
{
	/*Make the area bigger in y*/
	a->y1 = a->y1 & ~(0x7);
	a->y2 = a->y2 |  (0x7);
}

...

    lv_refr_set_round_cb(rounder);

```

6. And got this:
[image]

Please test it if you have time and tell your remarks!

Krastanov, 2018-08-26:
Found it, https://github.com/littlevgl/lvgl/blob/app_vdb/lv_hal/lv_hal_disp.h in app_vdb

Krastanov, 2018-08-26:
For screens where I need 1bpp **horizontally** packaged, am I right to assume that the code should look like this:

```
void lv_vdb_wr_1bpp_horiz(uint8_t * buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
	lv_color_t * buf_c = (lv_color_t *) buf;
	buf_c += buf_w * y;
	buf_c += x/8;
	if(lv_color_brightness(color) > 128) (*buf) |= (1 << (x % 8));
	else (*buf) &= ~(1 << (x % 8));
}
```

kisvegabor, 2018-08-26:
@Krastanov 
At first look, your code it seems correct.

The `app_vdb` branch is already merged to `dev-5.2`. Please use the `dev-5.2` instead (I just forget to delete `app_vdb`)

Krastanov, 2018-08-26:
Tested on SHARP memory lcd ultralow power screens and it works wonderfully on dev-5.2. There was a stupid mistake in my code above (forget to divide buf_w by 8), but the following works fine:

```
void lv_vdb_wr_1bpp_horiz(uint8_t * buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
    buf += buf_w/8 * y;
    buf += x/8;
    if(lv_color_brightness(color) > 128) {(*buf) |= (1 << (7 - x % 8));}
    else {(*buf) &= ~(1 << (7 - x % 8));}
}

void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    display_flush(y1, y2, (uint8_t *)color_p);
    lv_flush_ready();
}

void rounder(lv_area_t *a)
{
	a->x1 = 0;
	a->x2 = LCD_XRES-1;
}
```

kisvegabor, 2018-08-27:
Awesome! 
Just out of curiosity, how big VDB do you use?

Krastanov, 2018-08-27:
Here is the config I used:

```
#define LV_MEM_SIZE    (10U * 1024U)
#define LV_HOR_RES          (128)
#define LV_VER_RES          (128)
#define LV_VDB_SIZE         (16 * LV_HOR_RES)
#define LV_COLOR_DEPTH     1
```

kisvegabor, 2018-08-28:
It looks good to me!

As you successfully tested this feature I close this issue. If you find related issues feel free to reopen it! 

Thank you for the ideas, contribution and testes!

closed 2018-08-28 by kisvegabor