lvgl/lvgl #302: child lv_obj ignores parent's size and inner padding and sets WxH to hardcoded values?

aenchevich, 2018-07-08:
At present, inside lv_obj_create we have:

```
        /*Set coordinates left top corner of parent*/
        new_obj->coords.x1 = parent->coords.x1;
        new_obj->coords.y1 = parent->coords.y1;
        new_obj->coords.x2 = parent->coords.x1 +
                             LV_OBJ_DEF_WIDTH;
        new_obj->coords.y2 = parent->coords.y1 +
                             LV_OBJ_DEF_HEIGHT;
```

So the new object ignores parent's area available for drawing. I see 2 issues here:

A. New object's x1 and y1 ignore the inner padding of the parent. This will result in draws insider the inner border of the parent if parent_style->body.padding.inner is set (non-0).

B. New object's x2 and y2 ignore the W and H of the parent. Depending on the actual size of the parent and his inner padding value, this may result in draws inside the inner border of the parent, and, even worse (if the object's W or H is greater than the W or H of the parent) - in draws outside the parent's client area!

I propose this be fixed via:

```
    lv_coord_t parent_inner_padding = 0;
    lv_style_t * parent_style = lv_obj_get_style(parent);

    if(parent_style)
    {
        parent_inner_padding = parent_style->body.padding.inner;
    }

    /*Set coordinates left top corner of parent*/
    new_obj->coords.x1 = parent->coords.x1 + parent_inner_padding;
    new_obj->coords.y1 = parent->coords.y1 + parent_inner_padding;
    new_obj->coords.x2 = parent->coords.x2 - parent_inner_padding;
    new_obj->coords.y2 = parent->coords.y2 - parent_inner_padding;
```

This will result in the object occupying the entire client rect of the parent by default. Yes, in most cases this is not what the caller wants, but neither is all objects to be set to LV_OBJ_DEF_WIDTH x LV_OBJ_DEF_HEIGHT. I think that when creating new objects, the caller should know their dimensions and set them explicitly, instead of relying on some hard coded values, deep inside the library, and completely out of context of things like theme/style used, etc.

kisvegabor, 2018-07-10:
Hi,

Let's extend the question to every default value. E.g.
- Lables are created with "Text" default text
- Drop down list are created with: "Option1, Option2, Option3" options
- Buttons also have hardcoded default size which is a reasonable button size
- etc

The purpose of these default values is to after just create the objects show something which looks like the object is intended to look like. (E.g. an empty label after creating might be confusing)

What do you think? Are they still reasonable or are there some considerations to take into account?

Using padding for initial position and size might be useful only if the parent also uses the paddings. But for example, the images are not using it.

aenchevich, 2018-07-11:
Hi Gabor

We have discussed many things in the past and often have found new ideas in each other's suggestions, so I am really sorry to tell you that in this particular case I can not agree with anything you're saying... :)

--> Regarding the default string values for Text, Option etc., I don't think that this is a very valid comparison to the problem I am describing, because:
1. Setting Text or Option1 does not create potential bugs, whereas setting the object default WxH to LV_OBJ_DEF_WIDTH/HEIGHT could.
2. Setting these text values does not violate the rules of the lvgl library. And one of these rules (if I understand it correctly?) is that the actual area available to the children of the object must take into account the padding of the parent (more on padding in the last paragraph).

--> Regarding the button (or ANY object for that matter...) using a hardcoded default reasonable size:
1. I do not see the lv_btn constructor setting the button size in any way? I see that it calls the lv_cont constructor, but inside it I also did not see the size being set. Then lv_cont calls lv_obj_create so really it comes down to lv_obj (as far as I could see). I looked on branches beta and master.. maybe it is on some other branch?
2. I do not agree that an object itself can determine what is a "reasonable" size. Depending on the context a resonable size for a button can be a VERY different thing.
   The only component which could have knowledge about the reasonable size for a given GUI element, IMO, is the theme -  because the theme represents the Graphics Designer vision of the GUI as a whole and hence can "know" that in this type of theme/style a button should look like this.
   However ... 
... default sizes are not currently a part of lv_style (and consequently not part of the theme).
**Maybe they should be**? If they were, then I would agree completely that some reasonable default value, based on the theme, should be used when creating an object. And it should come from the theme or style of the parent.

--> Regarding creating an object with default values so that the object looks the way it is intended to look like:
This may be reasonable if we are talking about some demo software, e.g. in the context of the pc_simulator.
In the case of a real software which is used in different real world applications I think that this approach does more harm than good:
1. It is "philosophically" incorrect because it move the knowledge about the GUI look-and-feel from the theme/style to the private code of the object (and not even every object - but the base of all objects!).
2. It will result in unnecessary loss of memory due to some hardcoded const text strings being added to the app/lib (because in reality people will not use Option1, Option2..); 
and unnecessary waste of CPU cycles (to strcpy these strings to objects, when later they will be replaced by real strings).
3. It teaches people to be lazy by relying on some default values and actually creates a "ticking time bomb" situation. First I say  "Hey I don't need to set anything, it just works, and looks great!". 
But then I decide to make my own theme...  and the default values do not work anymore... and there is frustration because   a) it does not look great anymore. and    b) now I have to go and set all the values inmy code properly (something which I should have done in the beginning)

--> Regarding "padding ... useful only if the parent also uses the paddings".
Maybe I don't understand how paddings, margins and borders work in lvgl? If yes - my apologies, and if there is any docs on this - please let me know wher I can read about this?

Looking at the code I got the impression that lvgl's paddings work in the standard accepted way which is like the HTML DOM model and the CSS Box model (https://www.w3schools.com/css/css_boxmodel.asp)?

If the above is true then there are no objects which are "not using padding". There are objects with 0-padding, yes, and that's not a problem because my proposed change accounts for this possibility (x1 + 0 = x1).

kisvegabor, 2018-07-11:
Hi,

Not agreeing is great because this way we can improve things :)

**Styles:**
As you can see the styles describes a lot of thing at the same time. (similarly to CSS) These are general properties and every object picks what it needs. E.g. `text.font` is not used by a line.  Moreover, a container uses paddings ONLY it the `fit` or `layout` is enabled. Else it's just a simple rectangle. However, once `fit` is enabled the container will fit to involve all children and will keep padding distance. Or when the `layout` is enabled the padding is taken into account when positioning the children. 
Comparing LittlevGL's padding to CSS's padding the main difference is near all HTML element can use the padding. In LittlevGL only those who have something to deal with it. 
But there are similarities as well: e.g. in CSS the `padding` is not inherited by default (https://www.w3schools.com/cssref/pr_padding.asp) but the `color` (text color) does. So a `div` doesn't care about `color` but a label on it inherits and uses the `color`. It's similar in LittelvGL too but for performance reasons, only the whole style can be inherited and not single properties. 
It must be noted CSS is much more complex then LittelvGL's style system (which is designed for small, embedded systems). 
I'm totally open to any constructive ideas to make the style system more flexible if we can keep it performance-wise.

**Default values:**
You are right about "Text" and "Option1, Option2" ... default values have a negative effect on memory footprint and performance. I agree it should have a define in lv_conf.h to enable/disable these initial settings. While playing with the library it's better to have some init values but it's better to avoid them later when you overwrite them anyway. 

**About basic size:**
As we can't guess what user wants, so we can't set the size correctly during the creation of the object. As size must be set to a value, in my opinion,  we can only set a size which is not too small, not too big and just tries to not look ugly. That's why I used LV_DPI to create the objects. This why the result real size (in millimeter on the screen) is known and the screen size and resolution won't affect it.

hydra, 2018-07-12:
Just chiming in on this.

I'm not a fan of any of the default width/height/labels.  I would prefer is the default size for everything is 0, and the default text for anything is an empty/null string.

My thought is that the user should be specifying these values.

Today the default label value of "Text" caused me an issue with my actual text not displaying correctly, see #310 and I had to break out the debugger to find out what was going on.

hydra, 2018-07-12:
@aenchevich I don't like the idea of width/height being inside a style at all.  CSS uses min-width and min-height which sound a little more reasonable for a _style_

@kisvegabor and @aenchevich With regards to padding, in CSS there is the MARGIN and PADDING terms which are well understood.  In CSS you have a left/right/top/bottom MARGIN and left/right/top/bottom padding.  In LVGL the terms `ver`, `hor` and `inner` left me digging around in the codebase and I still don't think I understand when, where and why they are used.  `ver` and `hor` seems like MARGIN specifications, while `inner` seems like a PADDING specification that applies vertically and horizontally, but from the variable names themselves you can't tell, you have to read the code that uses them.

aenchevich, 2018-07-13:
@hydra - Regarding the variable names in LvGL: yes initially I also had some difficulties understanding which is which. After reading the code and after Gabor's clarification above ("it is enabled the container will fit to involve all children and will keep padding distance. Or when the layout is enabled the padding is taken into account when positioning the children.") it is more clear to me now.

I agree that using the well-known names for these values  (margin and padding) would probably be better but at present it may not be a good option to change them - too many noise changes in all branches... I'd say - Gabor's call :)

Regarding the default values: 
"_I don't like the idea of width/height being inside a style at all_"  ... ummm... ok... why? If you would like to have a discussion with the hopes of arriving at the best solution for this library and the people who will be using it - can you please specify WHY you don't think it is a good idea.

I *personally* also prefer 'null' values as default and the users to set them themselves. However there may be cases (especially when themes are involved) where these default values can be very useful and the good looks of the GUI would depend on certain values be used for certain elements. This was the reason why I suggested that default initial values be part of the '.body' structure.

@kisvegabor and @hydra Regarding the parallels which we all drew here to the CSS model:
 - Just as I was not very clear about how inner, ver and hor work in lvgl, I am also not very clear how border width works in lvgl. Gabor, could you please explain, or point me to some docs?

In that link I gave ([CSS box model](https://www.w3schools.com/css/css_boxmodel.asp)) there is an example about an object which is specified to have width 320 px and in reality becomes 350 px wide because the widths of margin, border AND(!) padding are added to 320. I personally find this confusing too (why add padding? it should be on the inside of the object!) but oh, well - that's what the w3c consortium decided.
How does it work in lvgl?

 - Gabor - I did NOT suggest that the object inherits parent's padding. I said that if the parent has non-0 padding the child should respect it and position itself at x1+padding, y1+padding.

kisvegabor, 2018-07-13:
**CSS boxing model**
Now LittelvGL didn't follow the CSS boxing model. Hor/Ver padding means the object should keep that distance from the children. It takes effect only in certain modes (`fit`, `layout`, etc.). Inner padding means the distance between the children on the parent.  True story, it's limited in some aspects:
- no left/right/top/bottom padding, however, it's intentional to save memory
- no separate margin and padding, which really might be confusing  and also limited

Now I'm thinking about the possibilities of following the CSS boxing model. So:
1. Object size = `core_size` + `padding` + `margin`
2. Object draw area: `core_size` + `padding`
3. Children can be only on  `core_area`
4. The border is now included to `core_area`. (Not like in CSS where border width is added to it). I feel it reasonable because this way a translucent border can be applied which look well :)
5. Left/Right/Top/Bottom padding and margin can be introduced as it's only 10 extra bytes/style (8 attributes vs 3)
6. It would be nice to make possible to overwrite padding and margin defined in the style to enable individual adjustment of objects.

It really looks simple and easy to understand. This way all object can use padding and margin.

It's a very interesting topic! I'm waiting a lot to hear you thoughts!

**Default size**
On the one hand, I feel inflexible to include size in styles. For example, a progress bar (lv_bar) and a chart (lv_chart) can have the same default style but they need significantly different default size. See their example photos here: https://littlevgl.com/object-types

On the other hand, in some cases it can be useful:
- Size of control buttons on lv_window
- Knob size of lv_slider
- Indicator size of lv_tabview
- The middle point of lv_gauge

So for me, it doesn't seem suitable to set the default size of a window and checkbox too because they look very different. But can be very useful to set the size of the "internal" elements of an object.

aenchevich, 2018-07-17:
@all - Firstly - a visual example so it's easier to discuss  :)
[image]

aenchevich, 2018-07-17:
Based on the above I think the sizes should be:

1.    Total object size: core_size + border + margin
2.    Object draw area: core_size - padding
    Children can be only on core_area
3.    lv_obj_set_size should set the total object size (the area with the green-yellow pattern)
4.    I did not understand your rationale here... I think that under no circumstances should the border go outside the total are of the object (e.g. even if margin is 0 and border is 10). This is the same as what was said in 1.
  Regarding "Left/Right/Top/Bottom padding and margin can be introduced" ... yes, sounds good!

  Regarding: "It would be nice to [...] overwrite padding and margin defined in the style to enable individual adjustment of objects."  - Absolutely! With all of the fields of the structs public nothing stops you from setting them of course... but maybe some setter/getter functions would be best.

NOTE: What I am suggesting is not blindly following the CSS model. 
I think that for lvgl, knowing the TOTAL area which the object may occupy is important. Therefore I am taking into account all the elements of the CSS model, but saying "we are calculating the offsets from the outside - in. We determine the total size/area of the object, then we step 25 pixels inside to respect the margin, then we draw a border 15 pixels wide, then we step in another 5 px to provide padding between our object and the children."

aenchevich, 2018-07-17:
@hydra - Ok I see what you're saying. But in reality the code will probably look something like: 

```
o = lv_something_create( parent, copy );      // this creates an object with default values
lv_obj_set_style( o, some_predefined_style ); // this is optional - if you want to change defaults
```

 - OR -

```
o = lv_something_create( parent, copy );      // this creates an object with default values
// set your_custom_style values however you like, e.g.
your_custom_style.body.padding.inner = 5;
lv_obj_set_style( o, your_custom_style ); 
```
So really the logic is not "if this then this unless this..." 
Its more like "When creating the object, use some default values, which you can look up in the style definition. Later you can overwrite them if you like."

kisvegabor, 2018-07-18:
Thanks for the image. It will be really easier this way :)
I also added a simple image to demonstrate something:

```html
<html>

<head>
<style>
p{
 background-color:red;
 width:100px;
 padding: 10px;
 margin:20px;
 border: dotted 5px blue;
}
</style>
</head>

<body>
<p>Some text</p>
</body>

</html>
```
[image]

[image]

Earlier I not defined the terms correctly. It seems we think the same but my terms were confusing:
- core size: blue in the box model. In CSS the children objects can be only here.
- draw area: where the object will draw itself (red rectangle)

So in the example above `core size` became 100 px because of `width` in CSS. The `draw area`  is `core_size` + `padding` + `border`.
The dotted border (the idea is taken from your image) shows well that the element is drawn under the border in CSS too. Just the size of the `draw area`  needs to be increased by the border width. It's reasonable because in my previous view a 20 px border could cover the children which is not good.

However - in my opinion - unlike CSS, `lv_obj_set_width()` should set `core_size` + `padding` + `border` (so the draw area). Why? The user thinks: "I want a 100 px wide button." This way he will get it. Let say he wants to set some extra space for the content on the left. To do this he can increase `padding-left` of the button and the button's size will remain.  For me, it seems more simple. 
`margin` comes to play only in case of layout and alignments. And the `margin` area is always transparent. It would be strange to me if I decreased the spacing between to buttons (reduce `margin`) and I would get larger buttons to keep the total size.

aenchevich, 2018-07-20:
Ok so in the box model example you gave there is 4 areas (BLUE, GREEN, YELLOW, ORANGE) and 4 rectangles (BLUE, BLUE+GREEN, BLUE+GREEN+YELLOW, and finally the big one with all 4)  let's define what we call those areas which are important and what they do? 

If I understand correctly, your idea is:

Area BLUE - "core_size", or "client [draw] area" (because this is the area which a container provides to its children to place and draw themselves. So if a child obj is a 1x1 pixel object which wants to go to coords (0,0) of the parent it will go to the upper left corner of the blue rect). The oibject paints this area with the background "brush" (color, pattern, image) before drawing the children.

Area GREEN - "padding" - The Parent paints this area with the background "brush" (color, pattern, image) before drawing the children. Nothing else draws there.

Area YELLOW - "border" - The Parent paints this area with the background "brush" (color, pattern, image) after drawing the background. The background shall be visible behind the border if the border has transparent areas.

Area ORANGE - "margin"  - The Parent paints this area with ...? the margin "brush"? Or always leaves it transparent? Is this where the shadow of the object is drawn?
 - - - 
Rect BLUE+GR: not important on its own

Rect BLUE+GRN+YELW: "draw_area"  in your terminology. Will be painted with the backgnd, then with the border. **lv_obj_set_size()** sets the size of this area (except when the parent container is smaller than this... some discussion on this below). If the object wants to have a border then the border must be drawn on the **inside** of this area.

Rect BLUE+GRN+YELW+ORNG:  ??? 
This last one is somewhat unclear. I, personally, completely agree with your logic about the two buttons. 
However: 
 - is the margin always transparent? If not then on object invalidate *this* must be the area provided for redraw, not the "draw_area" (BLUE+GRN+YELW);
 - what happens if the "draw_area" of the object is OK, but this area is bigger than the client area of the parent? 
 - When positioning an object which point should be taken as origin? The UL corner of this area? Or of the draw_area? If we use UL of draw area as origin ... and we have a 25 px margin ... and our parent has a 5px padding... Not only do we not respect the padding - we also overshoot the entire parent draw_area by 20 pixels!

So I think the proper solution (in order to keep the logic of "object size is object size, and margin has nothing to do with it") is to take the dimensions of the YELLOW rect as object size; and to take the UL of the margin as origin; and to refuse to draw objects, whose total size (i.e. draw_area + margin) would violate the dimensions of the client. Another possibility would be to reduce the client's size until it fits but I don't think it's a "clean" solution - at least when I tell a system to do something, I expect it either to do it exactly as instructed or to throw an error if impossible.

aenchevich, 2018-07-20:
.. oh and one more thing - I would like to disagree that margins only come into play when layouts are used. I think every object has the right to say "I need some space!" even when positioned without layouts :)

aenchevich, 2018-07-20:
... by the way - an example from life just came to my head:

You discover a new island and you decide to live there. So you want to build a house (your object), possibly put a fence around it (the borders) and possibly to have a garden inside the fence (the padding) and possibly even leave some area outside of the fence for some future needs (street, sidewalk, water pipes, electric cables, service road, etc..). 

Whatever you do you - it can not be bigger than the island, so everything must fit into the size of the island. 

On top of that, you can not build on sand, so you don't touch that area of the island and start building some distance away from the beach (this is the parent's padding).

Anyway, it's a big island and you are not a greedy person so you build a nice house, a garden and a fence, and leave some space outside of the fence for  and there is still lots of space on the island. 

Then I come. I build my house next to you. I also can not build on sand and I must respect your fence and garden and also the area which you marked for a street and sidewalk outside your property. 

So if I want a 100x100 house, with no fence and no garden and the space that's left is 100x140 - great. But if I also want a 1m fence and a 10m garden everywhere around my house then too bad - I can not build on this island because the W of the available space (100) is less than 1+10+100+10+1.

aenchevich, 2018-07-20:
hmm, the above suggests that objects can not be overlapping. And while at first glance it makes sense maybe you do need overlapping objects? Perhaps it's a good example up to the part where I come to the island, haha :) So basically client draw area must be respected (the area of the island without the beach/padding), but objects can overlap...?

kisvegabor, 2018-07-22:
Hi,

Your terminology translation was correct, I meant exactly the same.

So let's summarize what we know so far:
1. `lv_oj_set_size` sets `core_area` + `padding` + `border_width`
2. The object draws its background on `core_area` + `padding` + `border_width`
3. margin is always transparent
4. `lv_obj_set_pos(child, 0, 0)` moves `child` to the parent's `core_area` (0;0)
5. margin is ignored in positioning, so `core_area` + `padding` + `border_width` will be positioned
6. children can be visible only on the parent's `core_area`

However, the island example was very nice it not applies to GUI in every case: 
1. There is no reason to forbid overlapping. Moreover, it can be required during animations. For example, if you have a 3 button in a row and want to float them in from the left they might overlap each other during the animations.
2. It's natural that an object is bigger than its parent. Just think about the lists. If the scrollable area is too large it becomes scrollable.

What do you think?

aenchevich, 2018-07-23:
YES to everything with one little clarification:
 - If 'core_area' is the blue rectangle then also 'YES' to 4. 
 - Otherwise 4. I think should be:
    lv_obj_set_pos(child, 0, 0) moves child to (0;0) of the blue rectangle (i.e. parent padding must be respected!)

... and, yes  - the island example does not apply to all the GUI cases (damn!) :) If you don't mind I will delete my island posts, because, nice as they may be, they just confuse the readers and add noise?

kisvegabor, 2018-07-23:
In 4, yes it's the blue area because the parent's padding should be respected.

Please, do not remove the island example because it shows our way of thinking. No problem if we have side tracks.

hydra, 2018-07-24:
just a follow-up to my earlier statement 

> I'm not a fan of any of the default width/height/labels. I would prefer is the default size for everything is 0, and the default text for anything is an empty/null string.

I noted today that that default "Text" value causes memory allocation and de-allocation.  Consider:

```c
    lv_obj_t *btnSettingsLabel = lv_label_create(parent, NULL);
    lv_label_set_text(btnSettingsLabel, "My Text");
```

* The default "Text" is copied and allocated during `lv_label_create()` (as it internally calls `lv_label_set_text()`)
* "Text" is freed, "My Text" is copied during `lv_label_set_text`
* Needless memory fragmentation occurs if other allocations occur between the two calls.

aenchevich, 2018-07-24:
Yes, this was addressed by Gabor in one of his early comments:

_**Default values:**
You are right about "Text" and "Option1, Option2" ... default values have a negative effect on memory footprint and performance. I agree it should have a define in lv_conf.h to enable/disable these initial settings. While playing with the library it's better to have some init values but it's better to avoid them later when you overwrite them anyway._

So a define (in lv_conf.h) will govern whether the text objects will be created with some default text (for the lv_example apps) or withe empty strings (for when the lib is used 'in the field').

aenchevich, 2018-07-30:
Regarding usage of themes in an application, the approach that Gabor decided lvgl should use is (below is a quote from an email discussion):

> "You ***should*** use themes in your app if you want:
>  -  coding to be easy (objects are created with default "reasonable" sizes) 
>  -  graphics to look nice (the theme defines the entire look of your app). 
>
> You ***can*** build an app without themes, in which case you will need to manage all of this on your own".
> 
> It's kinda like using layouts: "hey, you want your life to be easy? Use layouts. You want it to be hard and to manage everything on your own? Don't use layouts."
>

kisvegabor, 2018-09-26:
Remainder based on  #422: Store "focused" styles in the themes

kisvegabor, 2019-03-02:
I'm clarifying things to implement in `v6.0`

In case of this issue, several months have passed but it seems users don't really need a more advanced boxing model (yet). There were some improvements in `fit` policies. See this [comment](https://github.com/littlevgl/lvgl/issues/838#issuecomment-466737762).

What seems really reasonable and I'm planning to add to `v6.0` is to replace `her/ver` padding with `left/right/top/bottom`. 

If later we decide to go for a more advanced boxing model we already have the ideas here.

Thank you very much for your ideas and contribution!

aenchevich, 2019-03-05:
Thank you for the update!

kisvegabor, 2019-03-17:
`padding.left/right/top/bottom` is added to `dev-6.0`. Still not fully tested but at first look, everything seems working.

closed 2019-03-17 by kisvegabor