lvgl/lvgl #303: lv_label bug with text that has a space at the end

upbeat27, 2018-07-10:
If you have a string like this:
`"test this text "`
The resulting label will only print:
`test this`

That is, when a string has a space at the end, the label will not print out the last word (even though there is room, i.e. the label is wide enough).
If you remove space (`' '`) from LV_TXT_BREAK_CHARS then it will print out everything.
If you remove the space from the string it will print out everything.

I looked into the code for lv_label, but I couldn't find what is causing this on first glance.

kisvegabor, 2018-07-11:
Hi,

your suggestion was correct! Thank for the examination of it. Trimming the last space, `\n` and `\r` is important when the label align is `LV_LABEL_ALIGN_CENTER` because in this case the line trailing spaces optically give an offset to the alignment. 

To solve it, I just removed the trimming of trailing spaces from `lv_txt_get_width` and done it in `lv_draw_label` instead. 

I pushed the update to the `beta` branch.

Test it, please!

kisvegabor, 2018-09-18:
As there was no activity here for a while I close it to keep the issues clear.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-09-18 by kisvegabor