lvgl/lvgl #304: Teensy with littlevgl (Arduino IDE), setup problem?

epikao, 2018-07-11:
Hello, I'm trying to compile the lvgl_test.ino from arduino library on my teensy 3.2.

Unfortunately I have follow first errors:
[image]
after change  #include <lvgl.h> to #include "lvgl.h" I have follow errors:
[image]

**BTW:** 
Is it correct that in the "my_disp_flush" function, I have to write the x/y and color data directly to the Interface of my Display (via SPI) ? So for example follow code:

```
	writecommand(0x2A); //x pixel-position 
	writedata(x); //start-position
	writedata(x); //start-position

	writecommand(0x2B); //y pixel-positon 
	writedata(y); //start-position
	writedata(y); //start-position
  
	writedata(color-data); //
	writedata(color-data);  //
```
I think the "my_disp_flush" is not a function which is called for every pixel? 
How is the structure of the "color_p" function/variable?
Where do I place the whole initialization of the Display? (in my case "init_ST7789V()"?

kisvegabor, 2018-07-11:
Hi,

Have you copied the `littlevgl` folder from [here](https://github.com/littlevgl/lv_boards/tree/master/arduino_library) to your library folder?

**disp_flush**
Here you can see how should a `disp_flush` look like: https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/0_porting/lv_tutorial_porting.c. 
Basically it should copy the `color_p` array to the specified area.

epikao, 2018-07-11:
> Have you copied the littlevgl folder from here to your library folder?
that's how my folder looks at the moment:

[image]

> disp_flush
> Here you can see how should a disp_flush look like: https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/0_porting/lv_tutorial_porting.c.
> Basically it should copy the color_p array to the specified area.

I need only 16bit Color values per Pixel...
so what is the data typo of "color_p", is it already a 16bit, or 24bit Value?
thank you

epikao, 2018-07-11:
> lv_conf.h should be NEXT TO the src folder (as the example shows in the lv_boards repo)

I already check that, also your tutorial... it's difficult for me to understand.
I have no clue why I always get the error that "lvgl.h" is not found?!! 

[image]

kisvegabor, 2018-07-11:
At first time, your error was "No such file or director: ../../lv_conf.h". It was because of the misplaced lv_conf.h.

What is your error message now?

epikao, 2018-07-11:
> What is your error message now?

[image]

thank you

epikao, 2018-07-12:
lvgl.h error only disappears if I change to "lvgl.h" instead of <lvgl.h>.
but than or now I still have follow error:
[image]

epikao, 2018-07-12:
so now I solved the problem of lv_conf.h and lvgl.h..., the problem is because I didn't read the "README.md" sorry to these circumstances :-/ 

Now I have follow errors see below, but I think is because I should configure something in the conf-file...
[image]

epikao, 2018-07-12:
Hello, at the moment I didn't find a solution for the errors above? (undefined reference to 'lv_flush_ready' etc...

kisvegabor, 2018-07-12:
I put together a working Ardunio project for Teensy 3.2
Here you can see the directory structure.

[Teensy_LittlevGL.zip](https://github.com/littlevgl/lvgl/files/2190245/Teensy_LittlevGL.zip)

epikao, 2018-07-13:
ok. thank you, now it works :-D
Follow Item in a "README" confused me: 
> 5. In your *main.c*: #include "lvgl/lvgl.h"

epikao, 2018-07-13:
now I finished the code, but it looks that function "my_disp_flush" is not called:

```
void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{
  /*TODO copy 'color_array' to the specifed coordinats of your disaply.*/
  writecommand(0x2A); //x pixel-position anfahren
  writedata(x1>>8); //start-position
  writedata(x1); //start-position
  writedata(x2>>8); //end-position
  writedata(x2); //end-position

  writecommand(0x2B); //y pixel-positon anfahren
  writedata(y1>>8); //start-position
  writedata(y1); //start-position
  writedata(y2>>8); //end-position
  writedata(y2); //end-position

  writecommand(0x2C); //RAM Daten senden
  digitalWrite(CS0, 0);
  digitalWrite(DC, 1); //Command = 0, Data = 1

  //SPI.transfer(color_array, 0, sizeof(color_array));
  SPI.transfer(0xA0);

  digitalWrite(CS0, 1);
  /*Tell the flushing is ready*/
  lv_flush_ready();
}
```

epikao, 2018-07-13:
now it looks that function "my_disp_flush" is not called:

```
void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{

  /*TODO copy 'color_array' to the specifed coordinats of your disaply.*/
  writecommand(0x2A); //x pixel-position anfahren
  writedata(x1>>8); //start-position
  writedata(x1); //start-position
  writedata(x2>>8); //end-position
  writedata(x2); //end-position

  writecommand(0x2B); //y pixel-positon anfahren
  writedata(y1>>8); //start-position
  writedata(y1); //start-position
  writedata(y2>>8); //end-position
  writedata(y2); //end-position

  writecommand(0x2C); //RAM Daten senden
  digitalWrite(CS0, 0);
  digitalWrite(DC, 1); //Command = 0, Data = 1

  //SPI.transfer(color_array, 0, sizeof(color_array));
  SPI.transfer(0xA0);

  digitalWrite(CS0, 1);
  /*Tell the flushing is ready*/
  lv_flush_ready();
}
```
```

void loop() {
  // put your main code here, to run repeatedly:

  lv_task_handler();
  /*If 'lv_tick_inc(5)' is not called in a Timer then call it here*/
  lv_tick_inc(5);
  delay(5);             /*Wait a little*/
}
```

kisvegabor, 2018-07-13:
Glad to hear it compiles!

Are you sure it's not called? Can you verify it by printing to the console something in `my_disp_flush`?

epikao, 2018-07-16:
just verified, the problem is that the program crashs :-( 
If I do the "lv_tick_inc(5)" without timer means in the loop and add a delay of about 1000ms the program calls the loop aprox 10 times and than it crashes.
If I remove the 1000ms in the program nothing more works.
The same if I do the "lv_tick_inc(5)" with Timer...

If I remove the "lv_tick_inc(5)" handling the program does not crash anymore. What's wrong??

According your porting Infos "lv handler, and tick" should called every 1...5ms, but is that not very stressful for the CPU??

thank you.

kisvegabor, 2018-07-16:
It's not so stressful because it calls tasks only if their period time is over.

Please attach you `ino` file to check what might be wrong.

epikao, 2018-07-17:
[lvgl_test - Kopie.txt](https://github.com/littlevgl/lvgl/files/2200492/lvgl_test.-.Kopie.txt)

see line 126, if I activate it, serial monitor(debug) shows follow:
setup
loop
.. nothing more 

if line 126 deactivated:
setup
loop
loop
loop
loop etc.

kisvegabor, 2018-07-17:
At first look, it seems correct. 
Please, post the summary of the crash issue to this thread: https://github.com/littlevgl/lvgl/issues/189
Here @rbultman also had a crash issue using Arduino. Join our forces to figure out what's going on :)

As the original question (set up)  is solved I close this issue. Let's continue in the other topic.

closed 2018-07-17 by kisvegabor