lvgl/lvgl #305: Ink effect added to buttons (give feedback)

kisvegabor, 2018-07-11:
Hi,

I added an ink effect to buttons. It similar to the one in Android: when you press a button a circle starts to grow to re-color the button with  pressed color.

In LittlevGL a growing circle will be drawn with `LV_BTN_STYLE_PR` or `LV_BTN_STYLE_TGL_PR`.

Example code:
```c
    lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
    lv_btn_set_ink_fill_time(btn, 300);
    lv_btn_set_ink_wait_time(btn, 100);

    lv_obj_t * label = lv_label_create(btn, NULL);
    lv_label_set_text(label, "Button");
```

[btn_ink.mp4.zip](https://github.com/littlevgl/lvgl/files/2185723/btn_ink.mp4.zip)

This feature is in the `dev-5.2` branch.

Please test it if you are curious, and let me know what do think about it!

seyyah, 2018-08-09:
This is good idea.

This effect's another name is ripple effect,

- https://codepen.io/madshaakansson/pen/ykode
- https://codepen.io/lehollandaisvolant/pen/dMQXYX
- https://materializecss.com/buttons.html

kisvegabor, 2018-08-10:
Thanks for the links! 
Especially the first looks awesome. I will check it what it does exactly.

kisvegabor, 2018-10-07:
I updated the ink effect according to the demos. 
It is already in v5.2.

Thank you!

closed 2018-10-07 by kisvegabor