lvgl/lvgl #306: Porting issue due to unitialized static variables

Zahriman, 2018-07-12:
Hi Gabor,

First of all, I haven't many experience with your library (playing with it for 3 days) but seems great (i.e. easy to understand), flexible and already implementing lots of features.

So I'm porting your library in an already existing project on TI C6000 DSP attached with a monochrome 128x64 LCD (hence I upvote for this kind of LCD support, but side story and will comment on appropriate issue). I started, cool simple init, driver registration and the flush callback to write! But it revealed to be not that easy. First run: the program gone crazy by jumping to random address. Solved by initializing static function pointers to null. Started working but faced another problem: depending on the code size, it worked or not. I tracked it down (by simply compiling or not an unused buffer) and it revealed to be that the lv_style_scr.body.empty is not initialized and it happened to be 0 or 1 depending on the variable place in memory. Once more due to a static variable not initialized. Solved by memsetting to 0 the lv_style_scr in lv_style_init.

Static variable go to a section commonly called .bss. While some argue that C standard impose unitialized variables to be set to 0 (or NULL), many compilers do it but that's not the case of the TI C6000 compiler in COFF legacy mode.

Of course I can force to set the .bss section to 0 (did not managed to as of now) but I think it would be better that the library does not rely on the linker behavior of setting to 0 the unitialized static variables. This would increase portability, ease of use (does not need to track this kind of problem), and compiler/linker independence.

What do you think?
Best regards

hydra, 2018-07-12:
Uninitialized variables should be set to `0` via the startup code, it's quite common practice.  Likely you can find or create some linker script and startup code to do this.

Having to initialise every variable to `0`/`null` is generally frowned upon by many on projects I've worked on.  The main reason is that a) developers forget, b) it creates uncertainty, b) it causes code bloat.

Likely other libraries that you use also expect unitialized variables to be `0`.

However, I would suggest the any porting instructions mention this requirement.

kisvegabor, 2018-07-12:
Hi,

As far as I know, it's a C standard to initialize all global variables to zero. I googled on it and others state the same. Maybe it's a bug in your SDK?

kisvegabor, 2018-07-12:
I found this: http://e2e.ti.com/support/development_tools/compiler/f/343/p/142730/841284

hydra, 2018-07-12:
Here's a couple of examples on the STM32 platform, for reference

https://github.com/cleanflight/cleanflight/blob/master/src/main/startup/startup_stm32f40xx.s#L114-L124
https://github.com/cortexm/baremetal/blob/master/src/startup/startup.cpp#L31-L37

Zahriman, 2018-07-13:
Hi all and thanks for your answers.

You're right, it seems to be C standard to initialiaze all global variables; I had a doubt after reading discussions with people debating whether it's standard or not.
But it's not a bug in the SDK, it's stated as this in the TI documentation:

> In ANSI C, global and static variables that are not explicitly initialized must be set to 0 before program execution. The C/C++ EABI compiler supports preinitialization of uninitialized variables by default. This can be turned off by specifying the linker option --zero_init=off. COFF ABI does not support zero initialization.

Unfortunately, the project I am working on uses the COFF ABI and is not going to be translated to EABI soon.

@hydra Thank you for the links, I already did that using gcc on an ARM platform; but here I discover the C6000 world. I'll investigate more on that, I have read that I can put a ```fill: 0``` in front of the section in the linker script.
By the way, I understand your a) and c) points but why do you state b) it creates uncertainty?

Maybe @hydra is right and it would just require a warning in the porting guide

hydra, 2018-07-13:
b) because if you see initialisation to zero where it is not needed see it you might continue to repeat it where it is not needed, leaving readers of the code wondering why it's there.

kisvegabor, 2018-07-25:
According to our discussion, this issue seems quite specific to your compiler and should be solved by some compiler settings.

Because of it, I close this issue. 

@Zahriman If you found a solution, please share here to help others. Thank you!

I will add a remark about it to the porting guide.

closed 2018-07-25 by kisvegabor