lvgl/lvgl #308: Performance issue with two page objects.

hydra, 2018-07-12:
I'm having an issue with two page objects when created side-by-side, the left hand one updates MUCH slower than the right hand side one.

Here's a video showing the issue.

[image]

https://youtu.be/5LI639yPyqw

Any ideas?

Things I tried:
* rotating the screen.
* moving the left object to the right.
* moving right hand side object to left and vice-versa.
* disabling all other UI objects (left hand side one is always slow).

here's the code:

```
static lv_obj_t *leftPanel;
static lv_obj_t *rightPanel;
static lv_obj_t *leftPage;
static lv_obj_t *rightPage;

lv_style_t titleStyle;

lv_style_t leftPanelStyle;
lv_style_t rightPanelStyle;

void ui_createPanels(void)
{
    lv_obj_t *root = lv_scr_act();

    lv_coord_t height = lv_obj_get_height(root);
    lv_coord_t width = lv_obj_get_width(root);

    lv_style_copy(&leftPanelStyle, &lv_style_transp_tight);
    leftPanelStyle.body.main_color = LV_COLOR_GREEN;
    leftPanelStyle.body.grad_color = leftPanelStyle.body.main_color;
    leftPanelStyle.body.empty = 0;
    leftPanelStyle.text.color = LV_COLOR_GRAY;

    lv_style_copy(&rightPanelStyle, &lv_style_transp_tight);
    rightPanelStyle.body.main_color = LV_COLOR_OLIVE;
    rightPanelStyle.body.grad_color = rightPanelStyle.body.main_color;
    rightPanelStyle.body.empty = 0;
    rightPanelStyle.text.color = LV_COLOR_GRAY;

    leftPanel = lv_cont_create(root, NULL);
    lv_obj_set_width(leftPanel, width / 2);
    lv_obj_set_height(leftPanel, height);
    lv_obj_set_style(leftPanel, &leftPanelStyle);
    lv_obj_align(leftPanel, root, LV_ALIGN_IN_LEFT_MID, 0, 0);

    leftPage = lv_page_create(leftPanel, NULL);
    lv_obj_set_width(leftPage, width / 2);
    lv_obj_set_height(leftPage, height - 16);
    lv_obj_set_pos(leftPage, 0, 16);
    lv_page_set_style(leftPage, LV_PAGE_STYLE_BG, &leftPanelStyle);
    lv_page_set_style(leftPage, LV_PAGE_STYLE_SCRL, &leftPanelStyle);
    //lv_obj_align(leftPage, leftPanel, LV_ALIGN_IN_LEFT_MID, 0, 16);

    rightPanel = lv_cont_create(root, NULL);
    lv_obj_set_width(rightPanel, width / 2);
    lv_obj_set_height(rightPanel, height);
    lv_obj_set_style(rightPanel, &rightPanelStyle);
    lv_obj_align(rightPanel, root, LV_ALIGN_IN_RIGHT_MID, 0, 0);

    rightPage = lv_page_create(rightPanel, NULL);
    lv_obj_set_width(rightPage, width / 2);
    lv_obj_set_height(rightPage, height - 16);
    lv_obj_set_pos(rightPage, 0, 16);
    lv_page_set_style(rightPage, LV_PAGE_STYLE_BG, &rightPanelStyle);
    lv_page_set_style(rightPage, LV_PAGE_STYLE_SCRL, &rightPanelStyle);
    //lv_obj_align(rightPage, rightPanel, LV_ALIGN_IN_RIGHT_MID, 0, 16);

    /*Create a label on the page*/
    lv_obj_t * label1 = lv_label_create(leftPage, NULL);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_BREAK);             /*Automatically break long lines*/
    lv_obj_set_width(label1, lv_page_get_scrl_width(leftPage));          /*Set the width. Lines will break here*/
    lv_label_set_text(label1, "Lorem ipsum dolor sit amet, consectetur adipiscing elit,\n"
                             "sed do eiusmod tempor incididunt ut labore et dolore magna aliqua.\n"
                             "Ut enim ad minim veniam, quis nostrud exercitation ullamco\n"
                             "laboris nisi ut aliquip ex ea commodo consequat. Duis aute irure\n"
                             "dolor in reprehenderit in voluptate velit esse cillum dolore\n"
                             "eu fugiat nulla pariatur.\n"
                             "Excepteur sint occaecat cupidatat non proident, sunt in culpa\n"
                             "qui officia deserunt mollit anim id est laborum.");

    /*Create a label on the page*/
    lv_obj_t * label2 = lv_label_create(rightPage, NULL);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_BREAK);             /*Automatically break long lines*/
    lv_obj_set_width(label2, lv_page_get_scrl_width(rightPage));          /*Set the width. Lines will break here*/
    lv_label_set_text(label2, "Lorem ipsum dolor sit amet, consectetur adipiscing elit,\n"
                             "sed do eiusmod tempor incididunt ut labore et dolore magna aliqua.\n"
                             "Ut enim ad minim veniam, quis nostrud exercitation ullamco\n"
                             "laboris nisi ut aliquip ex ea commodo consequat. Duis aute irure\n"
                             "dolor in reprehenderit in voluptate velit esse cillum dolore\n"
                             "eu fugiat nulla pariatur.\n"
                             "Excepteur sint occaecat cupidatat non proident, sunt in culpa\n"
                             "qui officia deserunt mollit anim id est laborum.");
}

```

hydra, 2018-07-12:
Quick update:

The problem seems to be caused by the scrollbar of the left hand page.

If this line is added to the above code the performance of the two pages is identical.

`lv_page_set_sb_mode(leftPage, LV_SB_MODE_OFF);`

hydra, 2018-07-12:
ok, I think I found the bug.

From my understanding too much of the display is being invalidated due to the incorrect calculation of the area.

```
diff --git a/lv_objx/lv_page.c b/lv_objx/lv_page.c
index d7e00c0..9451e7d 100644
--- a/lv_objx/lv_page.c
+++ b/lv_objx/lv_page.c
@@ -621,8 +621,8 @@ static lv_res_t lv_page_scrollable_signal(lv_obj_t * scrl, lv_signal_t sign, voi
                 lv_area_copy(&sb_area_tmp, &page_ext->sb.hor_area);
                 sb_area_tmp.x1 += page->coords.x1;
                 sb_area_tmp.y1 += page->coords.y1;
-                sb_area_tmp.x2 += page->coords.x2;
-                sb_area_tmp.y2 += page->coords.y2;
+                sb_area_tmp.x2 += page->coords.x1;
+                sb_area_tmp.y2 += page->coords.y1;
                 lv_inv_area(&sb_area_tmp);
                 page_ext->sb.hor_draw = 0;
             }
@@ -630,8 +630,8 @@ static lv_res_t lv_page_scrollable_signal(lv_obj_t * scrl, lv_signal_t sign, voi
                 lv_area_copy(&sb_area_tmp, &page_ext->sb.ver_area);
                 sb_area_tmp.x1 += page->coords.x1;
                 sb_area_tmp.y1 += page->coords.y1;
-                sb_area_tmp.x2 += page->coords.x2;
-                sb_area_tmp.y2 += page->coords.y2;
+                sb_area_tmp.x2 += page->coords.x1;
+                sb_area_tmp.y2 += page->coords.y1;
                 lv_inv_area(&sb_area_tmp);
                 page_ext->sb.ver_draw = 0;
             }
@@ -692,16 +692,16 @@ static void lv_page_sb_refresh(lv_obj_t * page)
         lv_area_copy(&sb_area_tmp, &ext->sb.hor_area);
         sb_area_tmp.x1 += page->coords.x1;
         sb_area_tmp.y1 += page->coords.y1;
-        sb_area_tmp.x2 += page->coords.x2;
-        sb_area_tmp.y2 += page->coords.y2;
+        sb_area_tmp.x2 += page->coords.x1;
+        sb_area_tmp.y2 += page->coords.y1;
         lv_inv_area(&sb_area_tmp);
     }
```

BesitzeRuf, 2018-07-12:
Hi hydra,
nice issue, the thing I can't to understand yet is why the left panel has decreased performance... I'm trying to swap initialization order of objects...but always left panel is affected...

hydra, 2018-07-12:
Here's a video showing the bug fixed using PR #309

[image]

https://youtu.be/U2ZFnagUIL8

BesitzeRuf, 2018-07-12:
Yes, I think it is fixed now (tested by demo example + some my own test project).

hydra, 2018-07-12:
@BesitzeRuf Great! 

This bug-fix will speed up scrolling of all scrollbars that are not drawn at the bottom or far-right of the screen.

hydra, 2018-07-12:
Also, since 'List' objects are built using Page objects it will fix them too.

kisvegabor, 2018-07-12:
Awesome work here guys! The observation and bugfix were perfect! 
Thank you very much!

closed 2018-07-12 by kisvegabor