lvgl/lvgl #310: Temporal coupling issues.

hydra, 2018-07-12:
Hi,

I'm having some temporal coupling issues with LittleVGL recently, here's an example.

A
```
zoneTitleLabel = lv_label_create(zoneHeaderPanel, NULL);
lv_obj_set_width(zoneTitleLabel, 160);
lv_label_set_long_mode(zoneTitleLabel, LV_LABEL_LONG_DOT);
lv_label_set_static_text(zoneTitleLabel, "My long title");
```

vs

B
```
zoneTitleLabel = lv_label_create(zoneHeaderPanel, NULL);
lv_label_set_long_mode(zoneTitleLabel, LV_LABEL_LONG_DOT);
lv_obj_set_width(zoneTitleLabel, 160);
lv_label_set_static_text(zoneTitleLabel, "My long title");
```

can you guess which one works and which one doesn't?

No, neither can I and that's the temporal coupling issue.

What happens in A is that the long mode is that the width is set to 160, then the long mode is set and during that call code in `lv_label_refr_text` eventually determines that the width should be the width of the word 'Text' using the whatever style is used by the label.  Then, when the text is set it doesn't fit in the incorrectly sized label which causes you to scratch your head wondering what you've done wrong.

'B', above, works just fine, but you'd never know from simply looking at the 4 lines of code above.

I've faced similar issues when applying styles, aligning objects and so on, all of which makes designing a UI very time consuming and somewhat frustrating as you constantly have to debug the LVGL code to find out which call is making things not work in the order you specified them.

Temporal coupling leads to a very fragile (application) code base.

kisvegabor, 2018-07-12:
Hi,

As the default long mode is `LV_LABEL_LONG_EXPAND` it will overwrite the width. So the order of function matters in this case. And it's not emphasised enough. So in my opinion, the first thing can be done is to mark this in the functions documentation. E.g. in: "LV_LABEL_LONG_EXPAND will overwrite the width you set... For LV_LABEL_LONG_BREAK/ROLL/DOT set the long mode first and then width..." or something like that.

I agree, that it might be frustrating. Have you got an idea to make is better?

But, IMHO, misalignment after resizing is natural because the alignment coordinates are not saved. Although, in https://github.com/littlevgl/lvgl/issues/208 were an idea to optionally save the alignment data (in `lv_obj_align`) to make possible to apply them with an `lv_obj_realign` function. The idea might be extended to automatically realign if the dimensions are changed. What do you think?

hydra, 2018-07-13:
in debug builds you could add assertions, i.e.  if long mode is on and width is set raise an assertion.

this could help developers get to grips with the library.

documentation would help for sure.

a design that avoids the issue entirely would be preferable of course.  a design that doesn't require details documentation would be even better :)

kisvegabor, 2018-07-13:
It just came to my mind that a log module is added (or will be added in v5.2) to LittlevGL. It's a great purpose for it to notify the user about these tricky situations.

And of curse the documentation...

kisvegabor, 2018-09-04:
Hi,

I notified the user in the comment of `lv_label_set_long_mode` about the size settings should come after it.

closed 2018-09-04 by kisvegabor