lvgl/lvgl #311: Background image and TabView

mubes, 2018-07-13:
Hi,

Just started playing with this on an IMXRT1052 EVB board and associated LCD....working great, and a really impressive piece of software. One question that's probably down to sticky-fingers;

Is it possible to have a background image bleed-through the background of a TabView? I've set the empty attribute on both the LV_TABVIEW_STYLE_BG and LV_TABVIEW_STYLE_BTN_BG but that just seems to lead to a solid background coming through, not the underlying image.

Regards

DAVE

kisvegabor, 2018-07-13:
Hi,

I tested it and it worked for me. The same code works for you too?

```c
/*Gradient on the screen*/
lv_style_scr.body.main_color = LV_COLOR_RED;
lv_style_scr.body.grad_color = LV_COLOR_BLUE;

/*Empty background style*/
static lv_style_t bg_style;
lv_style_copy(&bg_style, &lv_style_plain);
bg_style.body.empty = 1;

/*Create a tabview set the empty background style*/
lv_obj_t * tv = lv_tabview_create(lv_scr_act(), NULL);
lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BG, &bg_style);

/*Add two tabs*/
lv_obj_t * tab1 = lv_tabview_add_tab(tv, "First");
lv_obj_t * label1 = lv_label_create(tab1, NULL);
lv_label_set_text(label1, "Dummy text 1");

lv_obj_t * tab2 = lv_tabview_add_tab(tv, "Second");
lv_obj_t * label2 = lv_label_create(tab2, NULL);
lv_label_set_text(label2, "Dummy text 2");
```

And got this:
[image]

mubes, 2018-07-14:
Yes, that works perfectly.....must be something tangled in my code! 
Sticky fingers. Apologies for the noise, please close this.

Regards

DAVE

On 14/07/18 00:38, Gabor Kiss-Vamosi wrote:
> /*Gradient on the screen*/
> lv_style_scr.body.main_color = LV_COLOR_RED;
> lv_style_scr.body.grad_color = LV_COLOR_BLUE;
>
> /*Empty background style*/
> static  lv_style_t  bg_style; [...]

kisvegabor, 2018-07-15:
Great, no problem!

Glad to help :)

closed 2018-07-15 by kisvegabor