lvgl/lvgl #312: Issues about using group and external keyboard

as564335sa, 2018-07-16:
Hi,
I have two issues about using group and external keyboard,
1、How to delete chars of 'ta' object when using external keyboard？(such as a custom “del” key)

2、How to use a custom external keyboard “return” key to close a window or delete some objects, can it be implemented with this library?I have no idea.

kisvegabor, 2018-07-16:
Hi,

1. An `LV_GROUP_KEY_DEL` define is really missing.

2. So for example, you are typing in a text area, press "return" and you would like to close the window of the text area? If yes, then you can catch the "return" key in `indev_read` function and instead of setting it as the current key you can delete something. You can use the `lv_group_get_focused()` function to determine which object is focused.

I just added `LV_GROUP_KEY_DEL` and its handling to the Text area and done a minor update to be sure the delete of an object in `indev_read` won't break the library. The update is in the `beta` branch.

as564335sa, 2018-07-17:
Hi,
Great! I'll try it later. Thanks!

closed 2018-07-17 by as564335sa