lvgl/lvgl #313: mistaken spelling.

yuanzhaoming, 2018-07-17:
At line of 105 in the file misc/lv_area.c，there is a mistaken spelling 'her'. Actually，it needs to be 'here'.

BesitzeRuf, 2018-07-17:
Hi,

because this issue is opened, I want to add just other misspelling. in the same file (misc/lv_area.c) should be function:
```
bool lv_area_union(lv_area_t * res_p, const lv_area_t * a1_p, const lv_area_t * a2_p)
    /* Get the smaller area from 'a1_p' and 'a2_p' */
    res_p->x1 = LV_MATH_MAX(a1_p->x1, a2_p->x1);
    res_p->y1 = LV_MATH_MAX(a1_p->y1, a2_p->y1);
    res_p->x2 = LV_MATH_MIN(a1_p->x2, a2_p->x2);
    res_p->y2 = LV_MATH_MIN(a1_p->y2, a2_p->y2);
    
    /*If x1 or y1 greater then x2 or y2 then the areas union is empty*/
    bool union_ok = true;
    if((res_p->x1 > res_p->x2) ||
       (res_p->y1 > res_p->y2))
    {
        union_ok = false;
    }
    
    return union_ok;
}
```
renamed to lv_area_intersection?

kisvegabor, 2018-07-17:
Hi,

@yuanzhaoming  I fixed it in the `beta` branch. Please, keep reporting these typos or create pull requests to the `beta` branch.

@BesitzeRuf lv_area_union? How did I think that? It seems I had problems with the definitions... Alaso fixed in `beta`.

Thank you, guys!

yuanzhaoming, 2018-07-20:
Hi,
I have been reading your code these days，your code enlightened me.

In the file (lv_misc/lv_task.c) of beta version,I think there is a misspelling variable 'task_interruper'.

Should it be 'task_interrupter'?

kisvegabor, 2018-07-22:
Thank you, fixed!

kisvegabor, 2018-08-18:
As there wasn't activity here for a while I close this issue. 
If you find new typos, please reopen this issue!

closed 2018-08-18 by kisvegabor