lvgl/lvgl #314: How to receive input events

Bitblade, 2018-07-17:
Hi,

I feel kind of guilty for creating an issue for this, but I have been searching for a while now but I can't find it.  I would assume it's something obvious I'm missing, but sadly, am going to need some help. 

**I have:** 
USE_KEYBOARD, USE_LV_GROUP and otherwise relatively standard settings. 

**I need:**
A way to detect key presses (either by pushed events or polling) without there being an object visible on the screen to receive such events. An example of what I would use this for is to switch to an debug mode when the operator attaches a keyboard. 
[d]: Opens debug mode
[1]: Shows network information
[2]: Shows hardware information
[3]: Shows cache information

**I tried:**
While registering the keyboard driver, I got the drivers `lv_indev_t` pointer:
    `return lv_indev_drv_register(&keyboardDriver);`

In a separate thread, I then ran: 
```
 while(true){
    lv_indev_wait_release(this->driver);
    lv_indev_data_t data;
    lv_indev_read(this->driver, &data);
    //lv_indev_reset(this->driver);

    std::cout << "Input event: ";
    if(data.state == LV_INDEV_STATE_PR){
        std::cout << "PRESS[";
    }else{
        std::cout << "RELEA[";
    }
    std::cout << unsigned(data.key) << "]\n";
}
```

Sadly, this only gives me an endless stream of 
`Input event: RELEA[0]`, ignoring any actual input.

kisvegabor, 2018-07-17:
Hi,

Maybe `lv_indev_wait_release` is misleading. It makes the GUI to ignor the current pressing until a release.

Following your concept, I would suggest this:

```c
lv_indev_state_t last_state = LV_INDEV_STATE_REL;
while(true){
   lv_indev_data_t data;
   lv_indev_read(this->driver, &data);

   if(last_state == LV_INDEV_STATE_PR && data->state == LV_INDEV_STATE_REL) {
	   std::cout << "RELEA[" << unsigned(data.key) << "]\n";
   }

   last_state = data->state;
}
```

However, as LittelvGL also calls `lv_indev_read` it seems to me better to check the key releases in your `keyboard_read` function and do something there: e.g. `lv_scr_load(debug_scr)`
I didn't tested it so far, but is should work.

Bitblade, 2018-07-23:
Sorry for my late reply, a different project required priority. 

Thanks to your hints, I've managed to solve the problem. Indeed, I assumed wrongly that `lv_indev_wait_release` would block until a key was released. 

I have already started incorporating this in my keyboard_read function as you suggested.

closed 2018-07-23 by Bitblade

kisvegabor, 2018-07-23:
No problem, thank you for the feedback!

boraozgen, 2021-03-18:
I used the same approach, but it feels a bit hacky. Shouldn't there be a possibility to assign an "official" callback to key presses?

embeddedt, 2021-03-18:
@boraozgen Have you tried [`LV_EVENT_KEY`](https://docs.lvgl.io/latest/en/html/overview/event.html#related-to-keypad-and-encoder)?

boraozgen, 2021-03-18:
Do you mean subscribing to object events with `lv_obj_set_event_cb`? The use case here is to detect key presses globally, i.e. not tied to an object.

kisvegabor, 2021-03-19:
You can also use `feedback_cb` of the indev driver. It's called on every event sent to any object if there is an active input device.

boraozgen, 2021-03-22:
Thanks! `feedback_cb` works for my case as I am using it for the back button, which emits a `LV_EVENT_CANCEL` event when assigned to `LV_KEY_ESC,` but it probably wouldn't be suitable for other keys. Couldn't the `feedback_cb` also provide the pressed key for `LV_EVENT_KEY` or is this information not available during that callback?

kisvegabor, 2021-03-24:
In the `feedback_cb` you can 
```c
lv_indev_t * indev = lv_inedv_act();
uint32_t k = lv_indev_get_key(indev);
```

See the docs here: https://docs.lvgl.io/latest/en/html/porting/indev.html#other-features

boraozgen, 2021-03-24:
Great, thanks! Missed that section in the docs. I would still recommend considering a more intuitive API to be notified on a pressed key for the future versions.

kisvegabor, 2021-03-24:
I'm thinking about it, thanks!

meera1310, 2021-09-21:
I want to read keyboard inputs, my screen has id and password fields. And I need to check whether user entered correct id/pwd and validate those fields. 
e.g., id --------
       pwd --------
I want to check if user has entered "user" and "admin" as a id and password respectively.