lvgl/lvgl #315: lvgl appoach "event driven" instead of "polling"?

StephanSy, 2018-07-18:
Hi!
As far as I understood, the operation of the LvGL library is the following:
- perodic call of lv_task_handler()
- perodic call of lv_tick_inc()

The device I am working on will have a very limited LCD display and the only input device will be a rotary encoder with push button. Moreover, it can be expected that this interface is rarely used as the device is meant to be controlled via ethernet.

So I was thinking if it is possible to change this basic operation principle from polling to a event driven approach. That is the lv_task_handler() needs only be called on an event from input device or an update of the displayed data.

Or is there anything like a sleep mode?
Thanks in advance for your help.

BR,
Stephan

kisvegabor, 2018-07-18:
Hi,

Theoretically, it's possible. So there two main things as you mentioned:
1. Input device (encoder) read
2. Display refresh

If the GUI will change only if you change something by the encoder then it seems okay to call `lv_task_handler()` if the encoder is changed. It will call your `indev_read` function to handle the encoder.

However, you should call `lv_task_handler()` twice because it might happen that the display refresher task runs first, sees nothing happened (yet), and then comes the input device handling which changes the GUI. But in the next call of `lv_task_handler()` the refresher task will be called. 

Keep in mind the features which have an effect as time passes (e.g. animation) won't work. To handle long press with the encoder `lv_task_handler()` should be called during pressing.

It might be required to lower the values of `LV_INDEV_READ_PERIOD` (in lv_conf.h) to surely not miss a fast click. If `LV_INDEV_READ_PERIOD` is not elapsed since the last call of `lv_task_handler` then the input device handler task won't be called.

If the data from Ethernet can change the GUI then `lv_task_handler()` should be called after you refreshed the content.

Independently from this `lv_tick_inc(x)` still need to be called periodically.

Krastanov, 2018-08-25:
Could you elaborate on this.

- Why is lv_tick_inc(x) necessary?
- How much control do we have over when the "display refresher" runs?
- Can we sleep for a long time, detect a button press in non-lvgl code, only then run the display refresher, then run lv_task_handler which is told of the button press (which was saved to a queue for instance). and only then run the display refresher again?
- Is there a way to know whether all animations have ended, so that we can re-enter sleep?

kisvegabor, 2018-08-26:
> Why is lv_tick_inc(x) necessary?
The library runs task periodically (screen refresh, input device read, animations etc) These needs to know how much time has elapsed.

> How much control do we have over when the "display refresher" runs?
In lv_conf.h `LV_REFR_PERIOD` sets its period time

> Can we sleep for a long time, detect a button press in non-lvgl code, only then run the display refresher, then run lv_task_handler which is told of the button press (which was saved to a queue for instance). and only then run the display refresher again?

If you don't have animations it seems possible. If the GUI is idle then it only needs to read the input devices to see if something happens. So if you call `lv_task_handler` only while user actions are happening then it could work, but need to be tried out.

> Is there a way to know whether all animations have ended, so that we can re-enter sleep?
Not now, but I can add a function to get it. It's quite simple.

kisvegabor, 2018-08-27:
In the practice, you don't need to weak up from sleep to call `lv_tick_inc()` periodically. It's enough to call it on wake up with big value to be sure all the task will run once on the first call of `lv_task_handler()`. So for example `lv_tick_inc(1000)` should be fine on wake up. 

However, as you mentioned while the button or touchpad press lasts you need to call `lv_tick_inc()` periodically.

If *"officially" guaranteed"* means that this opportunity will be stated in the documentation and the upcoming features won't break it then yes, I can guarantee that for the foreseeable future. But a confirmation from your side is required to state this responsibly.

Krastanov, 2018-08-27:
Yeah, I am sorry, my phrasing was quite poor. I did indeed just wanted to be able to see this behavior be documented and not break in the upcoming versions.

kisvegabor, 2018-08-28:
Okay, I will update the documentation!

I was able to test in the PC simulator.

I modified the `mouse_handler()` in `mouse.c`:

```c
void mouse_handler(SDL_Event *event)
{
    switch (event->type) {
        case SDL_MOUSEBUTTONUP:
            if (event->button.button == SDL_BUTTON_LEFT)
                left_button_down = false;
            break;
        case SDL_MOUSEBUTTONDOWN:
            if (event->button.button == SDL_BUTTON_LEFT) {
                left_button_down = true;
                last_x = event->motion.x;
                last_y = event->motion.y;

            	lv_tick_inc(1000);                        /*Execute tasks on button press*/
            	lv_task_handler();
            }
            break;
        case SDL_MOUSEMOTION:
            last_x = event->motion.x;
            last_y = event->motion.y;

            break;
    }
}
```
And I modified the `while(1)`  like this:
```c
	bool print_sleep = true;
	bool print_wake_up = true;
	while(1) {

		if(lv_indev_get_inactive_time(NULL) < 1000) {
			lv_task_handler();
			lv_tick_inc(5);
			print_sleep = true;
			if(print_wake_up) {
				printf("Wake up\n");
				print_wake_up = false;
			}
		} else {
			print_wake_up = true;
			if(print_sleep) {
				printf("Sleep\n");
				print_sleep = false;
			}
		}
		usleep(5000);
	}
```

And of course, I removed the tick increment task.

So now the system wakes up on button press, and go sleep after 1000 ms inactivity.

kisvegabor, 2018-08-28:
MEanwhile I found that it's better to use `lv_tick_inc(LV_REFR_PERIOD);` in `mouse_handler` because the too big number can break some animations.

StephanSy, 2018-11-06:
Hi again,
Thanks for your answers so far.

Is there a possibility to check if there are animations currently running on the screen? Or that lv_task_handler had actually something to do?

kisvegabor, 2018-11-09:
I added the `lv_anim_count_running` function to `dev-5.3`.

There is no explicit way to know that the library has nothing to do but by default there is only 3 tasks:
- screen refresh: do something only if something happens
- indev read: can be checked with `lv_indev_get_inactive_time` 
- animation: can be checked with the new function: `lv_anim_count_running`

kisvegabor, 2018-11-09:
I updated the documentation:
https://github.com/littlevgl/lvgl/wiki/Porting#task-handling

embeddedt, 2018-11-09:
@StephanSy It's a bit ugly but maybe you could add some timestamp variable to your driver?

kisvegabor, 2018-11-09:
I updated the documentation as you suggested.

The suggestion of @embeddedt  seems quite flexible.

kisvegabor, 2018-11-30:
As this use case is added to the documentation I close this issue.

closed 2018-11-30 by kisvegabor