lvgl/lvgl #316: Action handler issues

hydra, 2018-07-19:
Hi!

So I'm using LVGL with C++.  I have a C++ class which has non-static button callback handlers which need to be called as the handlers need to use the non-static instance members to perform their function.  There are multiple instances of class so that the GUI elements can be repeated.

In order to call the correct NON-STATIC callback handler from LVGL when a button is pressed a static lv_action_t method is called, here's an example.

```
lv_res_t UI_Thing::btn_click_action(lv_obj_t * btn)
{
    UI_Thing *instance = (UI_Thing *)lv_obj_get_free_ptr(btn);
    return instance->onAction_btnClick(btn);
}

lv_res_t UI_Thing::onAction_btnClick(lv_obj_t * btn)
{
    someInstanceValue++; // do something with the instance member variables...
    return LV_RES_OK;
}

void UI_Thing::createElements(void) 
{
    lv_btn_set_action(instanceButton, LV_BTN_ACTION_CLICK, UI_Thing::btn_click_action);
    lv_obj_set_free_ptr(instanceButton, this);
}

class UI_Thing
{
public:
    void createElements(void);

private:
    static lv_res_t btn_click_action(lv_obj_t * btn);

    lv_res_t onAction_btnClick(lv_obj_t * btn);

    int someInstanceValue;
}
```

This seems fairly obvious stuff up to this point of course and works fine for one button with one action.

The problem, which you probably guessed by now, are:
a) when you want to add another button you have to duplicate both the instance method and static method for each button you want to click.
b) when you want to handle different events from a button (like LONG PRESS *and* CLICK) you again have to create two methods for each event.

The solution appears to be something like:
* use lv_set_free_num() using a different ID for each button.
* creating one static handler for each type of event that can occur, that function then lookup an instance method to call based on the lv_object and the free_num obtained from the object.

However, this still has too much duplication for my liking, since you have to create multiple static handlers.  What I want is a solution where there is a *single* static event handler which is used by all buttons, however this isn't possible as in such a static event handler there is no way to determine what the event is, so I can't say, iterate over a static map of "event + button_id" to "C++ method".

e.g.
```
EventMappingEntry eventMappings[] = {
    { action: LV_BTN_ACTION_CLICK, button: BUTTON_OK_ID, handler : &UI_Thing::onAction_btnOK_click }
    { action: LV_BTN_ACTION_PR, button: BUTTON_OK_ID, handler : &UI_Thing::onAction_btnOK_pressed }
    { action: LV_BTN_ACTION_REL, button: BUTTON_OK_ID, handler : &UI_Thing::onAction_btnOK_released }
    { action: LV_BTN_ACTION_CLICK, button: BUTTON_CANCEL_ID, handler : &UI_Thing::onAction_btnCancel_click }
    { action: LV_BTN_ACTION_PR, button: BUTTON_CANCEL_ID, handler : &UI_Thing::onAction_btnCancel_pressed }
    { action: LV_BTN_ACTION_REL, button: BUTTON_CANCEL_ID, handler : &UI_Thing::onAction_btnCancel_released }
};

...
    lv_btn_set_action(instanceButton1, LV_BTN_ACTION_CLICK, UI_Thing::handleEvent);
    lv_btn_set_action(instanceButton1, LV_BTN_ACTION_PR, UI_Thing::handleEvent);
    lv_btn_set_action(instanceButton1, LV_BTN_ACTION_REL, UI_Thing::handleEvent);
    lv_obj_set_free_num(instanceButton1, BUTTON_OK_ID);
    lv_obj_set_free_ptr(instanceButton1, this);
    lv_btn_set_action(instanceButton2, LV_BTN_ACTION_CLICK, UI_Thing::handleEvent);
    lv_btn_set_action(instanceButton2, LV_BTN_ACTION_PR, UI_Thing::handleEvent);
    lv_btn_set_action(instanceButton2, LV_BTN_ACTION_REL, UI_Thing::handleEvent);
    lv_obj_set_free_num(instanceButton2, BUTTON_CANCEL_ID);
    lv_obj_set_free_ptr(instanceButton2, this);
...
```
Also, the 'set_action' calls can be determined by iterating over the eventMappings table, to further reduce code duplication.

In order to do this, the static event handler needs to be given the action in addtion to the lv_obj_t * reference it's already given.

i.e.  something like this:

```
typedef lv_res_t (*lv_action_t) (struct _lv_obj_t * obj, int action);
```

and change use action calls in the code in methods like lv_btn_signal() like this:

```
...
res = ext->actions[LV_BTN_ACTION_PR](btn, LV_BTN_ACTION_PR);
...
```

Another alternative, is for the action handler to ask the object for the last action, but that requires additional state to be continually held which I don't like.

Thought? Better suggestions?

BesitzeRuf, 2018-07-19:
Hi **hydra**, 

onepossible sulution, because you are using C++, create you own object (classes) for ex. for button (BaseButton, ChildFromBaseButton1 -> override any handlers you need, ChildFromBaseButton2 -> override any handlers you need) ... Then In "class UI_Thing" create you elements with relevant classes. So the solution is in case of C++ use inheritance.

hydra, 2018-07-19:
Hi @BesitzeRuf, thanks for your reply.  

I don't want to go down the path of creating a wrapper for every type of LV object (yet, if it all).

I think that regardless of any inheritance model, the Button class still needs a way of working out what the Button instance is from the lv_object and still has no way of determining the action without a single handler for each type of action.

I simply want each of my UI elements (e.g. window with 3 buttons), to extend from a base class which can call instance methods.

I'm after something like this:

```
class UI_EventHandler {
  static lv_res_t handleEvent(lv_obj_t *btn) {
    // get UI_EventHandler instance from button.
    instance->onEvent(btn);
  }
  lv_res_t onEvent(lv_obj_t *btn, int action) {
    // get button id from btn.
    // use mapping table provided by derived class, find appropriate event mapping entry, call instance event handler for btn.
}

class UI_Clock : public UI_EventHandler {
  lv_res_t onAction_time_clicked(lv_obj *) { ... }
  lv_res_t onAction_configure_clicked(lv_obj *) { ... }

  void createElements() {
    // create buttons, set ids, set callback handler for each button to be UI_EventHandler::handleEvent
  }
  static vector<EventMappingEntry> eventMappings = {
    { action: LV_BTN_ACTION_CLICK, id: LABEL_TIME_ID, handler : &onAction_time_clicked},
    { action: LV_BTN_ACTION_CLICK, id: BUTTON_CONFIGURE_ID, handler : &onAction_configure_clicked}
  };
}
```

The idea is that all I want to define in the UI_Clock class is a static list of events, and their handlers.

hydra, 2018-07-19:
As a quick follow up, this is what I currently have to do.

ui_event_handler.h
```
#pragma once

extern "C" {
    #include "lvgl.h"
}

namespace UI {

class UI_EventHandler;

typedef lv_res_t (UI_EventHandler::*actionHandler)(lv_obj_t *btn);

enum UI_EventID_e {
    UI_EVENT_ON_BUTTON_CLICK,
    UI_EVENT_ON_BUTTON_PRESSED,
    UI_EVENT_ON_BUTTON_RELEASED
};

class EventMappingEntry {
public:
    LV_OBJ_FREE_NUM_TYPE elementId;
    UI_EventID_e eventId;
    actionHandler handler;
};

class UI_EventHandler {
public:

    UI_EventHandler(vector<EventMappingEntry> *_eventMappings) :
        eventMappings(_eventMappings)
    {}
    
    static lv_res_t onButtonClick(lv_obj_t *btn) {
        UI_EventHandler *instance = (UI_EventHandler *)lv_obj_get_free_ptr(btn);
        return instance->onAction(btn, UI_EVENT_ON_BUTTON_CLICK);
    }

    static lv_res_t onButtonPressed(lv_obj_t *btn) {
        UI_EventHandler *instance = (UI_EventHandler *)lv_obj_get_free_ptr(btn);
        return instance->onAction(btn, UI_EVENT_ON_BUTTON_PRESSED);
    }

    static lv_res_t onButtonPressed(lv_obj_t *btn) {
        UI_EventHandler *instance = (UI_EventHandler *)lv_obj_get_free_ptr(btn);
        return instance->onAction(btn, UI_EVENT_ON_BUTTON_RELEASED);
    }

    lv_res_t onAction(lv_obj_t *btn, UI_EventID_e eventId) {

        LV_OBJ_FREE_NUM_TYPE elementId = lv_obj_get_free_num(btn);

        for (vector<EventMappingEntry>::iterator i = eventMappings->begin(); i != eventMappings->end(); ++i) {
            EventMappingEntry entry = *i;

            if (entry.eventId == eventId && entry.elementId == elementId) {
                return (this->*entry.handler)(btn);
            }
        }
        return LV_RES_OK;
    }

private:
    vector<EventMappingEntry> *eventMappings;
};

} // namespace
```

example eventMappings and constructor
```
enum clockConfigureElementId_e {
    UI_CLOCK_CONFIGURE_OK
};

vector<EventMappingEntry> UI_ClockConfigure::uiEventMappings = {
    { {elementId: UI_CLOCK_CONFIGURE_OK, eventId: UI_EVENT_ON_BUTTON_CLICK, handler: (UI::actionHandler)&UI_ClockConfigure::onAction_btnOK_clicked} }
};

UI_ClockConfigure::UI_ClockConfigure() :
    UI_EventHandler(&uiEventMappings)
{
}
```

example create elements
```
    lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);
    lv_win_set_title(win, "Set Clock");

    lv_obj_t *btnOK = lv_win_add_btn(win, SYMBOL_OK, UI_ClockConfigure::onButtonClick); // <-- USING STATIC CALLBACK HANDLER IN UI_EventHandler HERE.
    lv_obj_set_free_num(btnOK, UI_CLOCK_CONFIGURE_OK);
    lv_obj_set_free_ptr(btnOK, this);

    lv_win_add_btn(win, SYMBOL_CLOSE, lv_win_close_action);

```

example handler
```
lv_res_t UI_ClockConfigure::onAction_btnOK_clicked(lv_obj_t *btn)
{
    // ... get the new time from other member elements
    // ... set the hardware clock

    lv_obj_t * win = lv_win_get_from_btn(btn);
    lv_obj_del(win);

    return LV_RES_INV;
}
```

hydra, 2018-07-20:
Hiya,

As mentioned, I don't want a solution that allows the event handler callback function to ask lvgl for the 'current action type' / `lv_btn_get_current_action_type` because it means that additional state is held internally in lvgl.  a solution where the data is passed at the time it is needed is better.

using `lv_btn_action_t` would only be ok in a handler that is only for buttons.  For a generic handler if other controls create call actions then the action value passed must be different from other action values.  i.e. you can't have two action identifiers with the same value unless the action handler code also looks up the object type which would then also have be to kept in the event mappings table.

a more general event handler would be something like:

```cpp
    static lv_res_t onAction(lv_obj_t *btn, lv_action_id actionId) {
        UI_EventHandler *instance = (UI_EventHandler *)lv_obj_get_free_ptr(btn);
        return instance->onAction(btn, id);
    }

    lv_res_t onAction(lv_obj_t *btn, lv_action_id actionId) {

        LV_OBJ_FREE_NUM_TYPE elementId = lv_obj_get_free_num(btn);

        for (vector<EventMappingEntry>::iterator i = eventMappings->begin(); i != eventMappings->end(); ++i) {
            EventMappingEntry entry = *i;

            if (entry.actionId == actionId && entry.elementId == elementId) {
                return (this->*entry.handler)(btn);
            }
        }
        return LV_RES_OK;
    }
```
and `onButtonClick`/`onButtonPressed`/`onButtonReleased` would be deleted.

hydra, 2018-07-23:
again, i prefer a solution that doesnt require additional state to be held in memory, i.e. for the value that the proposed `lv_indev_get_current_event` returns.

it will probably be ok if you *really* dont want to pass the action as a second parameter with the object, however passing the action to the handler while you already know the state and don't have to store anything seems the most logic time and place to do it to me.

kisvegabor, 2018-07-23:
In regard to memory usage, you are right. For example, button need to store only one function pointer for all action (instead of 3). Its only drawback is if you want to change only the CLICK action you need to change the whole action. But you can modify the CLICK's behaviour with values of variables so really it's not a big deal.

So I'm convinced. :)

However, as it's an API change it can be only in v6.0. :(

hydra, 2018-07-23:
buttons-only isnt a good solution. at the.moment.my example shows buttons, but anywhere a callback to user-code can happen some action reference is required to reduce the code duplication in user-code.

i wouldnt recommend using a define to switch api modes.  the less conditional logic, compile-time or run-time the better.

kisvegabor, 2018-07-23:
If you can wait for v6.0 then, yes it would be better to avoid the "dual-API".

embeddedt, 2018-12-01:
@kisvegabor I'm interested in implementing this feature. What is the final API and behavior supposed to be?

kisvegabor, 2018-12-02:
@embeddedt Great! 

My idea was to add an enum to `lv_obj.h`:
```c
typedef enum
{
  LV_ACTION_CLICK,
  LV_ACTION_PRESS,
...
} lv_action_type_t;
```
We should search the general action type, not use only the current button action. For example `LV_ACTION_CHANGED` can be used by a Text area.

It's only my current idea. We can/should develop/change it as needed.

embeddedt, 2018-12-02:
@kisvegabor That sounds like a reasonable implementation. My only concern is that it might be hard to hide that behind a `#define`.

kisvegabor, 2018-12-02:
> I assume then that a new API named lv_obj_set_action would be created?

Hm, it seems like an interesting idea to make the actions to a basic feature. Now buttons handle the calling of basic actions and it's not trivial in all cases. For example `Click` doesn't have to be called after dragging. What do you think, shall we move these features to `lv_obj`? 
Anyway, I don't see any obvious obstacle to use one generic action function.

To hide it with a `define` we should turn off all action related API functions and structure members. And use `obj->action` instead. Maybe it will look messy ang ugly but it need to be tried.

embeddedt, 2018-12-02:
@kisvegabor I think using a generic action function will solve the problems of @hydra and @amirgon. Let's do it this way.

The only question is how do we handle actions that only apply to specific objects? I think we could create a generic `LV_ACTION_CHANGED` and then leave it to the user to retrieve the change (i.e. for `lv_slider` the user would have to call `lv_slider_get_value`).

kisvegabor, 2018-12-02:
> The only question is how do we handle actions that only apply to specific objects? I think we could create a generic LV_ACTION_CHANGED and then leave it to the user to retrieve the change (i.e. for lv_slider the user would have to call lv_slider_get_value).

@embeddedt 
I agree it should be the user's responsibility to know to which object type was the action assigned. It will be still possible to use a generic button action as @amirgon said.

hydra, 2018-12-03:
you need 2 user-defined variables for c++, one for the object reference (free_ptr), and one for the action reference.

amirgon, 2018-12-03:
@kisvegabor 

> What about my earlier suggestion? You can use `free_ptr` like this:
> 
> ```c
> static void * ctx[8];
> ctx[0] = &x;
> ctx[1] = &y;
> ...
> lv_obj_set_free_ptr(obj, ctx);
> ```

This is possible, but it requires the user to know exactly how many callbacks are associated with each object, maintain `ctx` and refer to the correct `ctx` element in each of his callbacks. It feels like a technical detail we could spare the user.  
Also, I think this makes the API less clear (I need to know about `lv_obj_set_free_ptr` to implement callbacks, this is not immediately obvious).  

I see you point about memory usage.  
How about solving this by letting the user decide if he wants to pay the extra bytes or not? Just add another macro to `lv_conf.h` to enable/disable the extra `user_data` argument. On platforms that have enough RAM (such as ESP32), it can be enabled by default.
I prefer using `user_data` for Micropython binding support (so `user_data` would need to be enabled for Micropython, but platforms that support Micropython have enough RAM anyway).

amirgon, 2018-12-03:
@kisvegabor One more thing, if you worry about RAM.  
Even today you need to keep a function pointer for each callback, even if its not used later on. Why not allocate them dynamically (both the function pointer and `user_data`) when allocating RAM for the object?   
The only disadvantage I see is that the user would need to decide which callbacks he would need in advance, when creating the object (for example by a bitmask, you can know how many and which ones are needed).  
On the lv_obj struct you can have a [flexible array member](https://en.wikipedia.org/wiki/Flexible_array_member) with entry for each callback that the user actually wants to use.

embeddedt, 2018-12-03:
@kisvegabor @amirgon

> Why not allocate them dynamically (both the function pointer and user_data) when allocating RAM for the object?
> The only disadvantage I see is that the user would need to decide which callbacks he [...]

I think that will add too much complexity to the object-specific code. Since we don't know which callbacks the user is going to choose, there will have to be an internal function to determine the index of the callback function. This adds too much indirection and is not worth the 32 bytes saved in my opinion.

kisvegabor, 2018-12-03:
@amirgon You are right, now `lv_btn` has a `lv_action_t actions[LV_BTN_ACTION_NUM];` element which is 4 * 4 bytes. However, the new proposal to use only one action and pass the action type will reduce it to 1 * 4 bytes. So it seems to be a good improvement. 
In addition, with dynamic memory allocation:
- we need a pointer to store the allocated array of functions -> 4 bytes
- if you allocate a memory `lv_mem` it uses 4 extra bytes to store metadata 
- all the allocated actions need 4 bytes

So the equation is: 8 + (N * 4). So we need 12 bytes for 1 action.

Based on @amirgon's suggestion to make it an option to save the context for actions I suggest having a function like:  `lv_obj_set_action_ctx(obj, ACTION_TYPE, &ctx);` and its pair: `lv_obj_get_action_ctx(obj, ACTION_TYPE);`. This way the API can remain the same independently from the settings in `lv_conf.h`. 
What do you think?

amirgon, 2018-12-03:
> * we need a pointer to store the allocated array of functions -> 4 bytes

No you don't. That's the whole idea of [flexible array member](https://en.wikipedia.org/wiki/Flexible_array_member). You allocate it as part of your struct in the same malloc you use to allocate memory for your struct

> * if you allocate a memory `lv_mem` it uses 4 extra bytes to store metadata

No extra allocation.

> * all the allocated actions need 4 bytes
> 
> So the equation is: 8 + (N * 4). So we need 12 bytes for 1 action.

No. All you need is a few bytes for the mask. Apart from this you only need memory for the function pointers themselves, according to the number of enabled callbacks.

@kisvegabor I think you didn't understand my suggestion. Please let me know if you want me to clarify it further.

kisvegabor, 2018-12-03:
@amirgon Sorry, I really misunderstood your concept. Now it's clear. However, besides what  @embeddedt said ("You can only have one flexible array member per struct"), I'm not sure flexible arrays are really supported by every embedded compiler (even if it's a C99 standard). 

Let's summarize what we have so far:
1. Using 1 action function with an  ACTION_TYPE parameter would be a flexible and memory effective improvement. Do you agree?
2. We should store contexts (pointers) to every ACTION_TYPE, but ideally:
   - this feature needs to be optional to save memory
   - can't be added to every object because it'd consume RAM even is the object not uses actions
   - using `free_ptr` with a `void * ctx[] ` is not intuitive but possible

@amirgon said
> I prefer using user_data for Micropython binding support (so user_data would need to be enabled for Micropython, but platforms that support Micropython have enough RAM anyway).

According to it, I think the compromise could be to add an optional `void * ctx[LV_ACTION_TYPE_NUM]` to `lv_obj_t`. As you said the whole context thing becomes important on complex systems where the user works with `C++` or `Micropython`. 

So the signatures could be:
```c
void lv_obj_set_action(obj, action);               //If CTX feature is disabled
void lv_obj_set_action_ctx(obj, action, &ctx);     //If CTX feautre is enabled
```
I'd prefer 2 function prototypes for simplicity if the context feature is disabled. The `ctx` paramter might be confusing for the user who don't want to deal with it.

What do you think?

amirgon, 2018-12-03:
@kisvegabor 
> I'm not sure flexible arrays are really supported by every embedded compiler (even if it's a C99 standard).

Actually, before C99 you would write `arr[0]` instead of `arr[]` to achieve the same result. Flexible arrays are more common than you think. This would definitely be supported by all C compilers.

> 1. Using 1 action function with an  ACTION_TYPE parameter would be a flexible and memory effective improvement. Do you agree?

Yes.

> 2. We should store contexts (pointers) to every ACTION_TYPE, but ideally:
> * this feature needs to be optional to save memory
> * can't be added to every object because it'd consume RAM even is the object not uses actions

Actually no. With my suggestion an object instance that uses no actions will consume no memory at all except from the bitmask which could be as small as one byte (if 8 action types is enough).  
Actually, you won't need to keep even the function pointer you plan to keep if no actions are used, so my suggestion requires **less** RAM, even if you use a single action function with an  ACTION_TYPE parameter per object and even if you don't enable the `ctx` option (1 byte for mask vs. 4 bytes for function pointer that is not used, when the action is not used).

> According to it, I think the compromise could be to add an optional `void * ctx[LV_ACTION_TYPE_NUM]` to `lv_obj_t`

That would be good enough.

> I'd prefer 2 function prototypes for simplicity if the context feature is disabled. The `ctx` paramter might be confusing for the user who don't want to deal with it.

Ok.  
You might want to rename it `user_data` to make it more familiar for the user. I've seen it more often named `user_data` in different libraries that use callbacks.

kisvegabor, 2018-12-03:
@amirgon 
> Actually, before C99 you would write arr[0] instead of arr[] to achieve the same result. Flexible arrays are more common than you think. This would definitely be supported by all C compilers.

and 

> Actually, you won't need to keep even the function pointer you plan to keep if no actions are used,

They sound really good. I begin to be convinced about the flexible arrays. Finally a `void * action_and_ctx[1]` can be used too is `[0]` is not working. 
I suppose the bits of the mask can be set in `lv_obj_set_action_ctx` on demand. My only concern is if the mask changes in runtime than a lot of reallocation of `lv_obj_t` and realign of `ctx` pointers would be required. Of course, it can be implemented but it's not so elegant.

A cleaner solution could be to have only 2 flags:
1. `action_allocated`
2. `ctxs_allocated`

It could also work with flexible arrays. On `lv_obj_set_action` `lv_obj_t` is reallocated to hold an `action` pointer and `action_allocated` is set. On `lv_obj_set_action_ctx`  `action_allocated` and `ctxs_allocated` are set and `lv_obj_t` is reallocated to hold an action and all the contexts. It's not so memory effective as your solution but:
- still very memory effective
- `ACTION_CTX_ENABLE` define is not required, because it is represented only by 1 bit
- simple an clean

To have a common action prototype we can use: 
```c
lv_res_t my_action(lv_obj_t * obj, lv_action_type_t type) 
{
   void * ctx = lv_obj_get_action_ctx(obj, type);

   ...

   return LV_RES_OK;
}
```

And, yes `user_data` is probably more descriptive.

What do you think? @embeddedt I think the proposed solution is less hacky. We can use flexible arrays for one thing (because we can have only one of them). Why not use it for actions?

amirgon, 2018-12-03:
@kisvegabor 
> I suppose the bits of the mask can be set in `lv_obj_set_action_ctx` on demand. My only concern is if the mask changes in runtime than a lot of reallocation of `lv_obj_t` and realign of `ctx` pointers would be required. Of course, it can be implemented but it's not so elegant.

I agree reallocations are not so elegant. Originally I though the user would need to specify the action bit mask when calling the `create` function of an object (another argument to `create`), so no reallocation would be needed.

> It could also work with flexible arrays. On `lv_obj_set_action` `lv_obj_t` is reallocated to hold an `action` pointer and `action_allocated` is set. On `lv_obj_set_action_ctx` `action_allocated` and `ctxs_allocated` are set and `lv_obj_t` is reallocated to hold an action and all the contexts.

That would work. But If someone uses `ctx` on a single action it would cost him RAM to allocate **all** actions instead of only the action he is interested in.  
I'm not against you suggestion, it sounds good. But If you go this far, why not allocate the right number of  `ctx`, and hold a bitmap instead of a bit?  
Do you assume that any platform that would make use of `ctx` would have enough RAM to spare? Or that the use of `ctx` would be relatively rare?

> To have a common action prototype we can use:
> 
> ```c
> lv_res_t my_action(lv_obj_t * obj, lv_action_type_t type) 
> {
>    void * ctx = lv_obj_get_action_ctx(obj, type);
> ```

But why not also pass `ctx` parameter to `my_action`? It could be `NULL` if `ctx` is not used, and can be ignored by the user if he is not interested in it.  
Why ask the user to call `lv_obj_get_action_ctx` every time in each of his callbacks? (assuming the user is taking advantage of `ctx` in his callbacks)

kisvegabor, 2018-12-04:
> Originally I though the user would need to specify the action bit mask when calling the create function of an object (another argument to create), so no reallocation would be needed.

I think it's not so user friendly to set the mask explicitly. It should be hidden from the user. Adding it to the create function might be annoying because action (even more contexts) probably will be used only by a smaller number of objects. So why type the "empty mask" constantly?

> That would work. But If someone uses ctx on a single action it would cost him RAM to allocate all actions instead of only the action he is interested in.

Yes, your suggestion more memory effective but mine also uses extra RAM only if the context is explicitly used. (which will be only for a few objects and in a few cases if we have RAM anyway). But it still more simple and don't need to deal with the mask, reallocation on every new action etc.

> But why not also pass ctx parameter to my_action? It could be NULL if ctx is not used, and can be ignored by the user if he is not interested in it.

So far using `user_data` was not requested by people so I suppose they don't need it too often (or they used `free_ptr` instead). So I think we shouldn't add an extra parameter if they will use it very rarely.

BTW, can you share a dummy C++ code where the `ctx` is used? Just a few lines to illustrate the use case.

amirgon, 2018-12-04:
@kisvegabor I've done some rethinking about the whole thing.  
So you want to have **one callback** per object, and the callback would have an `action_type` parameter to specify what was the event that happened, right?  

In such case the user should register the callback function **exactly once** per object, but may specify **multiple events** that might trigger the callback.  

It's not clear to me how the action would be registered.  
Will there be a function for it (like `lv_btn_set_action`) that receives both the callback function and the action type? Or does the registration of the callback happen once, when the object is created or added? What happens if the user wants to specify multiple actions with his callback?  Does he call a function multiple times? Or uses some bit mask?

I'm also having second thoughts about how the callback context should be defined. After looking into it, it seems that at least for Micropython, I would need a context only **per callback function**, not **per `action_type`**. So if we have a single callback per object, a single context might be enough.

embeddedt, 2018-12-04:
@kisvegabor @amirgon Lots of possibilities being suggested here. I'm going to try to address my main concerns.

I think we should avoid reallocation where possible. My basis for saying that is that reallocation introduces more fragmentation which is not good for embedded systems. Heaps are less dangerous when there is little/no reallocation.

I think we should not allow the mask to be changed once the object is created. The user should know when they create the object which actions they want to use.

> It's not clear to me how the action would be registered.
> Will there be a function for it (like lv_btn_set_action) that receives both the callback function and the action type? Or does the [...]

Possibly the user could set a callback function and simply ignore events they're not interested in. This resembles how the Win32 API works.

embeddedt, 2018-12-04:
@kisvegabor 

> I think we can have LV_ACTION_FOCUS and LV_ACTION_DEFOCUS too to allow changing the image in your action function. What do you think @embeddedt and @amirgon?

That would be great!

kisvegabor, 2018-12-05:
@amirgon @embeddedt 

> Possibly the user could set a callback function and simply ignore events they're not interested in. This resembles how the Win32 API works.

Yes, that was the idea.

> I'm also having second thoughts about how the callback context should be defined. After looking into it, it seems that at least for Micropython, I would need a context only per callback function, not per action_type. So if we have a single callback per object, a single context might be enough.

Great news! In this case, I think, `free_ptr` is intuitive enough. Do you agree?

amirgon, 2018-12-05:
@kisvegabor 

> > Possibly the user could set a callback function and simply ignore events they're not interested in. This resembles how the Win32 API works.
> 
> Yes, that was the idea.

- How do you register a callback function? In the `create` function of the object? As a generic lv_obj function?

- Do you assume all callbacks would ever have the same prototype? Except from `action_type`, wouldn't some callbacks require additional parameters?  
For example today `lv_btnm_action_t` callback receives `const char *txt` parameter. How do you plan to convert it to a uniform callback prototype?

> Great news! In this case, I think, `free_ptr` is intuitive enough. Do you agree?

It is good enough, assuming only **one** callback function for each object.
Personally I would still keep a `user_data` pointer for every callback as a rule, since the user might want to use `free_ptr` for other purposes, but if you believe the usage of `user_data` is not so common then I guess we could live with `free_ptr`.

kisvegabor, 2018-12-05:
@amirgon 
> How do you register a callback function? In the create function of the object? As a generic lv_obj function?

Yes, we can use `lv_obj_set_action(obj, action)`.

> Do you assume all callbacks would ever have the same prototype? Except from action_type, wouldn't some callbacks require additional parameters?
For example today lv_btnm_action_t callback receives const char *txt parameter. How do you plan to convert it to a uniform callback prototype?

An extra parameter is really required in some cases. So this action prototype seems reasonable:
`lv_res_t my_action(lv_obj_t * obj, lv_action_type_t type, void * param)`

> Personally I would still keep a user_data pointer for every callback as a rule, since the user might want to use free_ptr for other purposes, but if you believe the usage of user_data is not so common then I guess we could live with free_ptr.

Now we have `LV_OBJ_FREE_NUM_TYPE`. Maybe we can add `LV_OBJ_FREE_PTR_TYPE` to enable the user to use a `struct` if required. Now the question seems obvious to why not have only one `struct` as `user_data` which can hold numbers and pointer as the user's requirements.

amirgon, 2018-12-05:
@kisvegabor 
> An extra parameter is really required in some cases. So this action prototype seems reasonable:
> `lv_res_t my_action(lv_obj_t * obj, lv_action_type_t type, void * param)`

That is not so pretty. The user who looks at the function prototype would need to guess how to cast `void*` instead of deducing it from the function prototype.  
In Micropython binding I rely on types to know how to convert between C types and Python types. In case of `void*`, I would have to leave it to the user to call the conversion function.  
How about keeping this information in the `lv_obj`, and providing a function to access it, instead of passing it as a `void*` parameter to the callback?

> Now we have `LV_OBJ_FREE_NUM_TYPE`. Maybe we can add `LV_OBJ_FREE_PTR_TYPE` to enable the user to use a `struct` if required. Now the question seems obvious to why not have only one `struct` as `user_data` which can hold numbers and pointer as the user's requirements.

I like this idea. Instead of a predefined free-num and a free-ptr, let the user define what information (and how much information) he would like to add to the underlying lv-obj struct.

embeddedt, 2018-12-06:
> Instead of a predefined free-num and a free-ptr, let the user define what information (and how much information) he would like to add to the underlying lv-obj struct.

How do you "define" this information?

kisvegabor, 2018-12-06:
@embeddedt 
A typedef n `lv_conf.h`?

```c
typedef struct {
   uint8_t num;
   void * ptr;
}lv_obj_user_data_t;
```

embeddedt, 2018-12-06:
@kisvegabor Perfect!

kisvegabor, 2019-02-15:
@amirgon  Reaction to https://github.com/littlevgl/lvgl/issues/785#issuecomment-463363341

> * Every callback function should have a separate registration function, and each registration function should register exactly one callback.

It's important when you set the callback by initializing a `struct`. In this case, we still can have function(s) to set the callback. It's also good when the user wants to modify the callback later.   
E.g. `lv_disp_set_flush_cb`

> * The registration function should receive both the function pointer and `void *user_data` (and can receive other parameters)
> * The callback function should receive `user_data` (and can receive other parameters)

When in `v6.0` object actions will be generalized on `disp` will have more callbacks. With one or two display the memory impact of the `user_data` for each callback is negligible. So this part is ok for me.

What I don't like is the extra parameter in the callback:   
`void disp_flush (void *user_data, lv_disp_t * disp, lv_area_t * area, lv_color_t * color_p);` 

This concept of `user_data` is required very rarely and in most of the cases it's just an extra unused paramter for the user. In a "casual" case one `user_data` even an `int` would be enough to identify the caller. 

Once the user data is a `void *` you can point to an array like:
```c
typedef struct {
   void * flush_data;
   void * fill_data;
}lv_disp_user_data_t
``` 

This `typedef` can go even to `lv_conf.h` and use `lv_disp_user_data_t * user_data` in `lv_disp_drv_t`.

Can it work?

amirgon, 2019-02-15:
@kisvegabor After giving it some more thought, I have another suggestion. (I think you would like it better than the previous one...)  

We need a separate `user_data` for each callback, we need a way  to set it automatically when setting the callback, and access it when the callback is called.
I think we could keep a `user_data` member per callback on the struct and use some **naming convention** to identify it and relate it back to the callback.  

So I'm thinking of something like:

```c
typedef struct {
    void (*disp_flush)(struct lv_disp_drv_t *context, int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p);    
    void (*disp_fill)(struct lv_disp_drv_t *context, int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t color);
    void *disp_flush_user_data;
    void *disp_fill_user_data;
    ...
} lv_disp_drv_t;
```

The important parts here are:

- `disp_flush_user_data` is user data for `disp_flush`  
- `disp_fill_user_data` is the user data for `disp_fill`
- In general, **every `<callback>` member on the struct should have a corresponding `void *<callback>_user_data` member**.
- The first parameter of `disp_flush` and `disp_fill` callback functions is `struct lv_disp_drv_t *context` which holds the user data for each callback.
- In general **every callback will pass a pointer to the struct as the first parameter**.

When setting `disp_flush` in `lv_disp_drv_t` instance, the Micropython binding will look for `disp_flush_user_data` member in the same struct and update it automatically to point to the Micropython object that is associated with the callback.  
When `disp_flush` callback is called, Micropython will take the first parameter of the `disp_flush` callback (which is `struct lv_disp_drv_t *context`) and will look for `disp_flush_user_data` in this `context`.  

I hope I'm not missing anything here, but this looks nice and clean, no need for `user_data` parameter on the callback functions themselves and no need for a special registration function at all (we can keep using something like `void lv_disp_drv_register(lv_disp_drv_t *driver)` to register the driver).  

What do you think?

kisvegabor, 2019-02-18:
@amirgon 
Sounds very good to me! 

Only one thing: my above proposed solution about having a `typedef` in `lv_conf.h` like
```c
typedef struct {
   void * flush_data;
   void * fill_data;
}lv_disp_user_data_t
```

would give more flexibility and still work with your idea. What do you think, can we use it?

amirgon, 2019-02-18:
> Only one thing: my above proposed solution about having a `typedef` in `lv_conf.h` like
> 
> ```c
> typedef struct {
>    void * flush_data;
>    void * fill_data;
> }lv_disp_user_data_t
> ```

@kisvegabor actually, I was thinking of coupling **a callback and its associated user data**, and not coupling together user_data of different callbacks.  

In my opinion this would be more natural and convenient:

```c
typedef struct {
    disp_flush_t disp_flush;
    void *disp_flush_user_data;

    disp_fill_t disp_fill;
    void *disp_fill_user_data;
    ...
} lv_disp_drv_t;
```

The reasons are:

- There is not special relation between user_data of different callbacks.
- When someone adds a callback, with your suggestion he would have to remember there is some separate `user_data_t` struct he has to update. Easy to forget and miss a user_data for a new callback.  
With my suggestion it's easier to see that every callback **must** be accompanied with a `user_data`.
- It's more generic so it would be easier to parse it with an automatic parser (only need to look for `<callback>_user_data` member on the same struct). No need for the parser to know the name of the user_data struct (`lv_disp_user_data_t` in your example).

kisvegabor, 2019-02-19:
> When someone adds a callback, with your suggestion he would have to remember there is some separate `user_data_t` struct he has to update. Easy to forget and miss a user_data for a new callback.

Having `user_data` for every callback is needed only by the Micropython binding so far. So it won't affect the "average" user.

> It's more generic so it would be easier to parse it with an automatic parser (only need to look for `<callback>_user_data` member on the same struct). No need for the parser to know the name of the user_data struct (`lv_disp_user_data_t` in your example).

In your suggestion you will see `disp->drv.disp_fill_user_data`
In my suggestion you will see:  `disp->drv.user_data.disp_fill`

It seems not a big difference to me.

If it could work for you too, I'd prefer the `typedef` because it gives more freedom. The user can `typedef` simply an `uint8_t id` data and use it as `disp->drv.user_data.id = 1`.

amirgon, 2019-02-19:
@kisvegabor 

> Having `user_data` for every callback is needed only by the Micropython binding so far. So it won't affect the "average" user.

I believe this will be required for any other bindings (at least one that implements callbacks in a generic way). It would also be needed for C++ support, if you would like to support class methods as a callbacks.

> In your suggestion you will see `disp->drv.disp_fill_user_data`
> In my suggestion you will see: `disp->drv.user_data.disp_fill`
> 
> It seems not a big difference to me.

Actually there is a difference you might have missed, and it's not in the syntax.  
It's in the support for **Forward Compatibility**, the ability to support new callbacks without changing the code. It stems from the fact the `user_data`, as you suggest, is part of lvgl configuration.

I'll try to explain, by an example:  

---

Suppose you added to lvgl a new callback called `disp->drv.disp_fill2`.  
Now you build Micropython with the new version of lvgl. With your suggestion you **will not** see the `disp_fill2` callback available automatically. (With my suggestion, you will).  

To enable `disp_fill2` callback you would have to edit `lv_conf.h` (or some other configuration file) and change the definition of `lv_disp_user_data_t` by adding `disp_fill2_user_data` member to it.   
You must edit this configuration file manually because the binding script only creates the C file for wrapping lvgl functions and structs. It does not change **lvgl configuration** so it does not generate any H file.  

---

So, if you insist on this approach, we have the following options:

- Drop Forward Compatibility and manage Micropython's `user_data` in lvgl micropython bindings repo. A new callback was added? It will not be supported until accepting a PR that updates Micropythons's `user_data` to support the new callback.
- Let the binding script generate the `user_data` struct for every callback group. This is less favorable and adds complexity because 
  1. I don't think the binding script should change lvgl configuration, since the same configuration could also be modified be the user.
  2. It's not generic. How would the binding script know which `user_data` to create? What happens when we have a new callback group (such as new driver type)?
  3. Today the binding script is simple in the sense that it creates one single C output. It's easy to decide where it is generated, how it is built in the Makefile, etc. Integrating the binding into Micropython is a very simple task.  
If the binding script starts to generate H files, it will make the task of integrating it to different Micropython forks more complex.
- Require the user to edit a configuration file every time a new callback is added or changed. When integrating the Micropython bindings to a new Micropython fork the user would have to define all `user_data` structs or copy it from some template. 
- Keep callbacks as today. Which means - no user data. Only one callback per object or driver is supported.

embeddedt, 2019-02-19:
@kisvegabor If you don't see much of a difference I would go for @amirgon's approach. It's not that different from a user's perspective but from what I can tell it simplifies things a lot.

kisvegabor, 2019-02-19:
@amirgon Okay, I'm convinced. Thank you for the detailed explanation. 

What I can still imagine to increase flexibility is to `typedef` the type of `disp_fill_user_data`. E.g.:
```c
typedef void * lv_disp_user_data_t;   /*Can be the default as it's the most flexible*/

/*OR*/

typedef struct {
  uint8_t id;
  const char * name;
} lv_disp_user_data_t;

/*OR*/

typedef enum {
  DISP_MAIN,
  DISP_AUX,
} lv_disp_user_data_t;

...

typedef struct {
    disp_flush_t disp_flush;
    lv_disp_user_data_t disp_flush_user_data;

    disp_fill_t disp_fill;
    lv_disp_user_data_t disp_fill_user_data;
    ...
} lv_disp_drv_t;
```
I mean the same approach for all user data.

kisvegabor, 2019-02-19:
Awesome! Very happy to see that we found a solution which fit every requirement! :+1:

kisvegabor, 2019-02-21:
Having multiple user data for displays and input devices are not big problem because you have only 1-2 display and indput device. However I realized that having multple user data in `lv_obj_t` for `signal_cb`, `design_cb` and `action_cb` would be really bad in 99% of the cases.

I got an idea. Why not have define like `#define LV_USE_MULTIPLE_USER_DATA` (or so) in `lv_conf.h` to have something like this:
```c
typedef struct {
    disp_flush_t disp_flush;
    disp_fill_t disp_fill;

#if LV_USE_MULTIPLE_USER_DATA
    lv_disp_user_data_t disp_flush_user_data;
    lv_disp_user_data_t disp_fill_user_data;
#else 
    lv_disp_user_data_t user_data;
#endif
} lv_disp_drv_t;
```

With this define all the concerns seems to be solved.

What do you think?

amirgon, 2019-02-21:
@kisvegabor It's possible but I don't like the fact that the callback and its `user_data` are separated.  
So let me suggest something similar that ensures that the per-callback user data won't be forgotten:

```c
#if LV_USE_MULTIPLE_USER_DATA
#define CALLBACK(callback_name) \
callback_name ## _t callback_name; \
callback_name ## _user_data_t callback_name ## _user_data
#else 
#define CALLBACK(callback_name) \
callback_name ## _t callback_name
#endif 

...

typedef struct {
    CALLBACK(disp_flush);
    CALLBACK(disp_fill);

#if LV_USE_SINGLE_USER_DATA
    lv_disp_drv_user_data_t lv_disp_drv_user_data;
#endif

} lv_disp_drv_t;

```

btw, I separated the macro to two: `LV_USE_MULTIPLE_USER_DATA` and `LV_USE_SINGLE_USER_DATA`, so if someone doesn't want `user_data` at all, he could set them both to 0.

kisvegabor, 2019-02-22:
Your suggestion really looks more robust. However, I'm not afraid about forgetting to add the user data. These things are very core things and changes very rarely. 

For better readability, I'd choose the "manual" way. 

Disabling the user data is a good idea!

manison, 2019-02-22:
I tested initial implementation of multiple user data and I don't find it useful. On the contrary, I think it's an overkill. As I commented here https://github.com/littlevgl/lvgl/issues/785#issuecomment-465998152:

> It would make a little sense if the user data would be passed to the callback directly, but `lv_disp_t` is being passed instead. So if anybody _really_ needs distinct user data for each callback, she can define

```c
struct lv_disp_user_data_t {
    void *data_for_flush_cb;
    void *data_for_render_cb;
    ...you get the idea...
};
```

> and then access specific user data in the callback through `lv_disp_t::driver`. But I think the usual scenario is that the driver needs the same user data (typically some low-level handle to underlying graphical context, framebuffer address etc) in all callbacks. Now I need to fill 4 user data for 4 callbacks with the same data. It also means 4 times larger memory footprint. I just think it should be the other way around.

It might be maybe useful for some reason to MicroPython, but that's just one special usage case of LittlevGL. I believe majority of users won't benefit from this and it will put a burden on them.

I have never seen any other framework that would provide distinct user data for each and every callback. Usually you get some handle that you can associate your data with (as with `lv_obj_set_free_ptr`) and then you receive the handle in callback and you can extract your data from it.

As of C++ (from initial post) one would use `this` as the user data and then do whatever she wants with it in the callback.

> a) when you want to add another button you have to duplicate both the instance method and static method for each button you want to click.
> b) when you want to handle different events from a button (like LONG PRESS _and_ CLICK) you again have to create two methods for each event.

Not true. You can use either inheritance or "events" (as C++ does not provide events as language construct, you can use e.g. `std::function` and lambdas).

```cpp
class BaseButton
{
private:
    lv_obj_t *_obj;

public:
    Button()
    {
        _obj = lv_btn_create();
        lv_obj_set_free_ptr(_obj, this);
        lv_obj_set_action(_obj, LV_BTN_ACTION_CLICK, Button::ClickActionStatic);
    }

private:
    static lv_res_t ClickActionStatic(lv_obj_t * obj)
    {
        BaseButton *that = reinterpret_cast<BaseButton *>(lv_obj_get_free_ptr(obj);
        return that->OnClick();
    }

protected:
    virtual lv_res_t OnClick() = 0;
};

class FirstButton : public BaseButton
{
protected:
    virtual lv_res_t OnClick() override
    {
        ...FirstButton clicked....
    }
}

class SecondButton : public BaseButton
{
protected:
    virtual lv_res_t OnClick() override
    {
        ...SecondButton clicked....
    }
}
```

or (maybe more ellegant):

```cpp
class Button
{
private:
    lv_obj_t *_obj;
    std::function<lv_res_t(void)> _click;

    static lv_res_t ClickActionStatic(lv_obj_t * obj)
    {
        BaseButton *that = reinterpret_cast<BaseButton *>(lv_obj_get_free_ptr(obj);
        return that->_click();
    }

public:
    void OnClick(std::function<lv_res_t(void)> handler)
    {
        _click = handler;
    }
};

Button btn;
btn.OnClick([]() {
    ...btn clicked...
});
```

I still thinks we don't need multiple user data at all but should they be supported then make it opt-in as @kisvegabor suggested with `LV_USE_MULTIPLE_USER_DATA`.

amirgon, 2019-02-22:
> Your suggestion really looks more robust. However, I'm not afraid about forgetting to add the user data. These things are very core things and changes very rarely.

@kisvegabor 
If this changes very rarely, **we could drop Forward Compatibility**. As I already wrote on a [previous post](https://github.com/littlevgl/lvgl/issues/316#issuecomment-465026732):

> So, if you insist on this approach, we have the following options:
>
> * Drop Forward Compatibility and manage Micropython's `user_data` in lvgl micropython bindings repo. A new callback was added? It [...]

But does it really happens very rarely? Can't we expect almost every new lvgl component to contain some new callback?  

I see a strong resistance to the `user_data` per-callback from @manison side, and I can see his point.  
In many cases a single `user_data` per struct is enough.   

So why do we need `user_data` **per-callback** in Micropython?  
Because we **can't predict** how the callback would be defined by the user. In case of Micropython, the user is free to define the callback however he wants, **for example, as a Python Closure!** I expect this to be very common, and you can see an example of this on the [demo script](https://github.com/littlevgl/lv_binding_micropython/blob/9b63af0c67946b8268e410aa3dcd6e0d6f45b7e0/gen/advanced_demo.py#L46). In such case every callback needs to be associated with a different `user_data`.  

A `user_data` per-callback could be defined: 

- In LittlevGL sources (with my suggestion of adding `user_data` per-callback next to each callback), or
- On the Micropython Bindings configuration (by setting the `...user_data_t` struct to contain `user_data` for each callback), 

In both cases, when you add a new callback you need to **actively add a `user_data` per callback, somewhere**.  
This is also true when adding a new lvgl component that contains some new callback.

To be honest, there is another option, which is more complex and expensive on the Micropython side.
The script could analyze (when parsing lvgl sources) which are all the callbacks per struct, and try to allocate `user_data` dynamically (when?) as an array of pointers to each per-callback `user_data`. I'm still not sure how difficult would it be to implement it this way, but it's worth mentioning that this option also exists. Recall I want to avoid changing lvgl configuration so I would not like to create the `...user_data_t` structs on lvgl configuration (on a header file), that's why I need to allocate `user_data` dynamically in this case.

---

**However, no matter what approach is selected, there is one thing which is the most important requirement for Micropython bindings:**

### A strict naming convention to identify the `user_data` for any callback.

`gen_mpy.py` analyzes lvgl headers in a **generic way**. It's not aware of what callbacks lvgl has, but it identifies them when it sees "a pointer to a function" function parameter or struct member.  

Now - it needs to find the corresponding `user_data`. How does it called? `free_ptr`?? `<callback>_user_data`?  Where can it be found?

My point is - there needs to be a convention for the `user_data` name and location so an automatic script could identify it for any existing or future callback.

manison, 2019-02-22:
> I see a strong resistance to the `user_data` per-callback from @manison side, and I can see his point.
In many cases a single `user_data` per struct is enough.

True, I'm against this idea 😃 I don't want to sound rough but if this complicates things for MicroPython then let it be complex on MicroPython side and do not complicate things for majority of other users. I really don't know the process MicroPython uses for bindings and would be happy to learn more to understand it. On the other hand as I already mentioned above I don't know any other framework providing the luxury of separate user data per callback. How does MicroPython bind in this situation? Can't you use C++ model I mentioned above, i.e. saving instance pointer as your user data?

> Recall I want to avoid changing lvgl configuration so I would not like to create the `...user_data_t` structs on lvgl configuration (on a header file), that's why I need to allocate `user_data` dynamically in this case.

Can't you generate definitions into separate header file that would then be included from the config header? That way you wouldn't modify the config itself.

> `gen_mpy.py` analyzes lvgl headers in a **generic way**.

This is your script or is it from MicroPython's distribution?

> ...there needs to be a convention for the `user_data` name and location so an automatic script could identify it for any existing or future callback.

This is doable. There can be also some kind of annotation in comment (just throwing ideas).

embeddedt, 2019-02-22:
I might be opening another can of worms here, but I have an idea... can we come up with some standard macro for declaring a callback inside a struct that will also generate a user data variable beside it? The macro itself would probably look very ugly, but here is a rough example from the top of my head:

```c
#define LV_CALLBACK(type) lv_##type##_cb_t type##_cb; void * type##_user_data;

typedef void (*lv_disp_flush_cb_t)(lv_area_t *area_p, int x1, int y1, int x2, int y2);

typedef struct {
        LV_CALLBACK(disp_flush);
} lv_some_struct_t;
```

That macro would then generate the following code:

```c
typedef struct {
        lv_disp_flush_cb_t disp_flush_cb;
        void * disp_flush_user_data;
} lv_some_struct_t;
```

I'm not sure if @amirgon's script is sophisticated enough to detect the use of the `LV_CALLBACK` macro but that might be solution to get the best of both worlds.

amirgon, 2019-02-22:
> True, I'm against this idea  I don't want to sound rough but if this complicates things for MicroPython then let it be complex on MicroPython side and do not complicate things for majority of other users.

@manison I believe we can find common grounds, for example by configuring either a single `user_data` or `user_data`-per-callback in `lv_conf.h`.

> I really don't know the process MicroPython uses for bindings and would be happy to learn more to understand it

I provided a lot of explanations on this thread and on #557. After reading them please let me know specifically what you don't understand and I'll be happy to clarify.

> How does MicroPython bind in this situation? Can't you use C++ model I mentioned above, i.e. saving instance pointer as your user data?

No, I don't think you got the idea. It is unrelated to C++.

> Can't you generate definitions into separate header file that would then be included from the config header? That way you wouldn't modify the config itself.

It may be possible but it complicates things. We already discussed this on this thread on the posts above.

> This is your script or is it from MicroPython's distribution?

This is my script. Micropython does not provide a way to automatically bind to external libraries.

> This is doable. There can be also some kind of annotation in comment (just throwing ideas).

The library we use to parse C headers works well only on preprocessed code, so no comments annotations are available.
We would need some naming convention on the identifier names themselves.
There was a lot of discussions on this subject on #557, please have a look there.

amirgon, 2019-02-22:
> I might be opening another can of worms here, but I have an idea... can we come up with some standard macro for declaring a callback inside a struct that will also generate a user data variable beside it?

@embeddedt could you clarify how this is different from [my suggestion here](https://github.com/littlevgl/lvgl/issues/316#issuecomment-466137769)?

embeddedt, 2019-02-22:
@amirgon Oops, that's my fault. I didn't realize you had suggested that before.

I think your idea is cleaner than adding it manually.

kisvegabor, 2019-02-23:
I prefer to add the callback and user data explicitly because it increases the readability. If the user checks the implementation of `lv_disp_drv_t` he will clearly see what is there.

I don't think too much new callbacks will be added. The new object features shall use the one generic action callback. Probably new action types (e.g. `LV_ACTION_CLICK`) will be added instead.

By manually adding user_data forward compatibility is still ensured for Micropython. Just care must be taken to not forget to add the user data. But I think it's not an issue because of rare changes.

I'm ok with having multiple user data if it makes binding easier. However if all user data/callback is used by micropython, no user data remains for the actual user to store an ID or something.  So we should have a define to enable user data for each callback and other to enable a simple user data. Do you agree?

amirgon, 2019-02-23:
@kisvegabor 

> I prefer to add the callback and user data explicitly because it increases the readability. If the user checks the implementation of `lv_disp_drv_t` he will clearly see what is there.

Ok.

> I don't think too much new callbacks will be added. The new object features shall use the one generic action callback. Probably new action types (e.g. `LV_ACTION_CLICK`) will be added instead.

That's good!

> By manually adding user_data forward compatibility is still ensured for Micropython. Just care must be taken to not forget to add the user data. But I think it's not an issue because of rare changes.

Yes. And care must also be taken to keep naming conventions, and **name the `user_data` fields correctly**.  
That's because Micropython bindings would rely on naming convention to find the right `user_data` for each callback.  
So if a callback is called, for example, `flush_cb`, its user_data must be called `flush_user_data` and nothing else. From a quick look at [`lv_hal_disp.h` on `multi-disp` branch](https://github.com/littlevgl/lvgl/blob/multi-disp/lv_hal/lv_hal_disp.h), it looks ok.

> I'm ok with having multiple user data if it makes binding easier. However if all user data/callback is used by micropython, no user data remains for the actual user to store an ID or something. So [...]

Do you mean two independent defines `LV_USE_MULTIPLE_USER_DATA` and `LV_USE_SINGLE_USER_DATA` as I suggested in a [previous post](https://github.com/littlevgl/lvgl/issues/316#issuecomment-466137769)?  
 I think it would be a good idea, to let the user configure whether he wants

- A single `user_data` per struct
- A `user_data` per callback
- No `user_data` at all.

This can be achieved by these defines.  

But I don't see a need for **both** single `user_data` per struct **and** `user_data` per callback, at least not for Micropython.  
Once you associate a Micropython object with a callback, you can keep all extra data (ID or something) in that Micropython object. So the Micropython user will not need it, the same way he wouldn't need `free_ptr` for lvgl object. (`free_ptr` is also used internally by Micropython Bindings, to associate lvgl object with Micropython object).

amirgon, 2019-02-24:
@kisvegabor  
Looking at  [`lv_hal_disp.h` on `multi-disp` branch](https://github.com/littlevgl/lvgl/blob/multi-disp/lv_hal/lv_hal_disp.h), I see that we need **two** steps to get from the callback to the user_data:

```c
void (*flush_cb)(struct _disp_t * disp, const lv_area_t * area, lv_color_t * color_p);
```

`flush_cb` receives `struct _disp_t * disp`. So the callback needs to access `disp->driver.flush_user_data`.   

- We know `disp`, this is the first parameter to the callback, by convention.
- We know `flush_user_data`, this is the user_data for `flush_cb` by convention.

**But what is the convention for the `driver` field?**

I'm trying to think how to generalize this for any callback. A callback doesn't have to be on a "driver", it could be an object callback for example, there we will not have a "driver" member to access.

**So the question is:**  
Given a "generic" callback, whether it's a driver callback or an object callback or any other callback: 
```
void (*some_callback_cb)(struct s_t *s, ...)
```
What is the convention to access `user_data` on the callback? (`s->[???].some_callback_user_data`)

kisvegabor, 2019-02-24:
> But I don't see a need for **both** single `user_data` per struct **and** `user_data` per callback, at least not for Micropython.
Once you associate a Micropython object with a callback, you can keep all extra data (ID or something) in that Micropython object. So the Micropython user will not need it, the same way he wouldn't need `free_ptr` for lvgl object. (`free_ptr` is also used internally by Micropython Bindings, to associate lvgl object with Micropython object).

I see, I really forget that in (Micro)python it's that simple. :) 
However, making the two `user_data` declaration independent is easy and increases flexibility without any disadvantages. So I say why not!?

> Looking at [`lv_hal_disp.h` on `multi-disp` branch](https://github.com/littlevgl/lvgl/blob/multi-disp/lv_hal/lv_hal_disp.h), I see that we need **two** steps to get from the callback to the user_data:

That's true and really messes up things. I've realized that it's fine to pass the drivers as a parameter. `lv_disp/indev_t` are containing only run-time data. The calling `indev` or `disp` can be get by `lv_indev_get_act()` and `lv_refr_get_disp_rendering()` in case if they are required for some reason.
I update the `multi-disp` branch accordingly.

amirgon, 2019-02-24:
@kisvegabor 
> However, making the two `user_data` declaration independent is easy and increases flexibility without any disadvantages. So I say why not!?

Sure, I agree there are no disadvantages.

> I've realized that it's fine to pass the drivers as a parameter

That's better now :smile:
btw I couldn't find where `lv_disp_user_data_t` is defined. It's `void*` by default, right?

kisvegabor, 2019-02-25:
I merged the `multi-disp` and other feature branches to `dev-6.0`. These types are added to `lv_conf_templ.h` And defined to `void *` by `lv_conf_checker.h` if not defined.

[8 more replies]

closed 2019-03-13 by kisvegabor