lvgl/lvgl #317: lv_obj_set_click function doesn't have impact on Drop Down List object.

TMarkivULCRobotics, 2018-07-19:
The Drop Down List object is still clickable after lv_obj_set_click function call.

The following line of a code is supposed to disable clicking of the object. But doesn't work with the Drop Down.

`lv_obj_set_click(ddlist_object, false);`

BesitzeRuf, 2018-07-19:
This function disables the clicking of the object, because the main object of the ddlist is page, but not labeles, which this page contains. As solution, you can add function directly to lv_ddlist.c, in it you need to disable clickable attribute of selected option. I think it should be enough.

kisvegabor, 2018-07-19:
Hi,

The label is not clickable by default. But as you mentioned  the `ddlist` is built from an `lv_page` which contains two objects:
1. background
2. scrollable part

`lv_obj_set_click(ddlist, false)` makes only the bacgrund object unclicable.

To make the scrollable part (which is usually clicked) unclickable you can use: `lv_obj_set_click(lv_page_get_scrl(obj), false)`

TMarkivULCRobotics, 2018-07-20:
Hi All,
Thank you for your responses.

What I understood from your posts is that:

1 the ddlist is the collections of labels. I can dis/enable click attribute for each label in the ddlist by modifying the library (lv_ddlist.c). But I dont want to do this. I have only two options in ddlist. And at some circumstances I want automacially set the ddlist to particular option and disable the ddlist (make it read only). lv_obj_set_click was considered as perfect function for this purpose ☺

2 background and scrollable part : I need a little bit of clarification here. I created a container:
    `lv_obj_t * container= lv_cont_create(main_container, NULL);`
Then I created a ddlist:
    `lv_obj_t * ddlist = lv_ddlist_create(container, NULL);`
a) May I say that the "container" is the background you mentioned?
b) Is the ddlist the scrollable part?
c) Is my ddlist still built from the page? How can I use that page in `lv_obj_set_click(lv_page_get_scrl(obj), false)` function call?

3 Is that documented? As for me it is obvious that when we have an object, here it is ddlist, and we want to disable the clicking of the object than we are disabling the ddlist and not its sub objects - labels.

Still have no luck with this issue.
Please correct me where I am wrong here.

TMarkivULCRobotics, 2018-07-20:
After debugging the `indev_search_obj()` function I found that in the ddlist case the library is working with both objects - the ddlist object itself and the  ddlist's page object. So if one of these objects is clickable, then I will have clickable ddlist object on the screen.

So I get this thing working as I wanted after the following code:
`lv_obj_t * ddlist = lv_ddlist_create(any_object, NULL);`
`lv_obj_set_click(lv_page_get_scrl(ddlist ), false);`
`lv_obj_set_click(ddlist , false);`

This approach is not optimal for embedded systems. And a recursion might cause a big and not easy for debugging problem particularly in embedded systems with small RAM memory size. Fortunately not my case ☺.

kisvegabor, 2018-07-22:
Hi,

In LittlevGL object types can be inherited from each other. The most basic example is the button:
Basic object (to set position, size etc.) -> Container (to have fit and layout features) -> Button (to handle actions). The derived object (button) can be passed to ancestor types' functions (`lv_cont_...`
 and `lv_obj_...`)

In the case of `ddlist` this inheritance looks like this: base object -> container -> page -> ddlist

An extra trick is that the `page` is a compounded object, built from 2 containers (a background and a scrollable). So when you write `lv_ddlist create` really an `lv_page` will be created which means two containers.

About the labels on the `ddlist`, it's only one label broken into multiple lines using `\n`.

It might seem complicated at first look but it ensures the reusability of the object types. It's not well documented because it is not important for the user who just uses the library.

However, in your case, it becomes important because you wanted to use something which is not a built-in feature. Why there isn't a built-in `lv_ddlist_set_click` function? Because it's not a typical use case to make the `ddlist` not clickable. The label is not clickable on the `ddlist` but you need to turn off the clickability of the two containers of the page.

TMarkivULCRobotics, 2018-07-23:
Thanks for support and explanation.

closed 2018-07-23 by TMarkivULCRobotics

kisvegabor, 2018-07-23:
Your very welcome!